ardera/flutter-pi #226: flickering area on the screen

iboldurev, 2022-02-18:
Hi, after starting the project, I have a problem, when update the state, I see blink on the screen

if run the project without rotate parameters everything works fine, but if I pass a rotate, get this problem

flutter-pi -r 270 /home/pi/testapp

attached video https://youtu.be/565MMXQY2Yk

ardera, 2022-02-18:
It should actually work even with the `could not find a preferred output mode` message. What pi are you using? Maybe attach the output if the `raspinfo` command

ardera, 2022-02-19:
Can you try commenting out the `hdmi_force_hotplug`, `hdmi_mode`, `hdmi_group` & `hdmi_timings` config options? I know the wiki for the display says you need those but I'm not sure how those work in combination with full KMS, could be they cause some problems. If you really need to force some specific timings there are other ways to do that for full KMS

(note: full KMS is the vc4-kms-v3d overlay, vc4-fkms-v3d means fake-KMS or firmware-KMS)

iboldurev, 2022-02-20:
If i comment lines

hdmi_group=2
hdmi_mode=87
hdmi_timings=320 0 80 16 32 1480 0 16 4 12 0 0 0 60 0 42000000 3

Then the screen does not start, only the backlight is visible, but if I run the application in the console, there are no errors.

ardera, 2022-02-20:
that basically means your display has a broken EDID (`hdmi_timings` etc is a workaround for that)

You can try adding a `video=HDMI-A-1:320x1480@62` parameter (again, with commented out `hdmi_*` in config.txt) to `/boot/cmdline.txt`. If that doesn't work you can try forcing a custom edid with the correct timings set. (See [here](https://wiki.archlinux.org/title/kernel_mode_setting), section `forcing modes and EDID`)

Other than that, I can add some commandline options to force a specific mode, but not sure when I'll get to that.

iboldurev, 2022-02-20:
Hi, @ardera everything worked when i added video=HDMI-A-1:320x1480@62,rotate=90 and comment all hdmi_*

but I faced a new problem (artifacts on the screen)

[image]

[image]

ardera, 2022-02-28:
Seems like something about the timings is still off. If you just put `video=HDMI-A-1:320x1480@62`, KMS will calculate some default timings (you can calculate them yourself [here](https://tomverbeure.github.io/video_timings_calculator)) which don't match what the panel expects, at least when looking at the values for the `hdmi_timings` command.

Not sure how you specify custom timings for a HDMI display though. I think you can only specify custom timings by editing the EDID

iboldurev, 2022-03-01:
This my display https://www.waveshare.com/wiki/11.9inch_HDMI_LCD

in docs I found this line Working with Raspberry P

max_framebuffer_height=1480
max_usb_current=1
hdmi_group=2
hdmi_mode=87
hdmi_timings=320 0 80 16 32 1480 0 16 4 12 0 0 0 60 0 42000000 3

Now I use hdmi_timings=320 0 80 16 32 1480 0 16 4 12 0 0 0 60 0 42000000 3

but I see https://user-images.githubusercontent.com/17103170/154847745-f31808d2-189c-4a2a-8eb2-aaf86172f6d3.jpeg

```
hdmi_timings=
<h_active_pixels>320 
<h_sync_polarity>0 
<h_front_porch>80 
<h_sync_pulse>16 
<h_back_porch>32 
<v_active_lines>1480 
<v_sync_polarity>0 
<v_front_porch>16 
<v_sync_pulse>4 
<v_back_porch>12 
<v_sync_offset_a>0 
<v_sync_offset_b>0 
<pixel_rep>0 
<frame_rate>60 
<interlaced>0 
<pixel_freq>42000000 
<aspect_ratio>3
```

iboldurev, 2022-03-08:
My EDID

```
edid-decode (hex):

00 ff ff ff ff ff ff 00 4c 2d b0 05 00 00 00 00
0d 13 01 03 80 10 09 78 0a ee 91 a3 54 4c 99 26
0f 50 54 bf ef 80 71 4f 81 00 81 40 81 80 95 00
95 0f a9 40 b3 00 02 3a 80 18 71 38 2d 40 58 2c
45 00 a0 5a 00 00 00 1e 01 1d 00 bc 52 d0 1e 20
b8 28 55 40 a0 5a 00 00 00 1e 00 00 00 fd 00 18
4b 1a 51 17 00 0a 20 20 20 20 20 20 00 00 00 fc
00 53 79 6e 63 4d 61 73 74 65 72 0a 20 20 01 d1

02 03 23 f1 4b 93 04 05 14 03 12 10 1f 20 21 22
23 09 07 07 83 01 00 00 e2 00 0f 67 03 0c 00 10
00 b8 2d 01 1d 80 d0 72 1c 16 20 10 2c 25 80 a0
5a 00 00 00 9e 01 1d 80 18 71 1c 16 20 58 2c 25
00 a0 5a 00 00 00 9e 01 1d 00 72 51 d0 1e 20 6e
28 55 00 a0 5a 00 00 00 1e 8c 0a d0 90 20 40 31
20 0c 40 55 00 a0 5a 00 00 00 18 8c 0a d0 8a 20
e0 2d 10 10 3e 96 00 a0 5a 00 00 00 18 00 00 57

----------------

Block 0, Base EDID:
  EDID Structure Version & Revision: 1.3
  Vendor & Product Identification:
    Manufacturer: SAM
    Model: 1456
    Made in: week 13 of 2009
  Basic Display Parameters & Features:
    Digital display
    Maximum image size: 16 cm x 9 cm
    Gamma: 2.20
    RGB color display
    First detailed timing is the preferred timing
  Color Characteristics:
    Red  : 0.6396, 0.3300
    Green: 0.2998, 0.5996
    Blue : 0.1503, 0.0595
    White: 0.3125, 0.3291
  Established Timings I & II:
    IBM     :   720x400    70.081663 Hz   9:5     31.467 kHz     28.320000 MHz
    DMT 0x04:   640x480    59.940476 Hz   4:3     31.469 kHz     25.175000 MHz
    Apple   :   640x480    66.666667 Hz   4:3     35.000 kHz     30.240000 MHz
    DMT 0x05:   640x480    72.808802 Hz   4:3     37.861 kHz     31.500000 MHz
    DMT 0x06:   640x480    75.000000 Hz   4:3     37.500 kHz     31.500000 MHz
    DMT 0x08:   800x600    56.250000 Hz   4:3     35.156 kHz     36.000000 MHz
    DMT 0x09:   800x600    60.316541 Hz   4:3     37.879 kHz     40.000000 MHz
    DMT 0x0a:   800x600    72.187572 Hz   4:3     48.077 kHz     50.000000 MHz
    DMT 0x0b:   800x600    75.000000 Hz   4:3     46.875 kHz     49.500000 MHz
    Apple   :   832x624    74.551266 Hz   4:3     49.726 kHz     57.284000 MHz
    DMT 0x10:  1024x768    60.003840 Hz   4:3     48.363 kHz     65.000000 MHz
    DMT 0x11:  1024x768    70.069359 Hz   4:3     56.476 kHz     75.000000 MHz
    DMT 0x12:  1024x768    75.028582 Hz   4:3     60.023 kHz     78.750000 MHz
    DMT 0x24:  1280x1024   75.024675 Hz   5:4     79.976 kHz    135.000000 MHz
    Apple   :  1152x870    75.061550 Hz 192:145   68.681 kHz    100.000000 MHz
  Standard Timings:
    DMT 0x15:  1152x864    75.000000 Hz   4:3     67.500 kHz    108.000000 MHz
    DMT 0x1c:  1280x800    59.810326 Hz  16:10    49.702 kHz     83.500000 MHz
    DMT 0x20:  1280x960    60.000000 Hz   4:3     60.000 kHz    108.000000 MHz
    DMT 0x23:  1280x1024   60.019740 Hz   5:4     63.981 kHz    108.000000 MHz
    DMT 0x2f:  1440x900    59.887445 Hz  16:10    55.935 kHz    106.500000 MHz
    DMT 0x30:  1440x900    74.984427 Hz  16:10    70.635 kHz    136.750000 MHz
    DMT 0x33:  1600x1200   60.000000 Hz   4:3     75.000 kHz    162.000000 MHz
    DMT 0x3a:  1680x1050   59.954250 Hz  16:10    65.290 kHz    146.250000 MHz
  Detailed Timing Descriptors:
    DTD 1:  1920x1080   60.000000 Hz  16:9     67.500 kHz    148.500000 MHz (160 mm x 90 mm)
                 Hfront   88 Hsync  44 Hback  148 Hpol P
[... 46 lines cut ...]
    Supports_AI
    DC_36bit
    DC_30bit
    DC_Y444
    Maximum TMDS clock: 225 MHz
  Detailed Timing Descriptors:
    DTD 3:  1920x1080i  50.000000 Hz  16:9     28.125 kHz     74.250000 MHz (160 mm x 90 mm)
                 Hfront  528 Hsync  44 Hback  148 Hpol P
                 Vfront    2 Vsync   5 Vback   15 Vpol P Vfront +0.5 Odd Field
                 Vfront    2 Vsync   5 Vback   15 Vpol P Vback  +0.5 Even Field
    DTD 4:  1920x1080i  60.000000 Hz  16:9     33.750 kHz     74.250000 MHz (160 mm x 90 mm)
                 Hfront   88 Hsync  44 Hback  148 Hpol P
                 Vfront    2 Vsync   5 Vback   15 Vpol P Vfront +0.5 Odd Field
                 Vfront    2 Vsync   5 Vback   15 Vpol P Vback  +0.5 Even Field
    DTD 5:  1280x720    60.000000 Hz  16:9     45.000 kHz     74.250000 MHz (160 mm x 90 mm)
                 Hfront  110 Hsync  40 Hback  220 Hpol P
                 Vfront    5 Vsync   5 Vback   20 Vpol P
    DTD 6:   720x576    50.000000 Hz   5:4     31.250 kHz     27.000000 MHz (160 mm x 90 mm)
                 Hfront   12 Hsync  64 Hback   68 Hpol N
                 Vfront    5 Vsync   5 Vback   39 Vpol N
    DTD 7:   720x480    59.940060 Hz   3:2     31.469 kHz     27.000000 MHz (160 mm x 90 mm)
                 Hfront   16 Hsync  62 Hback   60 Hpol N
                 Vfront    9 Vsync   6 Vback   30 Vpol N
Checksum: 0x57
```

ardera, 2022-03-18:
Can you try to edit the EDID and only leave one detailed timing descriptor that has the same timings as `hdmi_timings=320 0 80 16 32 1480 0 16 4 12 0 0 0 60 0 42000000 3` and then force it, similar to [these instructions](https://monado.freedesktop.org/edid-override.html)

BigGitWorld, 2022-06-25:
@iboldurev 
Did you solve the problem?
I have a similar problem and at first I want to test above recommended solutions. I hope that work for me and will report the result here.

iboldurev, 2022-06-25:
> @iboldurev Вы решили проблему? У меня похожая проблема, и сначала я хочу протестировать рекомендуемые выше решения. Я надеюсь, что это сработает для меня, и сообщу о результате здесь.

Hi, no 

if you can fix it let me know

iboldurev, 2022-06-25:
> @iboldurev Did you solve the problem? I have a similar problem and at first I want to test the above recommended solutions. I hope that work for me and will report the result here.

Hi, no

if you can fix it let me know

iboldurev, 2022-06-25:
Hi @BigGitWorld, no

if you can fix it let me know

closed 2023-09-08 by github-actions[bot]