ardera/flutter-pi #229: recent omxplayer build break

jwinarske, 2022-02-23:
Picked up this build break since your last set of changes.  I was pulling tip of tree, but now I'm considering on snapping to a commit.

I have disabled omxplayer in flutter-pi recipe for now.

```
ERROR: Task (/__w/meta-flutter/meta-flutter/recipes-graphics/flutter-pi/flutter-pi-release_git.bb:do_compile) failed with exit code '1'
[207](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:207)
| /__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/recipe-sysroot-native/usr/bin/x86_64-poky-linux/x86_64-poky-linux-clang -DBUILD_OMXPLAYER_VIDEO_PLAYER_PLUGIN -DBUILD_RAW_KEYBOARD_PLUGIN -DBUILD_TEXT_INPUT_PLUGIN -DHAS_EGL -DHAS_FBDEV -DHAS_GBM -DHAS_KMS -I/__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/build -I/__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/git/include -I/__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/git/include/plugins -I/__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/recipe-sysroot/usr/include/libdrm -target x86_64-poky-linux  -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -mlittle-endian -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -Wdate-time  --sysroot=/__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0=/usr/src/debug/flutter-pi-release/git-r0                      -fdebug-prefix-map=/__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0=/usr/src/debug/flutter-pi-release/git-r0                      -fdebug-prefix-map=/__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/recipe-sysroot=                      -fdebug-prefix-map=/__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/recipe-sysroot-native=  -target x86_64-poky-linux  -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -mlittle-endian -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -Wdate-time  --sysroot=/__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/recipe-sysroot -DNDEBUG -I/__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/recipe-sysroot/usr/include/libdrm -O2 -Wall -Wextra -Wno-unused-function -Wno-sign-compare -Wno-missing-field-initializers -ggdb -MD -MT CMakeFiles/flutter-pi.dir/src/plugins/omxplayer_video_player.c.o -MF CMakeFiles/flutter-pi.dir/src/plugins/omxplayer_video_player.c.o.d -o CMakeFiles/flutter-pi.dir/src/plugins/omxplayer_video_player.c.o -c /__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/git/src/plugins/omxplayer_video_player.c
[208](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:208)
| /__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/git/src/plugins/omxplayer_video_player.c:29:9: warning: 'LOG_ERROR' macro redefined [-Wmacro-redefined]
[209](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:209)
| #define LOG_ERROR(...) fprintf(stderr, "[omxplayer video player plugin] " __VA_ARGS__)
[210](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:210)
|         ^
[211](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:211)
| /__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/git/include/collection.h:439:9: note: previous definition is here
[212](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:212)
| #define LOG_ERROR(fmtstring, ...) fprintf(stderr, "[%s] " fmtstring, __file_logging_name, ##__VA_ARGS__)
[213](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:213)
|         ^
[214](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:214)
| /__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/git/src/plugins/omxplayer_video_player.c:33:12: warning: 'LOG_DEBUG' macro redefined [-Wmacro-redefined]
[215](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:215)
| #   define LOG_DEBUG(...) do {} while (0)
[216](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:216)
|            ^
[217](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:217)
| /__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/git/include/collection.h:441:9: note: previous definition is here
[218](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:218)
| #define LOG_DEBUG(fmtstring, ...) do {} while (0)
[219](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:219)
|         ^
[220](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:220)
| /__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/git/src/plugins/omxplayer_video_player.c:1531:77: error: too few arguments to function call, expected 2, have 1
[221](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:221)
|     return plugin_registry_is_plugin_present("omxplayer video_player plugin");
[222](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:222)
|            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                ^
[223](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:223)
| /__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/git/include/pluginregistry.h:99:6: note: 'plugin_registry_is_plugin_present' declared here
[224](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:224)
| bool plugin_registry_is_plugin_present(
[225](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:225)
|      ^
[226](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:226)
| /__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/git/src/plugins/omxplayer_video_player.c:1534:25: error: conflicting types for 'omxpvidpp_init'
[227](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:227)
| enum plugin_init_result omxpvidpp_init(struct flutterpi *flutterpi, void **userdata_out) {
[228](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:228)
|                         ^
[229](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:229)
| /__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/git/include/plugins/omxplayer_video_player.h:92:12: note: previous declaration is here
[230](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:230)
| extern int omxpvidpp_init(void);
[231](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:231)
|            ^
[232](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:232)
| /__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/git/src/plugins/omxplayer_video_player.c:1545:6: error: conflicting types for 'omxpvidpp_deinit'
[233](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:233)
| void omxpvidpp_deinit(struct flutterpi *flutterpi, void *userdata) {
[... cut ...]
| /__w/meta-flutter/yocto-honister/poky/build/tmp/work/core2-64-poky-linux/flutter-pi-release/git-r0/git/include/pluginregistry.h:115:13: note: expanded from macro 'FLUTTERPI_PLUGIN'
[248](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:248)
|                 .deinit = (_deinit) \
[249](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:249)
|                           ^~~~~~~~~
[250](https://github.com/meta-flutter/meta-flutter/runs/5311026200?check_suite_focus=true#step:16:250)
| 4 warnings and 3 errors generated.
```

ardera, 2022-02-24:
ah, yeah omxplayer videoplayer defaults to off now that's why I didn't catch it. (If you can, use the gstreamer video player instead)

- fixed in 2b9d25466c76eaeb1c62090d6867f751abe7a9a3

ardera, 2022-02-24:
though maybe it's better to make it default to on again, so people don't get breakages

closed 2022-02-24 by ardera

jwinarske, 2022-02-24:
@ardera that fixed it.  Thanks for the quick fix

jwinarske, 2022-02-25:
@ardera I updated recipe to default to gstreamer.  CI jobs are all passing, and images are available for test.  I suspect there will be some runtime deps missing.  Do you have a sample app to validate this with?

ardera, 2022-02-26:
@jwinarske The gstreamer video player shouldn't have any additional runtime dependencies, though I made it a soft-error if some dependencies for the gstreamer player couldn't be found while configuring, so people don't get build breakages. (It'll print a message in that case)

The new gstreamer video player just implements the default platform interface of the official `video_player` package, so you can use the video_player example:
```dart
import 'package:video_player/video_player.dart';
import 'package:flutter/material.dart';

void main() => runApp(VideoApp());

class VideoApp extends StatefulWidget {
  @override
  _VideoAppState createState() => _VideoAppState();
}

class _VideoAppState extends State<VideoApp> {
  VideoPlayerController _controller;

  @override
  void initState() {
    super.initState();
    _controller = VideoPlayerController.network(
        'https://flutter.github.io/assets-for-api-docs/assets/videos/bee.mp4')
      ..initialize().then((_) {
        // Ensure the first frame is shown after the video is initialized, even before the play button has been pressed.
        setState(() {});
      });
  }

  @override
  Widget build(BuildContext context) {
    return MaterialApp(
      title: 'Video Demo',
      home: Scaffold(
        body: Center(
          child: _controller.value.isInitialized
              ? AspectRatio(
                  aspectRatio: _controller.value.aspectRatio,
                  child: VideoPlayer(_controller),
                )
              : Container(),
        ),
        floatingActionButton: FloatingActionButton(
          onPressed: () {
            setState(() {
              _controller.value.isPlaying
                  ? _controller.pause()
                  : _controller.play();
            });
          },
          child: Icon(
            _controller.value.isPlaying ? Icons.pause : Icons.play_arrow,
          ),
        ),
      ),
    );
  }

  @override
  void dispose() {
    super.dispose();
    _controller.dispose();
  }
}
```

ardera, 2022-02-28:
> though I made it a soft-error if some dependencies for the gstreamer player couldn't be found while configuring, so people don't get build breakages. (It'll print a message in that case)

Turns out I forgot to push that part. It's working now,