ardera/flutter-pi #231: Has anybody experience with the i.MX7D

bottee, 2022-02-24:
Would be great if anybody has experiences about the graphics performance on a i.MX7D?

Especially as the i.MX7D has no GPU.

Thanks in advance!

LeoBound, 2022-05-13:
Did you manage to get this to work? I'm interested in trying it out on an i.MX6 but struggling to figure it out.

bottee, 2022-05-16:
I'm still(!) waiting for the dev board. But ist is quite clear that it will not work as the i.MX7D has no GPU.

About the i.MX6 I found the following:

- https://www.toradex.com/de/blog/flutter-on-torizon
- https://blog.basyskom.com/2020/how-to-run-flutter-on-an-embedded-device/

But I have no experience with the i.MX6 on my own.

ardera, 2022-05-16:
It won't work I'm afraid. One of the requirements is hardware 3D acceleration.
Though I'm working on making it run without 3D acceleration too.