ardera/flutter-pi #233: Gstreamer no audio output

redargon, 2022-02-28:
I read in #218 that you had not yet implemented audio for gstreamer. Is this still the case? I have tried a few different video formats, but do not get any audio output.

ardera, 2022-02-28:
Yep that's still the case, working on it

redargon, 2022-03-01:
Great, thanks for the fast feedback.

ioridev, 2022-05-18:
When will this be worked on?

Binozo, 2022-11-28:
@ardera 
Sorry but I couldn't get it to work.

Here are my logs:

```
[flutter-pi] WARNING: display didn't provide valid physical dimensions. The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.

(flutter-pi:627): GLib-GObject-WARNING **: 17:49:37.994: g_object_set_is_valid_property: object class 'GstPlayBin' has no property named 'force-sw-decoders'
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[gstreamer video_player] Could not query buffering state. (gst_element_query)
[... cut ...] laybin/GstPlaySink:playsink/GstBin:abin/GstOpenALSink:openalsink0:
sink not negotiated.)
[gstreamer video player] gstreamer error: code: 1, domain: gst-stream-error-quark, msg: Internal data stream error. (debug info: ../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0/GstSoupHTTPSrc:source:
streaming stopped, reason not-negotiated (-4))
[compositor] GPU does not supported the desired HW plane order.
  Some UI layers may be invisible.

```

ardera, 2022-11-28:
Are you running raspbian buster? buster has pretty old gstreamer 1.14, bullseye has 1.18. Gstreamer has seen a lot of improvements between those versions, I'm pretty sure upgrading to bullseye will fix it.

Or, if you rely on buster, try building & installing latest gstreamer yourself.

ardera, 2022-11-28:
@Binozo can you try installing `gstreamer1.0-alsa` if it's not installed already?

Binozo, 2022-11-28:
Now it started working :)

Thank you!

ardera, 2022-11-28:
Np, I'll add that to the dependencies section then.

tobiasht, 2023-03-02:
Has anyone managed go get audio working? I am using the [`audioplayers`](https://pub.dev/packages/audioplayers) package. Audio works fine in Mac and Web.

```
[audioplayers plugin] Call missing mandatory parameter player_id.
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(illegalargument, Expected `arg['player_id'] to be a string., null, null)
#0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:653:7)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:315:18)
<asynchronous suspension>
#2      AudioPlayer.play (package:audioplayers/src/audioplayer.dart:131:5)
<asynchronous suspension>
#3      GameNotifier.operatorNextStep (package:save_the_penguins/app/features/game/providers/game_providers.dart:93:7)
<asynchronous suspension>
```

tobiasht, 2023-03-14:
@ardera have you gotten audio to work?

innomatica, 2023-03-18:
I'm having exactly the same PlatformException as @ardera. Flutter version is 3.7.7.

According to 'audioplayers', this playerId is [set to uuid.v4 by default](https://github.com/bluefireteam/audioplayers/blob/main/packages/audioplayers/lib/src/audioplayer.dart#L99). 

This doesn't seem to be understood by the method channel when [the audio source is set.](https://github.com/bluefireteam/audioplayers/blob/main/packages/audioplayers/lib/src/audioplayer.dart#L228) And the PlatformException was thrown [by this](https://github.com/flutter/flutter/blob/master/packages/flutter/lib/src/services/message_codecs.dart#L168) 

I tried to set the playerId manually with a random string, but to no avail.

edlman, 2025-01-20:
Hello, I installed `flutter-pi` and run an app with VideoPlayer (using pkg flutterpi_gstreamer_video_player). It plays video, but there is no audio. When I play the same video file on the same RPi in different player (VLC, QtMediaPlayer, Chrome browser), it works. So the audio settings (alsa or whatever) is ok.
I'm confused and don't know how to fix it. 
My RPi is running RasbperryPi OS based on Debian 11 (Bullseye), latest flutter pi (downloaded and built yesterday), Flutter app is built using latest Flutter 3.27.2