ardera/flutter-pi #234: Video Player not working

dcbacarro, 2022-03-04:
Hi, I am trying to run the video_player example app on my Raspberry Pi 4 using flutter-pi but it seems to not work for me. There are logs about the asset file not being found, but it also fails to play network videos. I followed the guide to install all the pre-requisites. Please help. Thank you!

Here's the logs when I try to play videos with the example app:

```
[keyboard] Could not find "XKBMODEL" property inside "/etc/default/keyboard". Default value will be used.[keyboard] Could not find "XKBLAYOUT" property inside "/etc/default/keyboard". Default value will be used.[keyboard] Could not find "XKBVARIANT" property inside "/etc/default/keyboard". Default value will be used.[keyboard] Could not find "XKBOPTIONS" property inside "/etc/default/keyboard". Default value will be used.[flutter-pi] WARNING: display didn't provide valid physical dimensions. The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
===================================
display mode:
  resolution: 1920 x 1080
  refresh rate: 60Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_device EGL_EXT_platform_wayland EGL_KHR_platform_wayland EGL_EXT_platform_x11 EGL_KHR_platform_x11 EGL_MESA_platform_gbm EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 3.1 Mesa 20.3.5"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_draw_instanced GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_pack_reverse_row_order GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_NV_pixel_buffer_object GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_primitive_bounding_box GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_NV_image_formats GL_OES_draw_elements_base_vertex GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_EXT_buffer_storage GL_EXT_float_blend GL_EXT_geometry_point_size GL_EXT_geometry_shader GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_geometry_point_size GL_OES_geometry_shader GL_OES_shader_image_atomic GL_EXT_texture_compression_s3tc_srgb GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_EXT_EGL_image_storage GL_MESA_framebuffer_flip_y GL_EXT_texture_query_lod "
===================================
flutter: Observatory listening on http://127.0.0.1:46323/SGDwbYFk-08=/
[gstreamer video_player plugin] on_initialize
[compositor] Could not move cursor to front. Mouse cursor may be invisible. drmdev_plane_get_max_zpos_value: Invalid argument
[compositor] GPU does not supported the desired HW plane order.
  Some UI layers may be invisible.
[ERROR:flutter/lib/ui/ui_dart_state.cc(209)] Unhandled Exception: FormatException: Message corrupted
#0      StandardMessageCodec.decodeMessage (package:flutter/src/services/message_codecs.dart:325:7)
#1      BasicMessageChannel.send (package:flutter/src/services/platform_channel.dart:60:18)
<asynchronous suspension>
#2      VideoPlayerApi.setMixWithOthers (package:video_player_platform_interface/messages.dart:406:9)
<asynchronous suspension>
#3      VideoPlayerController.initialize (package:video_player/video_player.dart:348:7)
<asynchronous suspension>

[compositor] Preferred framebuffer depth for hardware cursor is not supported by flutter-pi.
[gstreamer video_player plugin] respond success on_create
[gstreamer video player] gstreamer error: code: 5, domain: gst-resource-error-quark, msg: Could not open resource for reading. (debug info: ../gst/gio/gstgiosrc.c(334): gst_gio_src_get_stream (): /GstPipeline:pipeline0/GstURIDecodeBin:src/GstGioSrc:source:
Could not open location file://video_player/assets/Butterfly-209.mp4 for reading: Operation not supported)
[gstreamer video player] gstreamer error: code: 5, domain: gst-resource-error-quark, msg: Could not open resource for reading. (debug info: ../gst/gio/gstgiobasesrc.c(123): gst_gio_base_src_start (): /GstPipeline:pipeline0/GstURIDecodeBin:src/GstGioSrc:source:
No input stream provided by subclass)
[gstreamer video player] gstreamer error: code: 4, domain: gst-core-error-quark, msg: GStreamer error: state change failed and some element failed to post a proper error message with the reason for the failure. (debug info: ../libs/gst/base/gstbasesrc.c(3556): gst_base_src_start (): /GstPipeline:pipeline0/GstURIDecodeBin:src/GstGioSrc:source:
Failed to start)
[gstreamer video player] gstreamer error: code: 5, domain: gst-resource-error-quark, msg: Could not open resource for reading. (debug info: ../gst/gio/gstgiosrc.c(334): gst_gio_src_get_stream (): /GstPipeline:pipeline0/GstURIDecodeBin:src/GstGioSrc:source:
Could not open location file://video_player/assets/Butterfly-209.mp4 for reading: Operation not supported)
[gstreamer video player] gstreamer error: code: 5, domain: gst-resource-error-quark, msg: Could not open resource for reading. (debug info: ../gst/gio/gstgiobasesrc.c(123): gst_gio_base_src_start (): /GstPipeline:pipeline0/GstURIDecodeBin:src/GstGioSrc:source:
No input stream provided by subclass)
[gstreamer video player] gstreamer error: code: 4, domain: gst-core-error-quark, msg: GStreamer error: state change failed and some element failed to post a proper error message with the reason for the failure. (debug info: ../libs/gst/base/gstbasesrc.c(3556): gst_base_src_start (): /GstPipeline:pipeline0/GstURIDecodeBin:src/GstGioSrc:source:
Failed to start)
[gstreamer video_player plugin] on_receive_evch
[ERROR:flutter/lib/ui/ui_dart_state.cc(209)] Unhandled Exception: FormatException: Message corrupted
#0      StandardMessageCodec.decodeMessage (package:flutter/src/services/message_codecs.dart:325:7)
#1      BasicMessageChannel.send (package:flutter/src/services/platform_channel.dart:60:18)
<asynchronous suspension>
#2      VideoPlayerApi.setMixWithOthers (package:video_player_platform_interface/messages.dart:406:9)
<asynchronous suspension>
[... cut ...]
#0      StandardMessageCodec.decodeMessage (package:flutter/src/services/message_codecs.dart:325:7)
#1      BasicMessageChannel.send (package:flutter/src/services/platform_channel.dart:60:18)
<asynchronous suspension>
#2      VideoPlayerApi.setMixWithOthers (package:video_player_platform_interface/messages.dart:406:9)
<asynchronous suspension>
#3      VideoPlayerController.initialize (package:video_player/video_player.dart:348:7)
<asynchronous suspension>

[gstreamer video_player plugin] on_receive_evch
```

ardera, 2022-03-18:
How are you creating the `VideoPlayerController`? Can you paste the dart code? Ideally the one where you try to playback from web, so I can try to reproduce it

dcbacarro, 2022-03-18:
Hi @ardera , it's not my actual code, I just tried to run the example app from the flutter `video_player` package [here](https://github.com/flutter/plugins/tree/main/packages/video_player/video_player/example).
Here's the code for the remote video:

```dart
class _BumbleBeeRemoteVideo extends StatefulWidget {
  @override
  _BumbleBeeRemoteVideoState createState() => _BumbleBeeRemoteVideoState();
}

class _BumbleBeeRemoteVideoState extends State<_BumbleBeeRemoteVideo> {
  late VideoPlayerController _controller;

  Future<ClosedCaptionFile> _loadCaptions() async {
    final String fileContents = await DefaultAssetBundle.of(context)
        .loadString('assets/bumble_bee_captions.vtt');
    return WebVTTCaptionFile(
        fileContents); // For vtt files, use WebVTTCaptionFile
  }

  @override
  void initState() {
    super.initState();
    _controller = VideoPlayerController.network(
      'https://flutter.github.io/assets-for-api-docs/assets/videos/bee.mp4',
      closedCaptionFile: _loadCaptions(),
      videoPlayerOptions: VideoPlayerOptions(mixWithOthers: true),
    );

    _controller.addListener(() {
      setState(() {});
    });
    _controller.setLooping(true);
    _controller.initialize();
  }

  @override
  void dispose() {
    _controller.dispose();
    super.dispose();
  }

  @override
  Widget build(BuildContext context) {
    return SingleChildScrollView(
      child: Column(
        children: <Widget>[
          Container(padding: const EdgeInsets.only(top: 20.0)),
          const Text('With remote mp4'),
          Container(
            padding: const EdgeInsets.all(20),
            child: AspectRatio(
              aspectRatio: _controller.value.aspectRatio,
              child: Stack(
                alignment: Alignment.bottomCenter,
                children: <Widget>[
                  VideoPlayer(_controller),
                  ClosedCaption(text: _controller.value.caption.text),
                  _ControlsOverlay(controller: _controller),
                  VideoProgressIndicator(_controller, allowScrubbing: true),
                ],
              ),
            ),
          ),
        ],
      ),
    );
  }
}
```

ioridev, 2022-05-17:
Same issue

> ===================================
OpenGL ES information:
  version: "OpenGL ES 3.1 Mesa 20.3.5"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_filter_anisotropic GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_draw_instanced GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_pack_reverse_row_order GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_NV_pixel_buffer_object GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_primitive_bounding_box GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_view GL_KHR_context_flush_control GL_NV_image_formats GL_OES_draw_elements_base_vertex GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_OES_texture_view GL_EXT_buffer_storage GL_EXT_float_blend GL_EXT_geometry_point_size GL_EXT_geometry_shader GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_geometry_point_size GL_OES_geometry_shader GL_OES_shader_image_atomic GL_EXT_texture_compression_s3tc_srgb GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_EXT_EGL_image_storage GL_MESA_framebuffer_flip_y GL_EXT_texture_query_lod "
===================================
flutter: The Dart VM service is listening on http://127.0.0.1:37697/FtLuoqxysOk=/
[gstreamer video_player plugin] on_initialize
[gstreamer video_player plugin] respond success on_create
[gstreamer video player] gstreamer error: code: 5, domain: gst-resource-error-quark, msg: Could not open resource for reading. (debug info: ../gst/gio/gstgiosrc.c(334): gst_gio_src_get_stream (): /GstPipeline:pipeline0/GstURIDecodeBin:src/GstGioSrc:source:
Could not open location file://flutter_assets/assets/sample.mp4 for reading: Operation not supported)
[gstreamer video player] gstreamer error: code: 5, domain: gst-resource-error-quark, msg: Could not open resource for reading. (debug info: ../gst/gio/gstgiobasesrc.c(123): gst_gio_base_src_start (): /GstPipeline:pipeline0/GstURIDecodeBin:src/GstGioSrc:source:
No input stream provided by subclass)
[gstreamer video player] gstreamer error: code: 4, domain: gst-core-error-quark, msg: GStreamer error: state change failed and some element failed to post a proper error message with the reason for the failure. (debug info: ../libs/gst/base/gstbasesrc.c(3556): gst_base_src_start (): /GstPipeline:pipeline0/GstURIDecodeBin:src/GstGioSrc:source:
Failed to start)
[gstreamer video player] gstreamer error: code: 5, domain: gst-resource-error-quark, msg: Could not open resource for reading. (debug info: ../gst/gio/gstgiosrc.c(334): gst_gio_src_get_stream (): /GstPipeline:pipeline0/GstURIDecodeBin:src/GstGioSrc:source:
Could not open location file://flutter_assets/assets/sample.mp4 for reading: Operation not supported)
[gstreamer video player] gstreamer error: code: 5, domain: gst-resource-error-quark, msg: Could not open resource for reading. (debug info: ../gst/gio/gstgiobasesrc.c(123): gst_gio_base_src_start (): /GstPipeline:pipeline0/GstURIDecodeBin:src/GstGioSrc:source:
No input stream provided by subclass)
[gstreamer video player] gstreamer error: code: 4, domain: gst-core-error-quark, msg: GStreamer error: state change failed and some element failed to post a proper error message with the reason for the failure. (debug info: ../libs/gst/base/gstbasesrc.c(3556): gst_base_src_start (): /GstPipeline:pipeline0/GstURIDecodeBin:src/GstGioSrc:source:
Failed to start)
[compositor] GPU does not supported the desired HW plane order.
  Some UI layers may be invisible.
[gstreamer video_player plugin] on_receive_evch

ardera, 2022-05-17:
Ah I see. Should be fixed now

ioridev, 2022-05-17:
Resolved. Thank you.

closed 2022-05-17 by ardera