ardera/flutter-pi #237: Release build error after including flutter_gpiod

PMG-Zachary, 2022-03-16:
When I attempt to make a release build of my app using gen_snapshot_linux_x64, I get the error:

> Inconsistency detected by ld.so: dl-version.c: 205: _dl_check_map_versions: Assertion `needed != NULL' failed!

I've made release builds of this app before without updating Flutter or my engine binaries, it stopped working after including [flutter_gpiod](https://github.com/ardera/flutter_packages/tree/main/packages).

I'm currently on Flutter version 2.8.1 and Dart 2.15.1 with the appropriate engine binaries / gen_snapshot_linux_x64_release file. I'm attempting to run the file in a WLS install of Kali Linux. These are the exact build commands I'm using:

> flutter clean

> flutter build bundle

> C:\Users\zach\Documents\flutter\bin\cache\dart-sdk\bin\dart.exe C:\Users\zach\Documents\flutter\bin\cache\dart-sdk\bin\snapshots\frontend_server.dart.snapshot --sdk-root C:\Users\zach\Documents\flutter\bin\cache\artifacts\engine\common\flutter_patched_sdk_product --target=flutter --aot --tfa -Ddart.vm.product=true --packages .packages --output-dill build\kernel_snapshot.dill --verbose --depfile build\kernel_snapshot.d package:ns_gui/main.dart

> ./gen_snapshot_linux_x64_release --deterministic --snapshot_kind=app-aot-elf --elf=./build/flutter_assets/app.so --strip --sim-use-hardfp ./build/kernel_snapshot.dill

ardera, 2022-03-18:
does it work if you remove flutter_gpiod again? To be honest I highly doubt it has anything to do with flutter_gpiod, that's just some dart code

It seems like that's a problem with the distro or some incompatibility of the compiled gen_snapshot. Did you do a `apt upgrade` (no idea what's the Kali equivalent) some time recently?

PMG-Zachary, 2022-03-18:
Have you seen this error before with the release build? I looked into it a bit and it seems to be some sort of issue with missing dependency, but I can't really find anything specific to this application.

I haven't done `apt upgrade`, updated my Flutter/Dart SDKs, or downloaded a new version of the gen_snapshot specifically to avoid compatibility issues. I'll try commenting out everything that uses flutter_gpiod and removing the dependency, if that doesn't work I'll install a fresh Linux distro and try upgrading all of my SDKs and gen_snapshot to the latest versions.

ardera, 2022-03-18:
I mean normally asserts should only be there to catch application bugs, and if you google the error that's what it says (for example [this](https://stackoverflow.com/questions/22564780/debugging-ld-inconsistency-detected-by-ld-so))

PMG-Zachary, 2022-03-18:
This is one of the threads I looked through when trying to debug, this snippet from the accepted answer leads me to believe it's a dependency issue:

    while (1)
      {
         ElfW(Vernaux) *aux;
         struct link_map *needed = find_needed (strtab + ent->vn_file, map);

        /* If NEEDED is NULL this means a dependency was not found
           and no stub entry was created.  This should never happen.  */
        assert (needed != NULL);
     ...

It seems fairly likely the Linux distro is to blame, but I wanted to check in here in case this is a known issue.

PMG-Zachary, 2022-03-18:
> does it work if you remove flutter_gpiod again? To be honest I highly doubt it has anything to do with flutter_gpiod, that's just some dart code
>
> It seems like that's a problem with the distro or [...]

Yeah, it's still happening with flutter_gpiod commented out. I'll go ahead and close this issue since it seems to just be a problem with my build environment.

closed 2022-03-18 by PMG-Zachary

ardera, 2022-03-18:
Okay, yeah
You can try running with `LD_DEBUG=all` (that's what it says in some comment of the same thread too), maybe that gives you some more info

PMG-Zachary, 2022-03-18:
I tried that before, but couldn't really glean anything useful. This is the full output with LD_DEBUG, in case it makes more sense to you.

        23:     symbol=__vdso_clock_gettime;  lookup in file=linux-vdso.so.1 [0]
        23:     binding file linux-vdso.so.1 [0] to linux-vdso.so.1 [0]: normal symbol `__vdso_clock_gettime' [LINUX_2.6]
        23:     symbol=__vdso_gettimeofday;  lookup in file=linux-vdso.so.1 [0]
        23:     binding file linux-vdso.so.1 [0] to linux-vdso.so.1 [0]: normal symbol `__vdso_gettimeofday' [LINUX_2.6]        23:     symbol=__vdso_time;  lookup in file=linux-vdso.so.1 [0]
        23:     binding file linux-vdso.so.1 [0] to linux-vdso.so.1 [0]: normal symbol `__vdso_time' [LINUX_2.6]
        23:     symbol=__vdso_getcpu;  lookup in file=linux-vdso.so.1 [0]
        23:     binding file linux-vdso.so.1 [0] to linux-vdso.so.1 [0]: normal symbol `__vdso_getcpu' [LINUX_2.6]
        23:     symbol=__vdso_clock_getres;  lookup in file=linux-vdso.so.1 [0]
        23:     binding file linux-vdso.so.1 [0] to linux-vdso.so.1 [0]: normal symbol `__vdso_clock_getres' [LINUX_2.6]        23:
        23:     file=libpthread.so.0 [0];  needed by ./gen_snapshot_linux_x64_release [0]
        23:     find library=libpthread.so.0 [0]; searching
        23:      search cache=/etc/ld.so.cache
        23:       trying file=/lib/x86_64-linux-gnu/libpthread.so.0
        23:
        23:     file=libpthread.so.0 [0];  generating link map
        23:       dynamic: 0x00007f44d821dd70  base: 0x00007f44d8202000   size: 0x0000000000020450
        23:         entry: 0x00007f44d8209c00  phdr: 0x00007f44d8202040  phnum:                 11
        23:
        23:
        23:     file=libm.so.6 [0];  needed by ./gen_snapshot_linux_x64_release [0]
        23:     find library=libm.so.6 [0]; searching
        23:      search cache=/etc/ld.so.cache
        23:       trying file=/lib/x86_64-linux-gnu/libm.so.6
        23:
        23:     file=libm.so.6 [0];  generating link map
        23:       dynamic: 0x00007f44d8200d90  base: 0x00007f44d80bd000   size: 0x0000000000144110
        23:         entry: 0x00007f44d80cc200  phdr: 0x00007f44d80bd040  phnum:                  9
        23:
        23:
        23:     file=libc.so.6 [0];  needed by ./gen_snapshot_linux_x64_release [0]
        23:     find library=libc.so.6 [0]; searching
        23:      search cache=/etc/ld.so.cache
        23:       trying file=/lib/x86_64-linux-gnu/libc.so.6
        23:
        23:     file=libc.so.6 [0];  generating link map
        23:       dynamic: 0x00007f44d80b5b60  base: 0x00007f44d7ef8000   size: 0x00000000001c4440
        23:         entry: 0x00007f44d7f1ff80  phdr: 0x00007f44d7ef8040  phnum:                 12
        23:
        Inconsistency detected by ld.so: dl-version.c: 205: _dl_check_map_versions: Assertion `needed != NULL' failed!

PMG-Zachary, 2022-04-05:
I fixed the problem, just in case anyone else has this issue and winds up here somehow. I had gen_snapshot_linux_x64 saved in my repo for building, but switching branches and pulling the file back down caused it to get corrupted. All I needed to do was redownload that file.