ardera/flutter-pi #241: flutter-pi localhost doubt

alexmac98, 2022-03-30:
Hello, I'm developing a flutter application that communicates with a RabbitMQ broker that runs on port 5672 on the machine (RaspberryPi).

I know that the Android emulators use the IP 10.0.2.2 to point to the machine's localhost. So, essentially, if the machine has a service running on localhost:5672, the flutter application would connect to 10.0.2.2:5672 in order to interact with that service.

Does this also happen in flutter-pi? Is the 10.0.2.2 used to point to the localhost of the machine? If not, what's the IP?

For some reason, the following error appears: 
[ERROR:flutter/lib/ui/ui_dart_state.cc(209)] Unhandled Exception: ConnectionFailedException: Could not connect to 10.0.2.2:5672 after 1 attempts. Giving up

ardera, 2022-03-30:
Flutter-pi doesn't do anything related to that, it just supplies flutter with input events and takes the rendered frames and presents them. So whatever port / IP you need to use has nothing to do with flutter-pi.
Though usually `127.0.0.1` is localhost, so maybe try that?

alexmac98, 2022-03-31:
Thank you! It worked. Best regards!

alexmac98, 2022-03-31:
Thank you! It worked!

closed 2022-03-31 by alexmac98