ardera/flutter-pi #243: Support for viewing PDF's

RadJKW, 2022-04-13:
My application requires video playback and pdf viewing on a raspberry pi.  

I switched to flutter-pi with the news that gstreamer is being used. My testing showed that using flutter-pi and gstreamer (video_player) works much better than dart_vlc. 

However, now I can no longer view my pdf documents using https://pub.dev/packages/printing. The package relies on pdfium to be compiled for the platform.  My current implementation results in this error `Error getting printing info: MissingPluginException(No implementation found for method printingInfo on channel net.nfet.printing)`. 

I've tried a few other pdf packages designed for android but no luck so far. Is there any solution available?

RadJKW, 2022-04-27:
@ardera Could you elaborate on this?

top-master, 2022-09-17:
"viewing"?

As said on #250:
>"**flutter-pi can only run without desktop.** Desktop support is not planned and won't be implemented. You can try using arm64 and the official linux desktop embedder, or sony's flutter-elinux, or toyotas embedder."

bojidartonchev, 2023-09-19:
@RadJKW I am currently working on an implementation for this using [ImageMagick's MagickWand C API](https://imagemagick.org/script/magick-wand.php). You can observe the progress on [my fork](https://github.com/bojidartonchev/flutter-pi/tree/feature/printing) (branch `feature/printing`).

eximius313, 2023-09-24:
@bojidartonchev, do I understand well, that with your fork it will be possible to display PDF file on the screen?

bojidartonchev, 2023-09-25:
Hi @eximius313,

I am using the pdf plugin to generate a pdf, then rasterize it and send it to thermal printer for printing. My implementation is rasterizing it as a bitmap, so you will need to decode it on flutter side.

```
await for (var page in Printing.raster(await doc.save())) {
  final image = i.decodeImage(page.pixels)!;

  bytes += generator.image(image);
}
```

The [esc_pos_utils](https://pub.dev/packages/esc_pos_utils) I am using for printing is a bit outdated and didn't work with the new [image](https://pub.dev/packages/image) plugin and I didn't have time for investigating it, so I proceed using [image_v3](https://pub.dev/packages/image_v3) instead. But if you need it only for rendering, [image](https://pub.dev/packages/image) should do the job.

eximius313, 2023-09-25:
thanks @bojidartonchev!