ardera/flutter-pi #245: Unhandled Exception: MissingPluginException(No implementation found for method create on channel plugins.flutter.io/desktop_webview_window)

marco-1988, 2022-04-21:
Hi everyone
I've been trying for a long time to get a webview to work on flutter-pi with no success partly due to lack of plugins that support this feature on Linux and partly because they don't work on flutter-pi.

Now I want to focus on the second aspect: I found a plugin called desktop webview window that works on pure Linux, opening the web page on a new window, and I would like it to open fullscreen above the existing one with the possibility of closing it later. However this plugin on flutter-pi doesn't work because it can't find the plugin (although I added it on pubspec)
This MissingPluginException problem on flutter-pi happens to me very often. In release mode it also happened to me with shared_preferences and I had to do a SharedPreferencesLinux.registerWith () to make it work, but with this plugin it didn't work.
Url_launcher also has the same problem, but I don't have a browser installed and maybe that's why. Speaking of browsers, I don't know if I could use a browser as an alternative to webview since it needs X and apparently flutter-pi is not compatible with X.
What suggestions do you give me to get out of this tangled situation?
Basically, I need to be able to open a link from the flutter application, possibly with a webview or alternatively in the browser if you really can't with the webview.
Thanks

ardera, 2022-04-21:
Theoretically it's possible to implement it but that's a lot of work. The MissingPluginException happens because the native code is not present. (A plugin == native code + dart code) The native code is not present because there is no native code the package provides for flutter-pi, only for the linux desktop embedder and so on. So you'd have to write that native code yourself, basically write your own chromium embedded framework client, pipe through all the input events, let chromium draw into an offscreen buffer, upload to a GL texture or some other GPU buffer, etc.

kukjevov, 2022-05-11:
Hmm I am fighting against similar issue. 

@ardera In readme page there is info that *shared_preferences* plugin should work, but in `--release` it does not. It fails with same exception as it is metioned above. Does this mean, that even plugins that are written for linux will not work on raspbian?

@marco-1988 You said that *I had to do a SharedPreferencesLinux.registerWith () to make it work*, what exactly that means? Where did you put that code? I was not able to enable `shared_preferences` nor `connectivity_plus`. Can you help me?

Thank you very much for any help

marco-1988, 2022-05-11:
Hi. You can add SharedPreferencesLinux.registerWith() importing shared_preferences_linux on main of app.

Regards

kukjevov, 2022-05-11:
Thank you both :) for answers :)

closed 2022-07-10 by ardera