ardera/flutter-pi #248: How to update to 3.0

kukjevov, 2022-05-17:
Hello i have been successfuly using flutter-pi with flutter 2.10, i have updated to 3.0 and i get following error during startup.

Error while initializing the Dart VM: Wrong full snapshot version, expected 'd56742caf7b3b3f4bd2df93a9bbb5503' found '1441d6b13b8623fa7fbf61433abebd31'

I had pulled latest engine binaries on development machine and called ./install.sh
I had called flutter upgrade on development machine.
I had pulled latest flutter-pi changes on raspberry os and removed build directory in flutter-pi and called following commands.

mkdir build && cd build
cmake ..
make -j`nproc`
sudo make install

What am i missing?

Thank you

pieter-scholtz, 2022-05-18:
Perhaps try the following : 
1. Make sure that your flutter SDK is checked out at 3.0.0 using `flutter doctor -v`
2. Inside the flutter SDK delete the cache folder **/flutter/bin/cache**

zoechi, 2022-06-08:
I run into the same issue.  
In my case it was because I didn't use the full path to the `dart.exe` and so the one from my Dart SDK install was executed, instead of `C:\flutter\bin\cache\dart-sdk\bin\dart.exe ...`

zoechi, 2022-06-08:
@pieter-scholtz yes, in my case it was only with the additional steps for the release build.

pieter-scholtz, 2022-06-08:
@zoechi Thanks , makes sense , when I first got the error it haunted me for a few weeks : D

zoechi, 2022-06-10:
It's on the dev machine when I try to create a release build, before rsyncing it to RPi

pieter-scholtz, 2022-06-10:
Yes exactly , the reason I suggested deleting **/flutter/bin/cache** (On the dev machine) , is because sometimes when switching flutter SDK versions using just git , the dart stuff in the cache does not get updated, if you delete it , the correct version will be downloaded automatically (I think running `flutter upgrade` is better in this regard)

kukjevov, 2022-06-13:
Well i have deleted `/flutter/bin/cache` and even removed all ignored files in flutter repo. I have upgraded flutter using `flutter upgrade` and problem is still there. When i was looking on stackoverflow for something like this, one guy had similar problem, that only full reinstall of flutter helped. That is one thing that i havent tried yet. So i hope full reinstall will help :). 

Anyway thx for replies :).

pieter-scholtz, 2022-06-13:
Before you do that , I would recommend you check that you have the right engine libraries on the pi , perhaps delete the old ones from `/usr/lib` on the pi , and then run the install for that.

kukjevov, 2022-06-21:
@ardera @pieter-scholtz  it looks like that problem was engine-binaries on rpi, now it is working thx :)