ardera/flutter-pi #254: Could not query DRM device list: No such file or directory

ardera, 2022-07-01:
flutter-pi outputs:
```
[flutter-pi] Could not query DRM device list: No such file or directory
```

and terminates.

drowsell, 2022-07-02:
Here is the info from my Pi. I was the user experiencing this error.

raspinfo
System Information
------------------

Raspberry Pi 3 Model B Rev 1.2
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 27a8050c3c06e567c794620394a8c2d74262a516, stage2

Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
Revision	: a02082
Serial		: 00000000811a5c75
Model		: Raspberry Pi 3 Model B Rev 1.2
Throttled flag  : throttled=0x50000
Camera          : supported=0 detected=0, libcamera interfaces=0

Videocore information
---------------------

Mar 24 2022 13:20:54 
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       59825532 4359740  53001044   8% /
devtmpfs          437272       0    437272   0% /dev
tmpfs             471928       0    471928   0% /dev/shm
tmpfs             188772     896    187876   1% /run
tmpfs               5120       4      5116   1% /run/lock
/dev/loop0        102784  102784         0 100% /snap/core/13315
/dev/loop1         50176   50176         0 100% /snap/core18/2406
/dev/loop2        171264  171264         0 100% /snap/flutter/124
/dev/mmcblk0p1    258095   31598    226497  13% /boot
tmpfs              94384       4     94380   1% /run/user/1000

Filename				Type		Size		Used		Priority
/var/swap                               file		102396		0		-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: (none)
raspberrypi-sys-mods:
  Installed: 20220224
openbox:
  Installed: (none)
lxpanel:
  Installed: (none)
pcmanfm:
  Installed: (none)
rpd-plym-splash:
  Installed: (none)

Networking Information
----------------------

eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether m.m.m.m  txqueuelen 1000  (Ethernet)
[... 513 lines cut ...]
[   16.652752] 8021q: 802.1Q VLAN Support v1.8
[   16.831672] Adding 102396k swap on /var/swap.  Priority:-2 extents:3 across:126976k SSFS
[   17.846335] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   18.304077] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   18.457980] smsc95xx 1-1.1:1.0 eth0: Link is Down
[   19.245398] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   19.333135] hwmon hwmon1: Undervoltage detected!
[   21.790198] Bluetooth: Core ver 2.22
[   21.791486] NET: Registered PF_BLUETOOTH protocol family
[   21.791520] Bluetooth: HCI device and connection manager initialized
[   21.791581] Bluetooth: HCI socket layer initialized
[   21.791611] Bluetooth: L2CAP socket layer initialized
[   21.791671] Bluetooth: SCO socket layer initialized
[   21.819804] Bluetooth: HCI UART driver ver 2.3
[   21.819859] Bluetooth: HCI UART protocol H4 registered
[   21.821315] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   21.828045] Bluetooth: HCI UART protocol Broadcom registered
[   22.636531] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.wlan0.retrans_time - use net.ipv6.neigh.wlan0.retrans_time_ms instead
[   23.346723] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   23.346754] Bluetooth: BNEP filters: protocol multicast
[   23.346779] Bluetooth: BNEP socket layer initialized
[   23.378790] NET: Registered PF_ALG protocol family
[   23.562208] loop3: detected capacity change from 0 to 8
[   25.381009] hwmon hwmon1: Voltage normalised
[   33.764975] cam-dummy-reg: disabling

ardera, 2022-07-02:
Thx, I'll compare that to what it says on my pi

What does `ls /dev/dri` output for you?

drowsell, 2022-07-02:
I looks like that folder does not exist for me:

ls: cannot access '/dev/dri': No such file or directory

ardera, 2022-07-03:
what display are you using?

can you send me the contents of your `/boot/config.txt`?

drowsell, 2022-07-03:
I am just using a regular HDMI monitor. I am waiting for a touchscreen to ship. Will this not even load without a touch screen? That could be the issue.

cat /boot/config.txt 
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
#dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
gpu_mem=64

ardera, 2022-07-04:
It seems like you have the graphics driver disabled:

```
...
#dtoverlay=vc4-kms-v3d
...
```

if you remove the `#` at the start, it should work.

> I am waiting for a touchscreen to ship. Will this not even load without a touch screen?

Yep it will, that shouldn't be a problem.

drowsell, 2022-07-04:
Thank-you so much! This is now working for me.

closed 2022-07-10 by ardera