ardera/flutter-pi #261: Dart Error: Can't load Kernel binary: Invalid SDK hash.

moritzxbe, 2022-07-07:
I am facing issues on a fresh installed raspberry pi with bullseye.

I am using flutter-pi e306d1b9ce8766c82f2415b1f4f7ae6ae91f90e3 and engine-binaries 6b09dd7ef4981ca52eec86fc78f47cc050fd330b and locally I have bundled it locally using the following flutter version
Flutter 3.0.4 • channel stable • https://github.com/flutter/flutter.git
Framework • revision 85684f9300 (7 days ago) • 2022-06-30 13:22:47 -0700
Engine • revision 6ba2af10bb
Tools • Dart 2.17.5 • DevTools 2.12.2

I am getting the following error:
[ERROR:flutter/shell/common/shell.cc(93)] Dart Error: Can't load Kernel binary: Invalid SDK hash.
[ERROR:flutter/runtime/dart_isolate.cc(143)] Could not prepare isolate.
[ERROR:flutter/runtime/runtime_controller.cc(382)] Could not create root isolate.
[ERROR:flutter/shell/common/shell.cc(600)] Could not launch engine with configuration.

Any idea how to fix that issue?

ardera, 2022-07-07:
What does `sha256sum /usr/lib/libflutter_engine.so.debug` say for you on the pi?

moritzxbe, 2022-07-08:
This is the output:
061bb7bc5e9eaa32a3ccdbfb61981831dde262682db2316646651fdc9bed7da4  /usr/lib/libflutter_engine.so.debug

ardera, 2022-07-08:
For me it's 
```console
pi@hpi4:~/devel/flutter-engine-binaries-for-arm $ sha256sum ./arm/libflutter_engine.so.debug
23e90022efaca95ec4b58eeb8591b64486cf7f6026fd5292fd067370377c379c  ./arm/libflutter_engine.so.debug
```

Try installing the engine binaries again

moritzxbe, 2022-07-11:
after reinstalling engine binaries again everything is working fine. problem might be related to Ansible deployment of flutter-pi and engine-binaries. Thank you 👍

closed 2022-07-11 by moritzxbe