ardera/flutter-pi #262: undefined reference to `eglDestroyImage'

bojidartonchev, 2022-07-11:
Hello, I am trying to build an Yocto image from the kirkstone branch using the [meta-flutter](https://github.com/meta-flutter/meta-flutter) layer, but flutter-pi is not compiling.

```
| /mnt/data/Projects/tinkerboard/build/tmp/work/armv8a-poky-linux/flutter-pi-release/git-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-ld: CMakeFiles/flutter-pi.dir/src/compositor.c.o: in function `destroy_drm_rbo':
| compositor.c:(.text+0x2b54): undefined reference to `eglDestroyImage'
| clang-14: error: linker command failed with exit code 1 (use -v to see invocation)
| ninja: build stopped: subcommand failed.
| WARNING: exit code 1 from a shell command.
ERROR: Task (/mnt/data/Projects/tinkerboard/meta-flutter/recipes-graphics/flutter-pi/flutter-pi-release_git.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1873 tasks of which 1872 didn't need to be rerun and 1 failed.
```

This is my local.conf

```
MACHINE ??= "rockchip-rk3399-sapphire-excavator"
CORE_IMAGE_EXTRA_INSTALL += "kernel-modules tzdata flutter-gallery-release flutter-pi-release"
IMAGE_FSTYPES = "wic.gz"
DISTRO_FEATURES += " systemd opengl x11"
DISTRO_FEATURES_BACKFILL_CONSIDERED += "sysvinit"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_initscripts = "systemd-compat-units"
GLIBC_GENERATE_LOCALES = "en_US.UTF-8"
IMAGE_LINGUAS = "en-us"
FLUTTER_RUNTIME_pn-flutter-drm-gbm-backend = "release"
LICENSE_FLAGS_ACCEPTED = "commercial"
FLUTTER_SDK_TAG = "3.0.1"
```

I don't know if you are familiar with the meta-flutter project, but if you have any hints on how to fix this, it would be helpful.
Best regards!

ardera, 2022-07-12:
I think your driver just doesn't support EGL 1.5 probably.

That's not a problem though, `eglDestroyImage` is available as `eglDestroyImageKHR` in earlier versions of EGL, and even if it's not, that function is only made use of by flutter-pi when using `omxplayer_video_player`.

I'll fix it

bojidartonchev, 2022-07-22:
The issue was resolved after changing from Jeffy's [meta-rockchip](https://github.com/JeffyCN/meta-rockchip) to mainline [meta-rockchip](https://git.yoctoproject.org/meta-rockchip). You can close this issue now or when you implement the fix you mentioned above.

closed 2023-08-22 by ardera