ardera/flutter-pi #263: Capture screenshot

mortenboye, 2022-07-16:
Is it possible to capture a screenshot of a running Flutter-pi app?

I've tried with [fbgrab](https://github.com/GunnarMonell/fbgrab) but I can only get it to capture an empty terminal prompt, not the app.

Any ideas?

mortenboye, 2022-07-17:
I ended up doing it within my app, using https://pub.dev/packages/screenshot

Closing

closed 2022-07-17 by mortenboye

SimonJFisherMB, 2023-03-02:
This is something I would like to do also. I have a full (800x440) screen app running on flutter_pi and would like to be able to capture a screenshot to file in order to remotely see what's on the screen for support purposes. I've tried a DRM screenshot (C) app but just returns a completely black bitmap.