ardera/flutter-pi #264: Examble on Readme

leoheck, 2022-07-17:
Hi, I am experimenting your setup, trying the example of the gallery that is on Readme.

This one
[image]

It seems it is not working. Since I don't know flutter, I cannot easily fix it, yet. Could you give me an idea on how to solve this?

```
➜  flutter_gallery git:(5da082d) git checkout 5da082d

HEAD is now at 5da082d null safety : Migrated standalone files to null safety (#586)
➜  flutter_gallery git:(5da082d) flutter build bundle

Running "flutter pub get" in flutter_gallery...                  2,619ms

💪 Building with sound null safety 💪

lib/feature_discovery/feature_discovery.dart:224:26: Warning: Operand of null-aware operation '?.' has
type 'SchedulerBinding' which excludes null.
 - 'SchedulerBinding' is from 'package:flutter/src/scheduler/binding.dart'
 ('../../../snap/flutter/common/flutter/packages/flutter/lib/src/scheduler/binding.dart').
        SchedulerBinding.instance?.addPostFrameCallback((_) {
                         ^
lib/feature_discovery/feature_discovery.dart:240:28: Warning: Operand of null-aware operation '?.' has
type 'SchedulerBinding' which excludes null.
 - 'SchedulerBinding' is from 'package:flutter/src/scheduler/binding.dart'
 ('../../../snap/flutter/common/flutter/packages/flutter/lib/src/scheduler/binding.dart').
          SchedulerBinding.instance?.addPostFrameCallback((_) {
                           ^
../../../snap/flutter/common/flutter/.pub-cache/hosted/pub.dartlang.org/animations-2.0.2/lib/src/open_cont
ainer.dart:631:24: Warning: Operand of null-aware operation '!' has type 'SchedulerBinding' which excludes
null.
 - 'SchedulerBinding' is from 'package:flutter/src/scheduler/binding.dart'
 ('../../../snap/flutter/common/flutter/packages/flutter/lib/src/scheduler/binding.dart').
      SchedulerBinding.instance!
                       ^
../../../snap/flutter/common/flutter/.pub-cache/hosted/pub.dartlang.org/animations-2.0.2/lib/src/open_cont
ainer.dart:665:24: Warning: Operand of null-aware operation '!' has type 'SchedulerBinding' which excludes
null.
 - 'SchedulerBinding' is from 'package:flutter/src/scheduler/binding.dart'
 ('../../../snap/flutter/common/flutter/packages/flutter/lib/src/scheduler/binding.dart').
      SchedulerBinding.instance!
                       ^
../../../snap/flutter/common/flutter/.pub-cache/hosted/pub.dartlang.org/win32-2.3.3/lib/src/structs.g.dart
:661:31: Error: Member not found: 'UnicodeChar'.
  int get UnicodeChar => Char.UnicodeChar;
                              ^^^^^^^^^^^
../../../snap/flutter/common/flutter/.pub-cache/hosted/pub.dartlang.org/win32-2.3.3/lib/src/structs.g.dart
:662:38: Error: Setter not found: 'UnicodeChar'.
  set UnicodeChar(int value) => Char.UnicodeChar = value;
                                     ^^^^^^^^^^^
../../../snap/flutter/common/flutter/.pub-cache/hosted/pub.dartlang.org/win32-2.3.3/lib/src/structs.g.dart
:664:29: Error: Member not found: 'AsciiChar'.
  int get AsciiChar => Char.AsciiChar;
                            ^^^^^^^^^
../../../snap/flutter/common/flutter/.pub-cache/hosted/pub.dartlang.org/win32-2.3.3/lib/src/structs.g.dart
:665:36: Error: Setter not found: 'AsciiChar'.
  set AsciiChar(int value) => Char.AsciiChar = value;
                                   ^^^^^^^^^
Target kernel_snapshot failed: Exception
Failed to build bundle.
➜  flutter_gallery git:(5d```

leoheck, 2022-07-17:
Oh, `flutter pub upgrade` has solved the Errors, it was left with 2 warnings only. 

```
➜  flutter_gallery git:(5da082d) ✗ flutter build bundle                                 

💪 Building with sound null safety 💪

lib/feature_discovery/feature_discovery.dart:224:26: Warning: Operand of null-aware operation '?.' has
type 'SchedulerBinding' which excludes null.
 - 'SchedulerBinding' is from 'package:flutter/src/scheduler/binding.dart'
 ('../../../snap/flutter/common/flutter/packages/flutter/lib/src/scheduler/binding.dart').
        SchedulerBinding.instance?.addPostFrameCallback((_) {
                         ^
lib/feature_discovery/feature_discovery.dart:240:28: Warning: Operand of null-aware operation '?.' has
type 'SchedulerBinding' which excludes null.
 - 'SchedulerBinding' is from 'package:flutter/src/scheduler/binding.dart'
 ('../../../snap/flutter/common/flutter/packages/flutter/lib/src/scheduler/binding.dart').
          SchedulerBinding.instance?.addPostFrameCallback((_) {
                           ^

```

leoheck, 2022-07-17:
Not quite there yet.

```
flutter_galerry  flutter-pi
➜  pi@drivethru  ~ flutter-pi /home/pi/flutter_galerry 
[flutter-pi] WARNING: display didn't provide valid physical dimensions. The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
==============Locale==============
Flutter locale:
  default: en_GB
  locales: en_GB en.UTF-8 en.UTF-8 en.UTF-8 en_GB en en.UTF-8
===================================
===================================
display mode:
  resolution: 1920 x 1080
  refresh rate: 60Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_device EGL_EXT_platform_wayland EGL_KHR_platform_wayland EGL_EXT_platform_x11 EGL_KHR_platform_x11 EGL_MESA_platform_gbm EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless"
  display extensions: "EGL_ANDROID_blob_cache EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 2.0 Mesa 20.3.5"
  shading language version: "OpenGL ES GLSL ES 1.0.16"
  vendor: "Broadcom"
  renderer: "VC4 V3D 2.1"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_EGL_image GL_OES_depth_texture GL_AMD_performance_monitor GL_OES_packed_depth_stencil GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_pack_reverse_row_order GL_EXT_occlusion_query_boolean GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_NV_pixel_buffer_object GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_separate_shader_objects GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_KHR_parallel_shader_compile GL_MESA_tile_raster_order "
===================================
detected VideoCore IV as underlying graphics chip, and VC4 as the driver.
Reporting modified GL_EXTENSIONS string that doesn't contain non-working extensions.
[ERROR:flutter/shell/common/shell.cc(93)] Dart Error: Can't load Kernel binary: Invalid SDK hash.
[ERROR:flutter/runtime/dart_isolate.cc(143)] Could not prepare isolate.
[ERROR:flutter/runtime/runtime_controller.cc(382)] Could not create root isolate.
[ERROR:flutter/shell/common/shell.cc(600)] Could not launch engine with configuration.
flutter: The Dart VM service is listening on http://127.0.0.1:40661/CPFVLzmYI2s=/
```

leoheck, 2022-07-17:
Actually, this is the important part.
```
[ERROR:flutter/shell/common/shell.cc(93)] Dart Error: Can't load Kernel binary: Invalid SDK hash.
[ERROR:flutter/runtime/dart_isolate.cc(143)] Could not prepare isolate.
[ERROR:flutter/runtime/runtime_controller.cc(382)] Could not create root isolate.
[ERROR:flutter/shell/common/shell.cc(600)] Could not launch engine with configuration.
```

How do I make sure which version of things I have? Or better, how do I fix it? I am new new to this flutter-world

leoheck, 2022-07-17:
So, on my computer (Ubuntu 22.04) I have this
```
➜  flutter_gallery git:(5da082d) ✗ flutter --version
Flutter 3.0.5 • channel stable • https://github.com/flutter/flutter.git
Framework • revision f1875d570e (3 days ago) • 2022-07-13 11:24:16 -0700
Engine • revision e85ea0e79c
Tools • Dart 2.17.6 • DevTools 2.12.2
```

leoheck, 2022-07-17:
I am using the latest engine binaries
```bash
➜  ~/flutter-engine-binaries-for-arm git:(main) git log
commit 6b09dd7ef4981ca52eec86fc78f47cc050fd330b (HEAD -> main, tag: engine_6ba2af10bb05c88a2731482cedf2cfd11cf5af0b, origin/main, origin/HEAD)
Author: Hannes Winkler <hanneswinkler2000@web.de>
Date:   Wed Jul 6 14:29:24 2022 +0200

    update binaries for flutter 3.0.4
```

It looks like they are not the same version as my flutter which is 3.0.5

leoheck, 2022-07-17:
Nice, could clone the flutter repo, select the tag 3.0.4 and use it. It worked!

ardera, 2022-07-18:
Yeah, just uploaded the 3.0.5 engine binaries. Should really setup CI for this

leoheck, 2022-07-18:
Ah, indeed. That would be awesome.

leoheck, 2022-07-18:
By the way, this flutter-pi is pretty awesome. Thanks for making this project and for making it available.