ardera/flutter-pi #268: Suport for mali GPU Tv Box Amlogic S905X

emboxs, 2022-07-24:
Is there a possibility for flutter-pi to support mali gpu, like the one on android tv box. I have tried it with armbian os but always get error

```
[modesetting] Could not commit atomic request. drmModeAtomicCommit: Permission denied
[compositor] Could not present frame. drmModeAtomicCommit: Permission denied
```

ardera, 2022-07-25:
Does it work if you run flutter-pi with `sudo`? (Wouldn't recommend doing that in a production environment though)
If it doesn't work is something else using the display maybe?

emboxs, 2022-07-25:
after running it with sudo I get an error like this:

`[flutter-pi] Warning: Could not load libflutter_engine.so from the asset bundle. dlopen: /home/station01/minipom/libflutter_engine.so: cannot open shared object file: No such file or directory. Trying to open libflutter_engine.so.debug...
flutter: The Dart VM service is listening on http://127.0.0.1:44151/K-DsTSFykPo=/
[compositor] GPU does not support reflecting the screen in Y-direction.
  This is required for rendering into hardware overlay planes though.
  Any UI that is drawn in overlay planes will look upside down.
[compositor] GPU does not supported the desired HW plane order.
  Some UI layers may be invisible.
[modesetting] Could not commit atomic request. drmModeAtomicCommit: Invalid argument
[compositor] Could not present frame. drmModeAtomicCommit: Invalid argument
[modesetting] Could not commit atomic request. drmModeAtomicCommit: Invalid argument
[compositor] Could not present frame. drmModeAtomicCommit: Invalid argument
`
any suggestions, what should i do?

ardera, 2022-08-01:
Can you try running kmscube and see if it works?

closed 2022-11-22 by github-actions[bot]