ardera/flutter-pi #272: [Unclear documentation] libflutter_engine.so: cannot open shared object file: No such file or directory

eximius313, 2022-07-28:
I've followed your guideline from [README](https://github.com/ardera/flutter-pi#building-the-appso-for-running-your-app-in-releaseprofile-mode) step by step and app was built just fine, but when I run it with `flutter-pi --release` I get:
```
/home/user/flutter_assets/libflutter_engine.so: cannot open shared object file: No such file or directory. Trying to open libflutter_engine.so.release...
```
Could you please clarify in docs where I can get `libflutter_engine.so.release` and where to put it?

Kind regards

ardera, 2022-07-28:
> ```
> /home/user/flutter_assets/libflutter_engine.so: cannot open shared object file: No such file or directory. Trying to open libflutter_engine.so.release...
> ```
That's not an error. There's multiple locations where the flutter engine can be located and this is just the first one that was tried.
Does the app run fine after that?

> Could you please clarify in docs where I can get `libflutter_engine.so.release` and where to put it?

The flutter engine will be installed in the first step of the [dependencies section](https://github.com/ardera/flutter-pi#dependencies)

eximius313, 2022-07-28:
> That's not an error. There's multiple locations where the flutter engine can be located and this is just the first one that was tried. Does the app run fine after that?

nope :(

> The flutter engine will be installed in the first step of the [dependencies section](https://github.com/ardera/flutter-pi#dependencies)

I have dependencies already installed and `flutter-pi` without `--release` works just fine

pieter-scholtz, 2022-07-29:
> nope :(
>
>
> I have dependencies already installed and `flutter-pi` without `--release` works just fine

Can you send a more complete error log ?

eximius313, 2022-07-29:
Sure

```
[locales] Warning: The system has no configured locale. The default "C" locale may or may not be supported by the app.
[flutter-pi] WARNING: display didn't provide valid physical dimensions. The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
==============Locale==============
Flutter locale:
  default: C
  locales: C
===================================
===================================
display mode:
  resolution: 1920 x 1080
  refresh rate: 60Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
===================================
error: XDG_RUNTIME_DIR not set in the environment.
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_device EGL_EXT_platform_wayland EGL_KHR_platform_wayland EGL_EXT_platform_x11 EGL_KHR_platform_x11 EGL_MESA_platform_gbm EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 3.1 Mesa 20.3.5"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_filter_anisotropic GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_draw_instanced GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_pack_reverse_row_order GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_NV_pixel_buffer_object GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_primitive_bounding_box GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_view GL_KHR_context_flush_control GL_NV_image_formats GL_OES_draw_elements_base_vertex GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_OES_texture_view GL_EXT_buffer_storage GL_EXT_float_blend GL_EXT_geometry_point_size GL_EXT_geometry_shader GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_geometry_point_size GL_OES_geometry_shader GL_OES_shader_image_atomic GL_EXT_texture_compression_s3tc_srgb GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_EXT_EGL_image_storage GL_MESA_framebuffer_flip_y GL_EXT_texture_query_lod "
===================================
[flutter-pi] Warning: Could not load libflutter_engine.so from the asset bundle. dlopen: /home/user/flutter_assets/libflutter_engine.so: cannot open shared object file: No such file or directory. Trying to open libflutter_engine.so.release...
[ERROR:flutter/lib/ui/ui_dart_state.cc(198)] Unhandled Exception: MissingPluginException(No implementation found for method getAll on channel dev.fluttercommunity.plus/package_info)
#0      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:165)
<asynchronous suspension>
#1      MethodChannel.invokeMapMethod (package:flutter/src/services/platform_channel.dart:367)
<asynchronous suspension>
#2      MethodChannelPackageInfo.getAll (package:package_info_plus_platform_interface/method_channel_package_info.dart:13)
```

pieter-scholtz, 2022-07-29:
```
[ERROR:flutter/lib/ui/ui_dart_state.cc(198)] Unhandled Exception: MissingPluginException(No implementation found for method getAll on channel dev.fluttercommunity.plus/package_info)
#0      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:165)
<asynchronous suspension>
#1      MethodChannel.invokeMapMethod (package:flutter/src/services/platform_channel.dart:367)
<asynchronous suspension>
#2      MethodChannelPackageInfo.getAll (package:package_info_plus_platform_interface/method_channel_package_info.dart:13)```
```
^^^ This is the error , the previous message is just a warning.

pieter-scholtz, 2022-07-29:
What does your debug log look like ?

eximius313, 2022-07-29:
```
[locales] Warning: The system has no configured locale. The default "C" locale may or may not be supported by the app.
[flutter-pi] WARNING: display didn't provide valid physical dimensions. The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
==============Locale==============
Flutter locale:
  default: C
  locales: C
===================================
===================================
display mode:
  resolution: 1920 x 1080
  refresh rate: 60Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
===================================
error: XDG_RUNTIME_DIR not set in the environment.
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_device EGL_EXT_platform_wayland EGL_KHR_platform_wayland EGL_EXT_platform_x11 EGL_KHR_platform_x11 EGL_MESA_platform_gbm EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 3.1 Mesa 20.3.5"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_filter_anisotropic GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_draw_instanced GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_pack_reverse_row_order GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_NV_pixel_buffer_object GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_primitive_bounding_box GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_view GL_KHR_context_flush_control GL_NV_image_formats GL_OES_draw_elements_base_vertex GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_OES_texture_view GL_EXT_buffer_storage GL_EXT_float_blend GL_EXT_geometry_point_size GL_EXT_geometry_shader GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_geometry_point_size GL_OES_geometry_shader GL_OES_shader_image_atomic GL_EXT_texture_compression_s3tc_srgb GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_EXT_EGL_image_storage GL_MESA_framebuffer_flip_y GL_EXT_texture_query_lod "
===================================
[flutter-pi] Warning: Could not load libflutter_engine.so from the asset bundle. dlopen: /home/user/flutter_assets/libflutter_engine.so: cannot open shared object file: No such file or directory. Trying to open libflutter_engine.so.debug...
flutter: The Dart VM service is listening on http://127.0.0.1:45385/yl0jcYDmzP4=/
```

eximius313, 2022-08-18:
Thank you.
The question is why it throws exception in "release" mode?

closed 2022-08-18 by eximius313

ardera, 2022-08-18:
Somehow in debug mode it recognizes it runs on linux and sets [`PackageInfoPlatform.instance` to linux](https://github.com/fluttercommunity/plus_plugins/blob/b9951b7e99d7ffc495a886def9154eef5cb1ab9d/packages/package_info_plus/package_info_plus_linux/lib/src/package_info.dart#L12), but in release mode it doesn't and it defaults to [`MethodChannelPackageInfo`](https://github.com/fluttercommunity/plus_plugins/blob/b9951b7e99d7ffc495a886def9154eef5cb1ab9d/packages/package_info_plus/package_info_plus_platform_interface/lib/method_channel_package_info.dart#L12) and you'll get the `MissingPluginException`.

Probably the dart plugin registrant isn't running in release mode somehow, though not sure why. But there's some other unsupported plugins that show similar behaviour: https://github.com/ardera/flutter-pi/issues/222

(If you do `flutter build linux` and then run `flutter build bundle --target=.dart_tool/flutter_build/generated_main.dart`, it'll probably run without the exception again)

eximius313, 2022-08-18:
thanks so much @ardera and @pieter-scholtz

eximius313, 2022-08-18:
@ardera and about:
> Probably the dart plugin registrant isn't running in release mode somehow

 is it a bug in Flutter itself?

pieter-scholtz, 2022-08-19:
Have you tried handling the exception ?

eximius313, 2022-08-19:
I want to use the `package_info` plugin, so handling the exception will not make me a difference ;)

ardera, 2022-08-19:
> @ardera and about:
> 
> > Probably the dart plugin registrant isn't running in release mode somehow
> 
> is it a bug in Flutter itself?

Not really, officially flutter only supports: android, ios, the official linux desktop embedder, the official windows desktop embedder, macos and so on. Flutter-pi is not officially supported. So it's surprising the dart plugin registrant is running in debug mode even. (But again, even if it ran in release mode, that doesn't change the fact that the plugin is not supported)

Though that doesn't stop you from writing your own `package_info` -like functionality of course