ardera/flutter-pi #274: Could not find "XKBMODEL" property inside "/etc/default/keyboard"

DanPKR, 2022-08-12:
Hi im trying to run the counter demo on a Raspberry 4 using raspbian but this message appears

`[keyboard] Could not find "XKBMODEL" property inside "/etc/default/keyboard". Default value will be used.[keyboard] Could not find "XKBLAYOUT" property inside "/etc/default/keyboard". Default value will be used.[keyboard] Could not find "XKBVARIANT" property inside "/etc/default/keyboard". Default value will be used.[keyboard] Could not find "XKBOPTIONS" property inside "/etc/default/keyboard". Default value will be used.[flutter-pi] Could not query DRM device list: No existe el fichero o el directorio
bash: [keyboard]: orden no encontrada`

you can find attached the build folder
[build.zip](https://github.com/ardera/flutter-pi/files/9313499/build.zip)

[Uploading build.zip…]()

ardera, 2022-08-19:
Can you try step 3 of the setup instructions? So:

3. _You can skip this if you're on Raspberry Pi 4 with Raspbian Bullseye_
Enable the V3D graphics driver:
`Advanced Options -> GL Driver -> GL (Fake KMS)`

Maybe something changed and you can no longer skip it.

ardera, 2022-09-19:
Closing this for now. Feel free to reopen if the issue persists.

closed 2022-09-19 by ardera