ardera/flutter-pi #279: install location of icudtl.dat

jwinarske, 2022-08-24:
`icudtl.dat` is not a system library, is generally unique to an application, and really shouldn't be installed in `/usr/lib/`.

In meta-flutter I install icudtl.dat to /usr/share/flutter.  This is shared between users, and specific to flutter.

DoumanAsh, 2022-08-25:
With this [commit](https://github.com/ardera/flutter-pi/commit/cf57b52a5848439dd4a53b657cbcb06706340ee9) you can put it inside bundle together with engine so it no longer requires system installation and allows to tie it to application.

I'm not sure it is wise to do anything else as this file probably compatible only with particular flutter versions

ardera, 2022-08-25:
@DoumanAsh true, I think bundling it with the app is even better, but since most people do install engine system-wide, we might as well install the icudtl system-wide as well.

@jwinarske Loading it from `/usr/share/flutter` is fine. I didn't really think about it too much when I put it inside `/usr/lib`. I'll prepare a PR and change the engine-binaries installer so it installs to `/usr/share/flutter`.

ardera, 2022-08-25:
Makes sense, my plan was to make it look inside the app dir first, then inside `/usr/share/flutter` and then inside `/usr/lib`.

ardera, 2023-08-18:
fixed by https://github.com/ardera/flutter-pi/pull/290

closed 2023-08-18 by ardera