ardera/flutter-pi #282: Rotation argument has no effect(vertical display, horizontal digitizer)

Taha-Firoz, 2022-08-28:
I'm in a peculiar position where I have a display which is vertical on it's main axis and the digitizer, i suspect, is landscape/horizontal
```
===================================
display mode:
  resolution: 800 x 1280
  refresh rate: 60Hz
  physical size: 107mm x 170mm
  flutter device pixel ratio: 1.967536
===================================
```

I've run the flutter gallery application with `-r 270`, `-r 90`, `-o landscape_left`, `-o landscape_right` and this is what it looks like regardless of any `-r` I pass. I'm sure it's a clash between the orientations of the display and touchscreen, however flutter-pi doesn't react to any different rotations I provide. Any idea how to fix 😭

[image]

https://user-images.githubusercontent.com/42237761/187097614-c3bb87e9-e99b-46d2-9fc7-3c0029a262f7.mp4

ardera, 2022-08-28:
The rotation/orientation argument rotates both the graphics and the touch input, so you can't use it to fix a misaligned touchscreen.

However there's another way to fix it: https://github.com/ardera/flutter-pi/issues/164#issuecomment-841879291

And I wrote an app to fix it as well: https://github.com/ardera/flutter_libinput_calibrator

Taha-Firoz, 2022-08-28:
Okay I'll look into the calibrator but the display still doesn't rotate even though I passed in the rotate argument with 90/270

ardera, 2022-08-28:
Ah, okay; that's a bug then yeah
I'll fix it

Taha-Firoz, 2022-08-29:
So won't the touchscreen still have problems if I add the rotate option even after I've calibrated it because all the libinput events would have the rotation applied to them or does that function differently? Also does the rotation option force the app into landscape or do we still have to do the set chrome orientation preference thing.

Taha-Firoz, 2022-08-29:
Okay no I understand now, I'll set the udev rotation matrix so when I start the app with the rotate option the libinput events will already be rotated before they reach flutter-pi.

ardera, 2022-08-29:
> Okay no I understand now, I'll set the udev rotation matrix so when I start the app with the rotate option the libinput events will already be rotated before they reach flutter-pi.

Exactly, yeah flutter-pi gets its input events from libinput and libinput will rotate them internally when it sees the udev property.

Taha-Firoz, 2022-08-31:
Okay Thank you so much! I got the touch working but now I just need it to rotate. Do I need to set the rotation preference thing inside of the application itself or can flutter-pi force it to rotate.

ardera, 2022-08-31:
you can set it inside the application ([`SystemChrome.setPreferredOrientations`](https://api.flutter.dev/flutter/services/SystemChrome/setPreferredOrientations.html)) but the flutter-pi rotation / orientation option can also set a default which is respected 99% of the time (if it wasn't bugged that is). Or you can use `RotatedBox` of course

Taha-Firoz, 2022-09-01:
Are you sure it's bugged, is there anyway I can force it on flutter gallery?

closed 2023-08-22 by ardera