ardera/flutter-pi #284: Flutter-pi is not picking Locale from the system

eximius313, 2022-09-03:


ardera, 2022-09-04:
flutter-pi gets its locale info from environment variables:
https://github.com/ardera/flutter-pi/blob/master/src/locales.c#L32-L56

First it checks `LANGUAGE`, then `LC_ALL`, then `LC_MESSAGES` and then `LANG`. If none of those are set it defaults to `C`.

Just putting it into `/etc/default/locale` might not be enough to make the system pickup the language. If you're on a pi, you can use `raspi-config` to configure the locale or otherwise you can use `localectl` and then `update-locale`.

eximius313, 2022-09-04:
Both `raspi-config` and `localectl` are already set:
```
# localectl
System Locale: LANG=en_GB.UTF-8
VC Keymap: n/a
X11 Layout: us
X11 Model: pc105
```

ardera, 2022-09-05:
Can you send me your cmdline output when you run `export`? Maybe something else mistakenly sets `LANGUAGE=C.UTF-8` and that overrides the `LANG` environment variable.

eximius313, 2022-09-08:
Ok, this is weird..
with reqular user, I get:
```
$ localectl
   System Locale: LANG=en_GB.UTF-8
       VC Keymap: n/a
      X11 Layout: us
       X11 Model: pc105
```

but:
```
$ echo "<$LANG>"
<>
```

with sudo, I get:
```
$ sudo echo "<$LANG>"
<>
```

but as a root I get:
```
# echo "<$LANG>"
<en_GB.UTF-8>
```

@DoumanAsh, when I run app like this:
```
$ LANGUAGE=LANG=en_GB.UTF-8 flutter-pi project
==============Locale==============
Flutter locale:
  default: en_GB
  locales: en_GB en.UTF-8 en.UTF-8 en.UTF-8 en_GB en en.UTF-8
===================================
```

Also
```
$ LANG=en_GB.UTF-8 flutter-pi project
==============Locale==============
Flutter locale:
  default: en_GB
  locales: en_GB en.UTF-8 en.UTF-8 en.UTF-8 en_GB en en.UTF-8
```

@ardera,
`export` output for `root` is indeed different than for my user (it has `LANG` set, while my **user does not**)
I have tried `sudo locale-gen "en_GB.UTF-8"` but without success...

It seems it's not the problem of `flutter-pi` but setting proper locale in user's shell.
Thanks for helping with pinpointing the issue

eximius313, 2022-09-08:
Ok, the problem is solved by enabling PAM....
https://askubuntu.com/a/749442

ardera, 2022-09-08:
Oh lol
glad you could fix it though