ardera/flutter-pi #285: Video player throws PlatformException

erdemduman, 2022-09-06:
When I run the application with Flutter-pi, [video_player](https://pub.dev/packages/video_player) throws this exception:

Unhandled Exception: PlatformException(channel-error, Unable to establish connection on channel., null, null)
VideoPlayerApi.create(package:video_player_platform_interface/messages.g.dart:156:7)
asynchronous suspension

MethodChannelVideoPlayer.create(package:video_player_platform_interface/method_channel_video_player.dart:53:37)
VideoPlayerController.initialize(package:video_player/video_player.dart:363:19)
asynchronous suspension

I am using the [example application](https://pub.dev/packages/video_player/example) that was given on the page of [video_player](https://pub.dev/packages/video_player) package.

Has anybody faced this issue before?

adicomdotir, 2022-09-28:
@ardera 
I installed all dependencies in raspberry but i still have the problem, i use flutter 2.8
Do I have to build flutterpi again?

ardera, 2022-11-01:
Yes, you have to rebuild flutter-pi after you've installed the dependencies, and it's better to reconfigure too. (Remove the build directory and re-run cmake).

closed 2022-11-22 by github-actions[bot]