ardera/flutter-pi #286: Running 2 flutter-pi

vigno88, 2022-09-06:
Hi,

I have modified `flutter-pi.c` so I could run 2 flutter-pi on 2 screens on my pi 4, but even after specifying 2 different output. I still have DRM related errors.

Is it a big change to allows 2 instances of flutter-pi to run at the same time?

Thanks a lot

ardera, 2022-09-19:
There can only be one process at a time that has control over display output. (called the DRM master) So just running 2 flutter-pi's won't work.

However, there's something called DRM leases that could help. With a DRM lease, you can give another process permissions to use a subset of the KMS resources directly, so you can effectively have multiple DRM masters (each one controlling some subset of the total KMS resources)

This could help: https://gerrit.automotivelinux.org/gerrit/gitweb?p=src/drm-lease-manager.git;a=blob;f=README.md

ardera, 2022-11-01:
Closing this now since it seems you no longer have the issue.

closed 2022-11-01 by ardera