ardera/flutter-pi #291: Touchscreen PenMount driver & calibration problem

pieroiezzi, 2022-09-29:
Hi I don't know  if this is the right place but I'm facing a problem.

I'm using a PenMount touchscreen, with standard drivers it works but seems like the touch area corresponds only a little smaller area of the interface. This happens non windows manager and on flutter app.

Anyway installing official drivers solves the problem on windows manager but not on flutter app. That's should be because drivers are for X11 and flutter-pi is unrelated to that.

Is there a way to use X11 driver and/or calibration or to replicate the same behavior?
I also tried few calibration tool but nothing seems to work.

Thanks

ardera, 2022-09-29:
Can you try running this on your pi: https://github.com/ardera/flutter_libinput_calibrator
It should help you calibrate the touch screen.

pieroiezzi, 2022-10-10:
Hi and thanks for the reply.
Sorry for the late response but i was able to make some tests just  today.

Building the application and launching it I was getting this error:
```
[ERROR:flutter/lib/ui/ui_dart_state.cc(198)] Unhandled Exception: Binding has not yet been initialized.
The "instance" getter on the WidgetsBinding binding mixin is only available once that binding has been initialized.
Typically, this is done by calling "WidgetsFlutterBinding.ensureInitialized()" or "runApp()" (the latter calls the former). Typically this call is done in the "void main()" method. The "ensureInitialized" method is idempotent; calling it multiple times is not harmful. After calling that method, the "instance" getter will return the binding.
In a test, one can call "TestWidgetsFlutterBinding.ensureInitialized()" as the first line in the test's "main()" method to initialize the binding.
If WidgetsBinding is a custom binding mixin, there must also be a custom binding class, like WidgetsFlutterBinding, but that mixes in the selected binding, and that is the class that must be constructed before using the "instance" getter.
#0      BindingBase.checkInstance.<anonymous closure> (package:flutter/src/foundation/binding.dart:281:9)
#1      BindingBase.checkInstance (package:flutter/src/foundation/binding.dart:363:6)
#2      WidgetsBinding.instance (package:flutter/src/widgets/binding.dart:306:53)
#3      MyWidgetsFlutterBinding.ensureInitialized (package:flutter_libinput_calibrator/widgets.dart:399:24)
#4      main (package:flutter_libinput_calibrator/main.dart:20:27)
#5      _runMainZoned.<anonymous closure>.<anonymous closure> (dart:ui/hooks.dart:130:25)
#6      _rootRun (dart:async/zone.dart:1426:13)
#7      _CustomZone.run (dart:async/zone.dart:1328:19)
#8      _runZoned (dart:async/zone.dart:1861:10)
#9      runZonedGuarded (dart:async/zone.dart:1849:12)
#10     _runMainZoned.<anonymous closure> (dart:ui/hooks.dart:126:5)
#11     _delayEntrypointInvocation.<anonymous closure> (dart:isolate-patch/isolate_patch.dart:297:19)
#12     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:192:12)
```

I was able to lunch the application with some changes but i get stuck on the page "Your touchscreen is now calibrated" (which is not, even for the current session).
While there i cannot proceed, cannot select a touchscreen, cannot check "Combine with current calibration" and no effect pressing "cancel" nor "Continue".

Can it be flutter version problem? This is my system status:
```
Flutter 3.0.5 • channel stable • https://github.com/flutter/flutter.git
Framework • revision f1875d570e (3 months ago) • 2022-07-13 11:24:16 -0700
Engine • revision e85ea0e79c
Tools • Dart 2.17.6 • DevTools 2.12.2
```

Thank you.
Piero

ardera, 2022-10-10:
> Building the application and launching it I was getting this error:

I think they changed `WidgetsBinding.instance` from nullable to non-nullable some time in the past, I'll fix it.

> I was able to lunch the application with some changes but i get stuck on the page "Your touchscreen is now calibrated" (which is not, even for the current session).

The `MyWidgetsFlutterBinding` is necessary for that session-wide calibration to work. (It's the thing that actually applies the calibration).

> Seems like the problem may bet the fact that my touchscreen is not being recognized as touchscreen but as a mouse. I'm investigating further.

Not sure how that would work, that's quite a difference. The type an input device has in linux (keyboard, touchscreen, mouse, tablet touchscreen with pen) is determined by a set of properties: https://www.kernel.org/doc/html/v4.12/input/event-codes.html#device-properties. You can look at what properties your device has using `evtest` for example.

pieroiezzi, 2022-10-10:
This is my USB device:
```
I: Bus=0003 Vendor=14e1 Product=6000 Version=0101
N: Name="DIALOGUE INC PenMount USB"
P: Phys=usb-fe980000.usb-1.2/input0
S: Sysfs=/devices/platform/soc/fe980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:14E1:6000.0007/input/input8
U: Uniq=
H: Handlers=mouse0 event0
B: PROP=0
B: EV=1b
B: KEY=70000 0 0 0 0 0 0 0 0
B: ABS=3
B: MSC=10
```

ardera, 2022-11-01:
@pieroiezzi Where did you get that output from? From that output, it seems like that device doesn't have `INPUT_PROP_DIRECT` set, which could be an issue

Can you run the following and send me the output?
```
$ sudo apt -y install libinput-tools
$ libinput list-devices
$ libinput debug-events
```

pieroiezzi, 2022-11-07:
Hi @ardera, sorry for the delay but I don't always have the device available.
Today i was able to run those command and these are the results:

```
 $ libinput list-devices
Device:           DIALOGUE INC PenMount USB
Kernel:           /dev/input/event0
Group:            1
Seat:             seat0, default
Capabilities:     pointer
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: disabled
Calibration:      identity matrix
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a
```

When tapping and moving I got this output:
```
-event0   DEVICE_ADDED     DIALOGUE INC PenMount USB         seat0 default group1  cap:p left scroll-nat calib
 event0   POINTER_MOTION_ABSOLUTE +119.01s       78.71/ 21.58
 event0   POINTER_BUTTON   +119.03s     BTN_LEFT (272) pressed, seat count: 1
 event0   POINTER_MOTION_ABSOLUTE +119.17s       78.81/ 21.58
 event0   POINTER_MOTION_ABSOLUTE +119.19s       78.91/ 21.58
 event0   POINTER_MOTION_ABSOLUTE +119.21s       79.10/ 21.58
 event0   POINTER_MOTION_ABSOLUTE +119.22s       79.39/ 21.58
 event0   POINTER_MOTION_ABSOLUTE +119.24s       79.59/ 21.58
 event0   POINTER_MOTION_ABSOLUTE +119.25s       79.98/ 21.48
 event0   POINTER_MOTION_ABSOLUTE +119.27s       80.27/ 21.48
 event0   POINTER_MOTION_ABSOLUTE +119.29s       80.66/ 21.39
 event0   POINTER_MOTION_ABSOLUTE +119.30s       80.96/ 21.29
 event0   POINTER_MOTION_ABSOLUTE +119.32s       81.35/ 21.09
 event0   POINTER_MOTION_ABSOLUTE +119.33s       81.74/ 20.90
 event0   POINTER_MOTION_ABSOLUTE +119.35s       82.13/ 20.70
 event0   POINTER_MOTION_ABSOLUTE +119.37s       82.62/ 20.41
 event0   POINTER_MOTION_ABSOLUTE +119.38s       83.11/ 20.12
 event0   POINTER_MOTION_ABSOLUTE +119.40s       83.50/ 19.82
 event0   POINTER_MOTION_ABSOLUTE +119.41s       83.98/ 19.43
 event0   POINTER_MOTION_ABSOLUTE +119.43s       84.38/ 19.14
 event0   POINTER_MOTION_ABSOLUTE +119.45s       84.86/ 18.75
 event0   POINTER_MOTION_ABSOLUTE +119.46s       85.16/ 18.36
 event0   POINTER_MOTION_ABSOLUTE +119.48s       85.64/ 17.87
 event0   POINTER_MOTION_ABSOLUTE +119.49s       86.04/ 17.48
 event0   POINTER_MOTION_ABSOLUTE +119.51s       86.43/ 16.99
 event0   POINTER_MOTION_ABSOLUTE +119.53s       86.82/ 16.60
 event0   POINTER_MOTION_ABSOLUTE +119.54s       87.21/ 16.11
 event0   POINTER_MOTION_ABSOLUTE +119.56s       87.50/ 15.72
 event0   POINTER_MOTION_ABSOLUTE +119.57s       87.89/ 15.23
 event0   POINTER_MOTION_ABSOLUTE +119.59s       88.09/ 14.84
 event0   POINTER_MOTION_ABSOLUTE +119.61s       88.38/ 14.45
 event0   POINTER_MOTION_ABSOLUTE +119.62s       88.77/ 14.06
 event0   POINTER_MOTION_ABSOLUTE +119.64s       88.96/ 13.67
 event0   POINTER_MOTION_ABSOLUTE +119.65s       89.26/ 13.28
 event0   POINTER_BUTTON   +119.69s     BTN_LEFT (272) released, seat count: 0
```

Is pretty clear that my thouchscreen is considered as a mouse.
I'm searching for something to do for changing that but, as always any advice is appreciated.

Thank you.

ardera, 2022-11-07:
Seems like this should've been fixed already: https://github.com/torvalds/linux/commit/ffe51d0d8abce3139a970c640ed48e73e9c360bb

Someone had the same problem as you, with PenMount 6000 not setting the "touch" USB-HID property, which makes the linux kernel driver emit `BTN_LEFT` (left mouse button) instead of `BTN_TOUCH` (user touched the screen), which makes userspace/libinput think it's a mouse instead of a touchscreen.

Maybe something about that driver broke and it's emitting `BTN_LEFT` again, at least that's what the log seem to indicate; or maybe somehow that driver is not built into the kernel in your case.

What distro & kernel are you using?

Maybe it'll work if you add a local libinput quirks file at `/etc/libinput/local-overrides.quirks`:
```
[DIALOGUE INC PenMount 6000 USB]
MatchBus=usb
MatchVendor=0x14E1
MatchProduct=0x6000
AttrInputPropEnable=INPUT_PROP_DIRECT
```

If that doesn't work, can you send me the output of `libinput debug-events` and `libinput record` again (with the quirk file still in place)? If we can't get it working this way, this maybe needs a kernel or libinput patch.

closed 2022-11-28 by github-actions[bot]