ardera/flutter-pi #294: Can't run flutter app on raspberry pi 4B

jmra88, 2022-10-05:
The pc on which I develop is a **64bit ubuntu 18.04**. When I type "**flutter --version**" it shows this:

```
Flutter 3.3.2 • channel stable • https://github.com/flutter/flutter.git
Framework • revision e3c29ec00c (3 weeks ago) • 2022-09-14 08:46:55 -0500
Engine • revision a4ff2c53d8
Tools • Dart 2.18.1 • DevTools 2.15.0
```

The raspberry pi where I try to run the app is a Raspberry 4 model B. If I type "**uname -a**" it shows: "**Linux raspberrypi 5.15.61-v7l+ #1579 SMP Fri Aug 26 11:13:03 BST 2022 armv7l GNU/Linux**"

I have followed the steps that are in: https://github.com/ardera/flutter-pi

When I run "**flutter-pi /home/pi/my_project**" then it shows this error:
```
[flutter-pi] WARNING: display has non-square pixels. Non-square-pixels are not supported by flutter.
[flutter-pi] WARNING: Error getting last vblank timestamp. drmCrtcGetSequence: Invalid argument
[flutter-pi] VSync will be disabled.
See https://github.com/ardera/flutter-pi/issues/38 for more info.
==============Locale==============
Flutter locale:
  default: en_GB
  locales: en_GB en.UTF-8 en.UTF-8 en.UTF-8 en_GB en en.UTF-8
===================================
===================================
display mode:
  resolution: 800 x 480
  refresh rate: 60Hz
  physical size: 154mm x 86mm
  flutter device pixel ratio: 1.367054
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_device EGL_EXT_platform_wayland EGL_KHR_platform_wayland EGL_EXT_platform_x11 EGL_KHR_platform_x11 EGL_MESA_platform_gbm EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 3.1 Mesa 20.3.5"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_filter_anisotropic GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_draw_instanced GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_pack_reverse_row_order GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_NV_pixel_buffer_object GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_primitive_bounding_box GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_view GL_KHR_context_flush_control GL_NV_image_formats GL_OES_draw_elements_base_vertex GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_OES_texture_view GL_EXT_buffer_storage GL_EXT_float_blend GL_EXT_geometry_point_size GL_EXT_geometry_shader GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_geometry_point_size GL_OES_geometry_shader GL_OES_shader_image_atomic GL_EXT_texture_compression_s3tc_srgb GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_EXT_EGL_image_storage GL_MESA_framebuffer_flip_y GL_EXT_texture_query_lod "
===================================
[flutter-pi] Warning: Could not load libflutter_engine.so from the asset bundle. dlopen: /home/pi/FLUTTER_PROJECTS/counter/flutter_assets/libflutter_engine.so: cannot open shared object file: No such file or directory. Trying to open libflutter_engine.so.debug...
flutter: The Dart VM service is listening on http://127.0.0.1:36621/HguwRKsRnvQ=/
[compositor] GPU does not supported the desired HW plane order.
  Some UI layers may be invisible.
[modesetting] Could not commit atomic request. drmModeAtomicCommit: Permission denied
[compositor] Could not present frame. drmModeAtomicCommit: Permission denied
[modesetting] Could not commit atomic request. drmModeAtomicCommit: Permission denied
[compositor] Could not present frame. drmModeAtomicCommit: Permission denied
```

closed 2022-10-06 by jmra88

niklasbartsch, 2022-11-02:
I run into the same problem. 

When I change the boot options to console it works. But I still get the error:

```
[text_input] warning: flutter requested native autocorrect, whichis not supported by flutter-pi.
[compositor] GPU does not supported the desired HW plane order.
  Some UI layers may be invisible.
```

ardera, 2022-11-02:
@niklasbartsch If you boot into desktop it doesn't work because flutter-pi and the desktop environment both try to use the display exclusively (that's why you switch to console mode in the readme)

> When I change the boot options to console it works. But I still get the error:
> 
> ```
> [text_input] warning: flutter requested native autocorrect, whichis not supported by flutter-pi.
> [compositor] GPU does not supported the desired HW plane order.
>   Some UI layers may be invisible.
> ```

Those are just warnings, if the app looks normal its perfectly fine.