ardera/flutter-pi #296: Could not find a fitting libflutter_engine.so, make sure you've installed the engine binaries.

MyisCARRY, 2022-10-25:
Hello!
I have followed the instructions in README.md and now I am stuck on launching debug app. Could you help me?
I have installed precompiled binaries.

Terminal log
```
[flutter-pi] WARNING: display didn't provide valid physical dimensions. The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
[flutter-pi] WARNING: Error getting last vblank timestamp. drmCrtcGetSequence: Invalid argument
[flutter-pi] VSync will be disabled.
See https://github.com/ardera/flutter-pi/issues/38 for more info.
==============Locale==============
Flutter locale:
  default: en_US
  locales: en_US en.UTF-8 en.UTF-8 en.UTF-8 en_US en en.UTF-8
===================================
===================================
display mode:
  resolution: 800 x 600
  refresh rate: 60Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
===================================
VMware: No 3D enabled (0, Success).
[flutter-pi] Could not create GBM Surface. gbm_surface_create_with_modifiers. Will attempt with gbm_surface_create: Function not implemented
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_device EGL_EXT_platform_wayland EGL_KHR_platform_wayland EGL_EXT_platform_x11 EGL_KHR_platform_x11 EGL_MESA_platform_gbm EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_image_dma_buf_export EGL_MESA_query_driver "
===================================
OpenGL ES information:
  version: "OpenGL ES 3.2 Mesa 20.3.5"
  shading language version: "OpenGL ES GLSL ES 3.20"
  vendor: "Mesa/X.org"
  renderer: "llvmpipe (LLVM 11.0.1, 256 bits)"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_compression_rgtc GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_float_linear GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_draw_instanced GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_NV_conditional_render GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_OES_viewport_array GL_ANGLE_pack_reverse_row_order GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_robustness GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_robustness GL_KHR_texture_compression_astc_ldr GL_NV_pixel_buffer_object GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_group_vote GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_tessellation_point_size GL_EXT_tessellation_shader GL_ANDROID_extension_pack_es31a GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_copy_image GL_EXT_draw_buffers_indexed GL_EXT_draw_elements_base_vertex GL_EXT_gpu_shader5 GL_EXT_polygon_offset_clamp GL_EXT_primitive_bounding_box GL_EXT_render_snorm GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_buffer GL_EXT_texture_cube_map_array GL_EXT_texture_norm16 GL_EXT_texture_view GL_KHR_blend_equation_advanced GL_KHR_context_flush_control GL_KHR_robust_buffer_access_behavior GL_NV_image_formats GL_OES_copy_image GL_OES_draw_buffers_indexed GL_OES_draw_elements_base_vertex GL_OES_gpu_shader5 GL_OES_primitive_bounding_box GL_OES_sample_shading GL_OES_sample_variables GL_OES_shader_io_blocks GL_OES_shader_multisample_interpolation GL_OES_tessellation_point_size GL_OES_tessellation_shader GL_OES_texture_border_clamp GL_OES_texture_buffer GL_OES_texture_cube_map_array GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_OES_texture_view GL_EXT_blend_func_extended GL_EXT_buffer_storage GL_EXT_float_blend GL_EXT_geometry_point_size GL_EXT_geometry_shader GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_geometry_point_size GL_OES_geometry_shader GL_OES_shader_image_atomic GL_EXT_clip_cull_distance GL_EXT_disjoint_timer_query GL_EXT_texture_compression_s3tc_srgb GL_MESA_shader_integer_functions GL_EXT_clip_control GL_EXT_color_buffer_half_float GL_EXT_texture_compression_bptc GL_KHR_parallel_shader_compile GL_EXT_EGL_image_storage GL_EXT_shader_framebuffer_fetch_non_coherent GL_EXT_texture_sRGB_R8 GL_EXT_texture_shadow_lod GL_MESA_framebuffer_flip_y GL_EXT_depth_clamp GL_EXT_texture_query_lod "
===================================
WARNING: Detected llvmpipe (ie. software rendering) as the OpenGL ES renderer.
         Check that flutter-pi has permission to use the 3D graphics hardware,
         or try running it as root.
         This warning will probably result in a "failed to set mode" error
         later on in the initialization.
[flutter-pi] Warning: Could not load libflutter_engine.so from the asset bundle. dlopen: /home/myiscarry/flutter_assets/libflutter_engine.so: cannot open shared object file: No such file or directory. Trying to open libflutter_engine.so.debug...
[flutter-pi] Warning: Could not load libflutter_engine.so.debug. dlopen: libflutter_engine.so.debug: cannot open shared object file: No such file or directory. Trying to open libflutter_engine.so...
[flutter-pi] Could not load libflutter_engine.so. dlopen: libflutter_engine.so: cannot open shared object file: No such file or directory
[flutter-pi] Could not find a fitting libflutter_engine.so, make sure you've installed the engine binaries.
```

Also showing how the directories are set
```
├── engine-binaries
│   ├── arm
│   └── arm64
├── flutter_assets
│   ├── fonts
│   ├── packages
│   │   └── cupertino_icons
│   │       └── assets
│   └── shaders
├── flutter-pi
│   ├── build
│   │   ├── CMakeFiles
│   │   │   ├── 3.18.4
│   │   │   │   ├── CompilerIdC
│   │   │   │   │   └── tmp
│   │   │   │   └── CompilerIdCXX
│   │   │   │       └── tmp
│   │   │   ├── CMakeTmp
│   │   │   └── flutter-pi.dir
│   │   │       └── src
│   │   │           └── plugins
│   │   └── _deps
│   │       ├── engine-version-build
│   │       ├── engine-version-src
│   │       └── engine-version-subbuild
│   │           ├── CMakeFiles
│   │           │   ├── 3.18.4
│   │           │   └── engine-version-populate.dir
│   │           └── engine-version-populate-prefix
│   │               ├── src
│   │               │   └── engine-version-populate-stamp
│   │               └── tmp
│   ├── include
│   │   └── plugins
│   ├── src
│   │   └── plugins
│   │       ├── audioplayers
│   │       └── gstreamer_video_player
│   └── test
```

mhoeckner, 2022-10-26:
Try installing the engine_binaries - the are needed - use the same version as used for building the app

MyisCARRY, 2022-10-26:
I did as said in https://github.com/ardera/flutter-engine-binaries-for-arm in step 1.
I build app with flutter 3.3.2

mhoeckner, 2022-10-26:
To precise my comment: you have to use the correct (arm/arm64) engine_binaries 
Also check the location of the binaries

**/home/myiscarry/flutter_assets/libflutter_engine.so**

The engine binaries are not found at this location. Or install them via **install.sh** located in **engine-binaries**

ardera, 2022-10-26:
If you use the `install.sh` (with `sudo` of course), the engine binaries will be installed to `/usr/lib`.

Following files should be present there:
- `libflutter_engine.so.debug`
- `libflutter_engine.so.profile`
- `libflutter_engine.so.release`
- `icudtl.dat`

If one of these is not present, something may have gone wrong while installing, you can try moving them there manually from the cloned engine binaries repo.
If they are there and it still doesn't work, make sure they're executable (`chmod +x ...`)

mhoeckner, 2022-10-26:
@ardera  - thank you for the great work!

MyisCARRY, 2022-10-27:
I found the problem.
`install.sh` from engine-binaries didn't work because I am using Virtualbox with Raspberry Pi Desktop OS and `uname -m` returns `x86_64` instead of `arm` or `arm64`.
That's my bad

ardera, 2022-11-01:
@mhoeckner Appreciate it, thanks!

@MyisCARRY Np, glad you could fix the issue. I'll close this now

closed 2022-11-01 by ardera