ardera/flutter-pi #298: grab all input devices

ardera, 2022-10-26:
use `EVIOCGRAB` to grab all input devices for exclusive use.
otherwise, if people have a keyboard attached and a console running, the keyboard can be used to enter things on the console.

See discussion in #256

niklasbartsch, 2022-11-17:
And how do I use it?

Since the latest Binary update to Flutter 3.3.7 the keyboard input doesn't work anymore in my Flutter App. It only appears in the console.

ardera, 2022-11-19:
@niklasbartsch Should work OOB. If it doesn't, could be libinput exposes your keyboard as both a key and pointer device, and at least in the master branch, that case isn't handled correctly yet.

Though it's handled correctly on the `feature/compositor-ng` branch, can you try there and lmk if it works?

niklasbartsch, 2022-11-20:
Still doesn't work, event with the `feature/compositor-ng` Branch

ardera, 2022-11-20:
@niklasbartsch That's a bug then, can you run `libinput debug-events`, press some keys, and send me the complete output?

Also what does your flutter-pi output look like?

niklasbartsch, 2022-11-21:
I opened the Issue #304.
How do I use `libinput debug-events`?

yaliv, 2024-08-22:
I know, it's really annoying when the keyboard inputs also appear on the console.
Let's say you have a login screen, and some users know that they can close the app with `Ctrl+C`, that would be a security issue.

In my case, I create an autorun script to run my app. So I can use `stty -echo` command as a workaround. [See this SO answer](https://stackoverflow.com/a/4316765/4513925).