ardera/flutter-pi #302: Incorrect touchscreen coordinates

MyisCARRY, 2022-11-14:
Hello!
I have problem with my touchscreen - it works but I need to tap in the "opposite" corner of the screen to tap a button.
Weird thing is that it works perfectly fine when I am in desktop mode.

Raspberry Pi 4 B 4gb
4inch rpi lcd a with xpt2046 touch controller

Let me know what details/logs do you need

ardera, 2022-11-14:
Hey, can you run this on your raspberry pi? It should help you calibrate the screen.
https://github.com/ardera/flutter_libinput_calibrator

MyisCARRY, 2022-11-15:
I build bundle with the same flutter sdk version as engine binaries. Run the app on raspberry pi. Followed the instructions on screen and it worked :D

Thanks for help!

closed 2022-11-15 by MyisCARRY

ardera, 2022-11-15:
np, glad it helped :)