ardera/flutter-pi #303: Screen margins support

bojidartonchev, 2022-11-14:
Hello,

I have a display with actual resolution 640x240, which is detected as a 640x480 display (It is a display controller issue, but it is not an option to modify it). So half of the screen is cut off. Is there a way to patch flutter-pi and achieve bottom margin of 240 pixels?

Best regards!

bojidartonchev, 2022-11-15:
```
Fixed it with the following patch

diff --git a/src/flutter-pi.c b/src/flutter-pi.c
index 8d3b4d6..10963da 100644
--- a/src/flutter-pi.c
+++ b/src/flutter-pi.c
@@ -1452,7 +1452,7 @@ static int init_display(void) {
     // prefer higher refresh rates. After that, prefer progressive scanout modes.
     mode = NULL;
     for_each_mode_in_connector(connector, mode_iter) {
-        if (mode_iter->type & DRM_MODE_TYPE_PREFERRED) {
+        if (mode_iter->hdisplay == 640 && mode_iter->vdisplay == 480 && mode_iter->vrefresh == 60) {
             mode = mode_iter;
             break;
         } else if (mode == NULL) {
@@ -1475,7 +1475,7 @@ static int init_display(void) {
     }
 
     flutterpi.display.width = mode->hdisplay;
-    flutterpi.display.height = mode->vdisplay;
+    flutterpi.display.height = mode->vdisplay / 2;
     flutterpi.display.refresh_rate = mode->vrefresh;
 
     if ((flutterpi.display.width_mm == 0) || (flutterpi.display.height_mm == 0)) {

```

closed 2022-11-15 by bojidartonchev

ardera, 2022-11-21:
Glad you could fix it :)

I think this is a pretty unique use-case (broken HDMI EDID), so not sure it makes sense to support it upstream. Maybe I'll implement it in `feature/compositor-ng`.