ardera/flutter-pi #309: Is V3D graphics driver required?

yannickvg, 2023-01-12:
Hi,

first of all, thanks a lot of this project! I'm a Flutter developer and I love that I can use Flutter on some raspberry pi's I have lying around the house!

The documentation states that the v3d graphics driver needs to be enabled. I'm running Flutter on a raspberry pi 2 with the official 7-inch touchscreen. That touchscreen has the problem that by default the image is upside down and the solution is to add lcd_rotate=2 in the config.txt file. As the people of the Pi Hut state on this page (https://support.thepihut.com/hc/en-us/articles/115002072865-Official-touchscreen-is-upside-down) for some people the property dtoverlay=vc4-fkms-v3d also needs to be disabled for the rotation property to work. 

Apparently I'm one of those people so I have two options:

- If I disable that property the Flutter app launches but it it upside down (together with the console before launching Flutter)
- If I enable that v3d property the device launches with the screen in the correct rotation but the Flutter app doesn't launch. It gives no error in the console but I suppose it is because of the v3d driver not being enabled.

Do you have any ideas for a solution? Is that v3d driver a requirement for Flutter Pi or is there a workaround?

Thanks very much in advance for any help!

Yannick

ardera, 2023-01-13:
Hey Yannick,

> first of all, thanks a lot of this project! I'm a Flutter developer and I love that I can use Flutter on some raspberry pi's I have lying around the house!

Thanks! Glad you find it useful :)

There's some other ways you can fix it, without `lcd_rotate`:
- run flutter-pi with `-r 180`
- use a `RotatedBox` or a `Transform.rotate` inside flutter

yannickvg, 2023-01-13:
Cool, the -r 180 option works like charm! thanks!

closed 2023-01-13 by yannickvg