ardera/flutter-pi #312: VideoPlayer is not working (Gstreamer).

edgardiano, 2023-01-27:
First of all, thanks for creating flutter-pi, this is such a great project!

I am having some issues after trying to run an application with the video_player package, exactly the one recommended in the example part of Readme (https://pub.dev/packages/video_player#example). Please fin the details below:

-flutter_assets bundle generated in Ubuntu 18.04 machine (all process)
-video_player: 2.5.1 (and with some older versions too like 2.4.5) 
-Device: STM32MP157F-EV1 (32 bits MPU)
-Flutter: 3.3.10
-flutter-pi: Latest version available in Master
-engine-binaries: Latest version available in your repo.

Command: `./flutter-pi -r 270 --release /usr/local/flutter331/flutter_assets/`

Output:
`[locales] Warning: The system has no configured locale. The default "C" locale may or may not be supported by the app.
[keyboard] Could not load keyboard configuration from "/etc/default/keyboard". Default keyboard config will be used. load_file: No such file or directory
[flutter-pi] WARNING: display has non-square pixels. Non-square-pixels are not supported by flutter.
==============Locale==============
Flutter locale:
  default: C
  locales: C
===================================
===================================
display mode:
  resolution: 720 x 1280
  refresh rate: 50Hz
  physical size: 68mm x 122mm
  flutter device pixel ratio: 2.786378
===================================
EGL information:
  version: 1.5
  vendor: "Vivante Corporation"
  client extensions: "EGL_EXT_client_extensions EGL_EXT_platform_base EGL_KHR_platform_wayland EGL_EXT_platform_wayland EGL_KHR_platform_gbm"
  display extensions: "EGL_KHR_fence_sync EGL_KHR_reusable_sync EGL_KHR_wait_sync EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_gl_renderbuffer_image EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_lock_surface EGL_KHR_create_context EGL_KHR_no_config_context EGL_KHR_surfaceless_context EGL_KHR_get_all_proc_addresses EGL_EXT_buffer_age EGL_ANDROID_native_fence_sync EGL_WL_bind_wayland_display EGL_WL_create_wayland_buffer_from_image EGL_KHR_partial_update EGL_EXT_swap_buffers_with_damage EGL_KHR_swap_buffers_with_damage"
===================================
OpenGL ES information:
  version: "OpenGL ES 2.0 V6.4.3.279124"
  shading language version: "OpenGL ES GLSL ES 1.0.0"
  vendor: "Vivante Corporation"
  renderer: "Vivante GCNano"
  extensions: "GL_OES_vertex_half_float GL_OES_element_index_uint GL_OES_mapbuffer GL_OES_vertex_array_object GL_OES_compressed_ETC1_RGB8_texture GL_OES_compressed_paletted_texture GL_OES_texture_npot GL_OES_rgb8_rgba8 GL_OES_depth_texture GL_OES_depth_texture_cube_map GL_OES_depth24 GL_OES_depth32 GL_OES_packed_depth_stencil GL_OES_fbo_render_mipmap GL_OES_get_program_binary GL_OES_fragment_precision_high GL_OES_standard_derivatives GL_OES_EGL_image GL_OES_EGL_image_external GL_OES_EGL_image_external_essl3 GL_OES_EGL_sync GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_texture_type_2_10_10_10_REV GL_EXT_texture_compression_dxt1 GL_EXT_texture_format_BGRA8888 GL_EXT_texture_compression_s3tc GL_EXT_read_format_bgra GL_EXT_multi_draw_arrays GL_EXT_frag_depth GL_EXT_discard_framebuffer GL_EXT_blend_minmax GL_EXT_robustness GL_EXT_texture_sRGB_decode GL_EXT_texture_rg GL_VIV_direct_texture "
===================================
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: UnimplementedError: init() has not been implemented.
#0      VideoPlayerPlatform.init (package:video_player_platform_interface/video_player_platform_interface.dart:43)
#1      _videoPlayerPlatform (package:video_player/video_player.dart:28)
#2      VideoPlayerController.initialize (package:video_player/video_player.dart:362)
#3      _VideoAppState.initState (package:video_test_331/main.dart:22)
#4      StatefulElement._firstBuild (package:flutter/src/widgets/framework.dart:5015)
#5      ComponentElement.mount (package:flutter/src/widgets/framework.dart:4853)
#6      Element.inflateWidget (package:flutter/src/widgets/framework.dart:3863)
#7      Element.updateChild (package:flutter/src/widgets/framework.dart:3592)
#8      RenderObjectToWidgetElement._rebuild (package:flutter/src/widgets/binding.dart:1195)
#9      RenderObjectToWidgetElement.mount (package:flutter/src/widgets/binding.dart:1164)
#10     RenderObjectToWidgetAdapter.attachToRenderTree.<anonymous closure> (package:flutter/src/widgets/binding.dart:1111)
#11     BuildOwner.buildScope (package:flutter/src/widgets/framework.dart:2605)
#12     RenderObjectToWidgetAdapter.attachToRenderTree (package:flutter/src/widgets/binding.dart:1110)
#13     WidgetsBinding.attachRootWidget (package:flutter/src/widgets/binding.dart:945)
#14     WidgetsBinding.scheduleAttachRootWidget.<anonymous closure> (package:flutter/src/widgets/binding.dart:925)
#15     Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
#16     _Timer._runTimers (dart:isolate-patch/timer_impl.dart:398)
#17     _Timer._handleMessage (dart:isolate-patch/timer_impl.dart:429)
#18     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:192)

[compositor] GPU does not support reflecting the screen in Y-direction.
  This is required for rendering into hardware overlay planes though.
  Any UI that is drawn in overlay planes will look upside down.
[compositor] GPU does not supported the desired HW plane order.
  Some UI layers may be invisible.`

I have also tried VideoPlayerController.asset instead of .network, but issue persists.

In addition, during flutter-pi building, after doing make command, different errors were displayed:
`[  4%] Building C object CMakeFiles/flutter-pi.dir/src/flutter-pi.c.o
In file included from /usr/local/flutter331/flutter-pi/src/flutter-pi.c:47:
/usr/local/flutter331/flutter-pi/include/flutter-pi.h:57:2: error: expected specifier-qualifier-list before 'FlutterEngineResult'
   57 |  FlutterEngineResult (*FlutterEngineCreateAOTData)(const FlutterEngineAOTDataSource* source, FlutterEngineAOTData* data_out);
      |  ^~~~~~~~~~~~~~~~~~~
In file included from /usr/local/flutter331/flutter-pi/src/flutter-pi.c:47:
/usr/local/flutter331/flutter-pi/include/flutter-pi.h:160:8: error: unknown type name 'FlutterTransformation'
  160 |  const FlutterTransformation t,
      |        ^~~~~~~~~~~~~~~~~~~~~
/usr/local/flutter331/flutter-pi/include/flutter-pi.h: In function 'apply_flutter_transformation':
/usr/local/flutter331/flutter-pi/include/flutter-pi.h:168:10: error: request for member 'scaleX' in something not a structure or union
  168 |   *px = t.scaleX*x + t.skewX*y + t.transX;`

I could solve them by setting FLUTTER_EMBEDDER_HEADER (in the CMake file) to the flutter_embedder.h provided in your engine-binaries repo. 

Thanks in advance!

ardera, 2023-01-29:
The default video player platform implementation was removed a while ago, I'm working on a fix.

For now you'll have to explicitly add a dependency on `android_video_player: 2.3.0` to `pubspec.yaml` and call `AndroidVideoPlayer.registerWith` in main.

Binozo, 2023-02-04:
Hi ardera,

first of all thank you for this awesome project!
I'm having the same issue and I'm looking forward for a fix.
But no pressure, take your time 😄

Vallevard, 2023-02-23:
@edgardiano did you find a solution?
Thanks

ardera, 2023-02-27:
@Binozo @Vallevard 
Try using the [`flutterpi_gstreamer_video_player`](https://pub.dev/packages/flutterpi_gstreamer_video_player) package and the [`feature/gstreamer-player-from-pipeline`](https://github.com/ardera/flutter-pi/tree/feature/gstreamer-player-from-pipeline) branch.

Binozo, 2023-02-27:
@ardera I tried using the package and your suggested branch but I got the following error:

```
[flutter-pi] WARNING: display didn't provide valid physical dimensions. The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
==============Locale==============                                                                                      
Flutter locale:                                                                                                           
default: en_GB                                                                                                          
locales: en_GB en.UTF-8 en.UTF-8 en.UTF-8 en_GB en en.UTF-8                                                           ===================================                                                                                     ===================================                                                                                     
display mode:                                                                                                             
resolution: 1920 x 1080                                                                                                 
refresh rate: 60Hz                                                                                                      
physical size: 0mm x 0mm                                                                                               
flutter device pixel ratio: 1.000000                                                                                  ===================================                                                                                     
EGL information:                                                                                                          
version: 1.4                                                                                                           
 vendor: "Mesa Project"                                                                                                  
client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_device EGL_EXT_platform_wayland EGL_KHR_platform_wayland EGL_EXT_platform_x11 EGL_KHR_platform_x11 EGL_MESA_platform_gbm EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless"                                                                                                       display extensions: "EGL_ANDROID_blob_cache EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "             
===================================
mmap of bo 1 (offset 0x00000000113f5000, size 524288) failed
Aborted
```

ardera, 2023-02-27:
@Binozo mh that's interesting. is that the complete log?

I'll try using HDMI too, I've only tried using DSI. Though not sure that makes a difference.

Can you send me the following infos:
- pi model (pi 3/4 a/b, how much RAM)
- flutter code where you create the video player controller
- can you run: 
   ```bash
   echo 0x1FF | sudo tee /sys/module/drm/parameters/debug # Enable verbose DRM logging
   sudo dmesg -C # Clear kernel logs
   dmesg -w >dmesg.log & # Continuously write DRM logs to a file, in the background
   flutter-pi ... # Run flutter-pi again, manually terminate it after the error has happened
   fg # Kill dmesg with Ctrl+C
   echo 0 | sudo tee /sys/module/drm/parameters/debug # Disable DRM logging
   ```
   and attach the `dmesg.log` file?

Binozo, 2023-02-27:
@ardera It runs and terminates instantly
I'm using a Raspberry Pi Zero 2 w which is connected to my tv with hdmi.

I think flutter-pi doesn't even start to run the flutter app but anyway here is my code:

`main.dart`
```dart
import 'package:flutterpi_gstreamer_video_player/flutterpi_gstreamer_video_player.dart';
import 'package:video_player/video_player.dart';
import 'package:youtube_explode_dart/youtube_explode_dart.dart';
import 'package:flutter/material.dart';

void main() {
  print("Registering.....");
  FlutterpiVideoPlayer.registerWith();
  runApp(const MyApp());
  print("hier");
}

class MyApp extends StatelessWidget {
  const MyApp({super.key});

  // This widget is the root of your application.
  @override
  Widget build(BuildContext context) {
    return MaterialApp(
      title: 'Flutter Demo',
      theme: ThemeData(
        // This is the theme of your application.
        //
        // Try running your application with "flutter run". You'll see the
        // application has a blue toolbar. Then, without quitting the app, try
        // changing the primarySwatch below to Colors.green and then invoke
        // "hot reload" (press "r" in the console where you ran "flutter run",
        // or simply save your changes to "hot reload" in a Flutter IDE).
        // Notice that the counter didn't reset back to zero; the application
        // is not restarted.
        primarySwatch: Colors.blue,
      ),
      home: const MyHomePage(title: 'Flutter Demo Home Page'),
    );
  }
}

class MyHomePage extends StatefulWidget {
  const MyHomePage({super.key, required this.title});

  // This widget is the home page of your application. It is stateful, meaning
  // that it has a State object (defined below) that contains fields that affect
  // how it looks.

  // This class is the configuration for the state. It holds the values (in this
  // case the title) provided by the parent (in this case the App widget) and
  // used by the build method of the State. Fields in a Widget subclass are
  // always marked "final".

  final String title;

  @override
  State<MyHomePage> createState() => _MyHomePageState();
}

class _MyHomePageState extends State<MyHomePage> {
  int _counter = 0;
  String _stream = "";
  late VideoPlayerController _controller;
  final url = "https://rr4---sn-h0jeenld.googlevideo.com/videoplayback?expire=1675469345&ei=wU3dY-TqJIKR7QSY54b4Ag&ip=91.89.115.158&id=o-ALKl6ig8FMPhZBjNicz05SyGPrFEHcM9-VXu-8ZSVzgC&itag=22&source=youtube&requiressl=yes&mh=OK&mm=31%2C26&mn=sn-h0jeenld%2Csn-4g5ednkl&ms=au%2Conr&mv=m&mvi=4&pl=22&initcwndbps=1875000&vprv=1&mime=video%2Fmp4&cnr=14&ratebypass=yes&dur=1188.954&lmt=1627902618655246&mt=1675447049&fvip=3&fexp=24007246&c=ANDROID&txp=5535432&sparams=expire%2Cei%2Cip%2Cid%2Citag%2Csource%2Crequiressl%2Cvprv%2Cmime%2Ccnr%2Cratebypass%2Cdur%2Clmt&sig=AOq0QJ8wRgIhAKP_JtzNRrYKXdBcxCUMVx8w7UGqHjm0CeK9oUIpp31BAiEA0mGTgT72km8KJjFQHEGNcbBG6XTmtEljCDSi6LhTNYo%3D&lsparams=mh%2Cmm%2Cmn%2Cms%2Cmv%2Cmvi%2Cpl%2Cinitcwndbps&lsig=AG3C_xAwRQIgfpZFBscIwfZlUMEZmfLJ_2Rx7m8pSaMb9aFuhJVqBU4CIQD8ONs7H_EsTk-GAr48yjw6vMP6RKWKfqvnGvMUOoHVqw%3D%3D";

  void _incrementCounter() async {
    var yt = YoutubeExplode();
[... 1560 lines cut ...]
[ 2456.739622] vc4-drm soc:gpu: [drm:drm_crtc_vblank_helper_get_vblank_timestamp_internal [drm]] crtc 3 : v p(0,1076)@ 2456.738833 -> 2456.722892 [e 0 us, 0 rep]
[ 2456.739983] vc4-drm soc:gpu: [drm:drm_update_vblank_count [drm]] crtc 3: Calculating number of vblanks. diff_ns = 83369534, framedur_ns = 16666666)
[ 2456.740335] vc4-drm soc:gpu: [drm:drm_update_vblank_count [drm]] updating vblank count on crtc 3: current=146615, diff=5, hw=0 hw_last=0
[ 2456.740703] vc4-drm soc:gpu: [drm:drm_crtc_vblank_helper_get_vblank_timestamp_internal [drm]] crtc 3 : v p(0,26)@ 2456.739914 -> 2456.739529 [e 0 us, 0 rep]
[ 2456.741063] vc4-drm soc:gpu: [drm:drm_update_vblank_count [drm]] crtc 3: Calculating number of vblanks. diff_ns = 16637326, framedur_ns = 16666666)
[ 2456.741416] vc4-drm soc:gpu: [drm:drm_update_vblank_count [drm]] updating vblank count on crtc 3: current=146620, diff=1, hw=0 hw_last=0
[ 2456.756322] vc4-drm soc:gpu: [drm:drm_crtc_vblank_helper_get_vblank_timestamp_internal [drm]] crtc 3 : v p(0,-45)@ 2456.755523 -> 2456.756189 [e 0 us, 0 rep]
[ 2456.756746] vc4-drm soc:gpu: [drm:drm_update_vblank_count [drm]] crtc 3: Calculating number of vblanks. diff_ns = 16660230, framedur_ns = 16666666)
[ 2456.757102] vc4-drm soc:gpu: [drm:drm_update_vblank_count [drm]] updating vblank count on crtc 3: current=146621, diff=1, hw=0 hw_last=0
[ 2456.757553] [drm:drm_atomic_state_default_clear [drm]] Clearing atomic state 5a63026c
[ 2456.757920] [drm:drm_mode_object_put.part.0 [drm]] OBJ ID: 32 (5)
[ 2456.758271] [drm:drm_mode_object_put.part.0 [drm]] OBJ ID: 32 (4)
[ 2456.758646] [drm:drm_mode_object_put.part.0 [drm]] OBJ ID: 292 (2)
[ 2456.759009] [drm:drm_mode_object_put.part.0 [drm]] OBJ ID: 291 (3)
[ 2456.759407] [drm:__drm_atomic_state_free [drm]] Freeing atomic state 5a63026c
[ 2456.759774] vc4-drm soc:gpu: [drm:drm_client_dev_restore [drm]] fbdev: ret=0
[ 2456.772986] vc4-drm soc:gpu: [drm:drm_crtc_vblank_helper_get_vblank_timestamp_internal [drm]] crtc 3 : v p(0,-45)@ 2456.772187 -> 2456.772854 [e 0 us, 0 rep]
[ 2456.773415] vc4-drm soc:gpu: [drm:drm_update_vblank_count [drm]] crtc 3: Calculating number of vblanks. diff_ns = 16664733, framedur_ns = 16666666)
[... cut ...]
[ 2456.774135] vc4-drm soc:gpu: [drm:vblank_disable_fn [drm]] disabling vblank on crtc 3
[ 2456.774492] vc4-drm soc:gpu: [drm:drm_crtc_vblank_helper_get_vblank_timestamp_internal [drm]] crtc 3 : v p(0,57)@ 2456.773703 -> 2456.772859 [e 1 us, 0 rep]
[ 2456.774849] vc4-drm soc:gpu: [drm:drm_update_vblank_count [drm]] crtc 3: Calculating number of vblanks. diff_ns = 4619, framedur_ns = 16666666)
[ 2456.775200] vc4-drm soc:gpu: [drm:drm_update_vblank_count [drm]] updating vblank count on crtc 3: current=146623, diff=0, hw=0 hw_last=0

```

ardera, 2023-02-27:
@Binozo Can you try again with with `feature/gstreamer-player-from-pipeline`? Should be fixed now.

Binozo, 2023-02-27:
@ardera you fixed that error yay😄

But sadly I get another error while starting the flutter app:

<details>

<summary> Logs </summary>

```

[flutter-pi] WARNING: display didn't provide valid physical dimensions. The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
==============Locale==============
Flutter locale:
  default: en_GB
  locales: en_GB en.UTF-8 en.UTF-8 en.UTF-8 en_GB en en.UTF-8
===================================
===================================
display mode:
  resolution: 1920 x 1080
  refresh rate: 60Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_device EGL_EXT_platform_wayland EGL_KHR_platform_wayland EGL_EXT_platform_x11 EGL_KHR_platform_x11 EGL_MESA_platform_gbm EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless"
  display extensions: "EGL_ANDROID_blob_cache EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 2.0 Mesa 20.3.5"
  shading language version: "OpenGL ES GLSL ES 1.0.16"
  vendor: "Broadcom"
  renderer: "VC4 V3D 2.1"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_EGL_image GL_OES_depth_texture GL_AMD_performance_monitor GL_OES_packed_depth_stencil GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_pack_reverse_row_order GL_EXT_occlusion_query_boolean GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_NV_pixel_buffer_object GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_separate_shader_objects GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_KHR_parallel_shader_compile GL_MESA_tile_raster_order "
===================================
detected VideoCore IV as underlying graphics chip, and VC4 as the driver.
Reporting modified GL_EXTENSIONS string that doesn't contain non-working extensions.
flutter: The Dart VM service is listening on http://127.0.0.1:38075/MC__DGINt-Q=/
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: UnimplementedError: init() has not been implemented.
#0      VideoPlayerPlatform.init (package:video_player_platform_interface/video_player_platform_interface.dart:43:5)
#1      _videoPlayerPlatform (package:video_player/video_player.dart:28:21)
#2      VideoPlayerController.initialize (package:video_player/video_player.dart:362:25)
#3      _MyHomePageState.initState (package:sandbox/main.dart:74:9)
#4      StatefulElement._firstBuild (package:flutter/src/widgets/framework.dart:5101:55)
#5      ComponentElement.mount (package:flutter/src/widgets/framework.dart:4944:5)
#6      Element.inflateWidget (package:flutter/src/widgets/framework.dart:3953:16)
#7      Element.updateChild (package:flutter/src/widgets/framework.dart:3682:18)
#8      SingleChildRenderObjectElement.mount (package:flutter/src/widgets/framework.dart:6377:14)
#9      Element.inflateWidget (package:flutter/src/widgets/framework.dart:3953:16)
#10     Element.updateChild (package:flutter/src/widgets/framework.dart:3682:18)
#11     ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:4993:16)
#12     Element.rebuild (package:flutter/src/widgets/framework.dart:4690:5)
#13     ComponentElement._firstBuild (package:flutter/src/widgets/framework.dart:4950:5)
#14     ComponentElement.mount (package:flutter/src/widgets/framework.dart:4944:5)
#15     Element.inflateWidget (package:flutter/src/widgets/framework.dart:3953:16)
#16     Element.updateChild (package:flutter/src/widgets/framework.dart:3682:18)
#17     SingleChildRenderObjectElement.mount (package:flutter/src/widgets/framework.dart:6377:14)
#18     Element.inflateWidget (package:flutter/src/widgets/framework.dart:3953:16)
[... cut ...]
#696    WidgetsBinding.scheduleAttachRootWidget.<anonymous closure> (package:flutter/src/widgets/binding.dart:906:7)
#697    Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18:15)
#698    _Timer._runTimers (dart:isolate-patch/timer_impl.dart:398:19)
#699    _Timer._handleMessage (dart:isolate-patch/timer_impl.dart:429:5)
#700    _RawReceivePort._handleMessage (dart:isolate-patch/isolate_patch.dart:192:26)

```

</details>

Did I implemented your flutter gstreamer package wrong or what do you think could cause that error?

ardera, 2023-02-27:
@Binozo You need to call `FlutterpiVideoPlayer.registerWith()` in main, before runApp :-)

Binozo, 2023-02-28:
@ardera yeah I think I am already doing it 😅

```dart
void main() { 
    print("Registering.....");
    FlutterpiVideoPlayer.registerWith();
    runApp(const MyApp()); 
    print("hier"); 
}
```

ardera, 2023-02-28:
@Binozo I'll try reproducing it, though the [example app](https://github.com/ardera/flutter_packages/tree/main/packages/flutterpi_gstreamer_video_player/example) I wrote works for me :sweat_smile: 

Very weird error. `FlutterpiVideoPlayer.registerWith()` just sets `VideoPlayerPlatform.instance` to `FlutterpiVideoPlayer()`. And in your error message it seems like `VideoPlayerPlatform.instance` was never set to anything and left at default. (The default implementation just throws an `UnimplementedError` everywhere)

Vallevard, 2023-02-28:
Hello,
For me it works using as @ardera said: [flutterpi_gstreamer_video_player](https://pub.dev/packages/flutterpi_gstreamer_video_player) package and the [feature/gstreamer-player-from-pipeline](https://github.com/ardera/flutter-pi/tree/feature/gstreamer-player-from-pipeline) branch and calling **FlutterpiVideoPlayer.registerWith()**

OS: Raspian OS 64bits
Device: Raspberry Pi 4, 8gb RAM.

I use flutter for signage with raspberry and I have to loop one video or if there are multiple videos reproduce each video and start again when the last video end. 
FYI
I'm not if the problem is related with the package or gstreamer but now with the video package updated does not reproduce again the videos and I receive the error: [gstreamer_video_player] Failed to set custom HTTP headers because gstreamer source element has no 'extra-headers' property.

If I find something, I'll back to you.
Thank you.

bojidartonchev, 2023-05-04:
Hello everyone,

Any progress on this? @ardera Do you plan to bring it to master any time soon?

Best regards,
Bozhidar

jtkeyva, 2023-05-12:
Hello,
Great job with flutter-pi!! I think the missing link is video playback (and camera video recording).

I have tried what seems like everythin and my latest attempt is using the gstreamer-player-from-pipeline branch and 
running the example app from the flutterpi_gstreamer_video_player 0.1.0+1  package.

The app loads and i see the tabs "Example Video" and "Camera" but the main screen area is grey background with controls and then turns white. No video. (Also no mouse cursor)

Is it possible to record a youtube video or something of how to get this working as of May 11 2023? This could save a lot of time and head banging. I've gone through the issues and tried what seems like everything. 

Any help greatly appreciated.

Thanks!

ardera, 2023-05-16:
@jtkeyva The above error message sounds like the gstreamer player isn't built into flutter-pi, or maybe you're not using the `feature/gstreamer-player-from-pipeline` branch there. Can you check the configure log, if it says anything about the video player not being built?

Audio should be working by default, it works for me at least

ardera, 2023-05-17:
What video are you playing back? The video player does not change resolution or color depth in any way, it just uses whatever the source provides it with.

Good point, I haven't tried if HDMI sound works yet. I've tried AUX in the past and that worked

dqbd, 2023-05-19:
Tried the branch `feature/gstreamer-player-from-pipeline` together with `flutterpi_gstreamer_video_player` and video works fine. However, audio is not being played at all. Tried both HDMI and AUX (changing via `raspi-config` as well). 

Example app using the bee.mp4 (https://github.com/ardera/flutter-pi/assets/1443449/c7a5ebed-75c4-4943-934b-87c0e9d34d62):
```
import "package:flutter/material.dart";
import "package:video_player/video_player.dart";
import "package:flutterpi_gstreamer_video_player/flutterpi_gstreamer_video_player.dart";
import "package:flutter/foundation.dart" show kIsWeb;

void main() {
  if (!kIsWeb) {
    FlutterpiVideoPlayer.registerWith();
  }
  runApp(const MyApp());
}

class VideoPlayerScreen extends StatefulWidget {
  const VideoPlayerScreen({super.key});

  @override
  State<VideoPlayerScreen> createState() => _VideoPlayerScreenState();
}

class _VideoPlayerScreenState extends State<VideoPlayerScreen> {
  late VideoPlayerController _controller;
  @override
  void initState() {
    super.initState();
    _controller = VideoPlayerController.asset("assets/bee.mp4")
      ..initialize().then((_) {
        _controller.setVolume(1.0);
        _controller.setLooping(true);
        _controller.play();
        setState(() {});
      });
  }

  @override
  Widget build(BuildContext context) {
    return SizedBox(
      width: 800,
      height: 600,
      child: VideoPlayer(_controller),
    );
  }

  @override
  void dispose() {
    super.dispose();
    _controller.dispose();
  }
}

class MyApp extends StatelessWidget {
  const MyApp({super.key});

  @override
  Widget build(BuildContext context) {
    return const MaterialApp(
      title: 'Flutter Demo',
      home: MyHomePage(title: 'Flutter Demo Home Page'),
    );
  }
}

class MyHomePage extends StatefulWidget {
  const MyHomePage({super.key, required this.title});

  final String title;

  @override
  State<MyHomePage> createState() => _MyHomePageState();
}

class _MyHomePageState extends State<MyHomePage> {
  @override
  Widget build(BuildContext context) {
    return const Scaffold(
      backgroundColor: Colors.black,
      body: Center(
        child: Column(
          mainAxisAlignment: MainAxisAlignment.center,
          children: <Widget>[VideoPlayerScreen()],
        ),
      ),
    );
  }
}

```

dqbd, 2023-05-19:
Video ceases to play after being in a loop for a while:

```
[gstreamer video_player] gstreamer error: code: 1, domain: gst-stream-error-quark, msg: Internal data stream error. (debug info: ../gst/isomp4/qtdemux.c(6545): gst_qtdemux_loop (): /GstPipeline:pipeline0/GstURIDecodeBin:src/GstDecodeBin:decodebin0/GstQTDemux:qtdemux0:
streaming stopped, reason error (-5))
[gstreamer video_player] gstreamer error: code: 1, domain: gst-stream-error-quark, msg: Internal data stream error. (debug info: ../plugins/elements/gstmultiqueue.c(2670): gst_multi_queue_sink_event (): /GstPipeline:pipeline0/GstURIDecodeBin:src/GstDecodeBin:decodebin0/GstMultiQueue:multiqueue0:
streaming stopped, reason error (-5))
```

jtkeyva, 2023-05-19:
> @jtkeyva What video are you playing back? The video player does not change resolution or color depth in any way, it just uses whatever the source provides it with.
> 
> Good point, I haven't tried if HDMI sound works yet. I've tried AUX in the past and that worked

I tried the bee video and several others, no sound working via HDMI. 

In the past I as able to modify config.txt and set the pi's color depth, resolution etc. in raspbian. is this still a thing with the newer raspberry pi os? i notice my images have banding when displayed with flutter. i particularly notice a subpar quality when connected to my tv.

btw: i'm on raspberry pi 2b.

in my previous setup i was using omx player along with pygame and it worked and looked good but i would way rather use flutter if i can get the same video playback quality with sound and high resolution full color image rendering quality.

jvictorsoto, 2023-05-29:
Hi @ardera, player works fine, but status is not updated (videoPlayerController.value.position is allways 0 & videoPlayerController.value.isPlaying does not change to false when video ends). Also no callback to eventListener when playing stops.
Right now I'm using a timer running parallel to the player as a workarround.

jtkeyva, 2023-05-30:
> Hi @ardera, player works fine, but status is not updated (videoPlayerController.value.position is allways 0 & videoPlayerController.value.isPlaying does not change to false when video ends). Also no [...]

@jvictorsoto what pi hardware and OS are you running? you getting sound? is the video quality crisp?
thanks

jvictorsoto, 2023-05-30:
> @jvictorsoto what pi hardware and OS are you running? you getting sound? is the video quality crisp? thanks

@ardera its a custom board running rpi CM4 with 2560x1440 screen, quality is good (i'm using h264 main profile 10mbps), no sound in the board and no sound in the video files so I can not answer about the sound quality.

ardera, 2023-08-22:
fixed by #323

closed 2023-08-22 by ardera