ardera/flutter-pi #313: printer

AndroidDesigner, 2023-01-27:
Hi everyone.
I want to use a usb label printer with raspberry pi.
I did try different packages and plugins such as [printing](https://pub.dev/packages/printing) but none of them works.
Do I have to use Linux commands such as ``lp`` directly or did you have use a flutter package/plugin successfully?

ardera, 2023-01-27:
There's no printing package for flutter-pi I know of, I'm afraid. Technically you could port the printing package to flutter-pi, but that requires some low-level knowledge of flutter-pi and flutter plugins.
I'd try the `lp` approach you mentioned

AndroidDesigner, 2023-01-28:
Do you think can we use ``lp`` command in flutter-pi?
There are multiple packages to run linux command in flutter (such as [process_run](https://pub.dev/packages/process_run))
Can you please test it in a sample project?

ardera, 2023-01-29:
You don't need a package to run processes in dart, there's standard Dart API for that. (dart:io Process.run)

I think you should try it yourself, I'm pretty busy. :-)

AndroidDesigner, 2023-01-31:
Unfortunately, I could not find a label printer with arm64 driver. so nothing is printed with ``lp`` and ``lpr`` and ``cups admin page``.
What is your solution?

AndroidDesigner, 2023-02-01:
Finally I did it.

closed 2023-02-01 by AndroidDesigner