ardera/flutter-pi #315: Some issues related to older i915 rendering on a Celeron G470

Abnormalpingu, 2023-02-14:
Hi, 

We have some x86 Shuttle pc's and a variety of older intel based hardware out in the wild, our plan is to network boot all our clients from a central local pxe server in read only, as all we need is all the clients starting up, logging in and starting our app (based on flutter as you might've already guessed). 

The rpi4's will be booted with raspian (latest stable) 64-bit, with no desktop environment, it works perfectly.
The shuttle's and the older "custom" machines will be running debian 11 64-bit (latest stable) - the debian network boot version is built using debootstrap just to minimize it as much as possible. 
Initial testing on the shuttle machines works perferctly after we installed the **libdrm-intel1** package. 

Our older intel hardware did not go as we hoped, and there are some stuff we'd really appreciate some pointers and ultimately some help getting to work.

We first tried networkbooting the old client like the already working Shuttle, OS wise it works fine, but when we get to flutter-pi and running our app we ran into issues. 
_Just to test stuff (and without the network boot complexities) I've installed Debian 11 from scratch on the older machine._ 

When starting the app we see the following; [link](https://pastebin.com/HPj0LqmL) 

If we try removing the 'libdrm-intel1' package we see the following; [link](https://pastebin.com/NQJU8UgB)

(just to illustrate the intel package actually does something)

I did some digging and came over the 'kmscube' package, just to see if it renders anything at all, which also works when we do 'kmscube --atomic' 

Here is the output from kmscube with the libdrm-intel1 package installed; [link](https://pastebin.com/WP1dr0j4)
Here is the output without the libdrm-intel1 package installed; [link](https://pastebin.com/4uYfjJhQ)

Our "older intel"-stuff runs an ASUS H61M-F motherboard with a [Celeron G470 CPU](https://www.intel.com/content/www/us/en/products/sku/74390/intel-celeron-processor-g470-1-5m-cache-2-00-ghz/specifications.html)
After som digging the CPU has a i915 (driver mentions i965 but not sure if this is relevant, see the log) hd gpu (apu*), which should be sufficient for our requirements (the app doesn't have any overly complicated graphics).

The app is built on a windows 10 box using (powershell; Flutter build foldername)
Flutter 3.7.3
Dart 2.19.2
DevTools 2.20.1

Debian 11 client is running 5.10.0-20-amd64
Flutter-pi is built from https://github.com/ardera/flutter-pi
https://github.com/sony/flutter-embedded-linux/releases/download/b24591ed32/elinux-x64-debug.zip
https://raw.githubusercontent.com/ardera/flutter-engine-binaries-for-arm/main/flutter_embedder.h
https://github.com/ardera/flutter-engine-binaries-for-arm/raw/main/arm64/icudtl.dat

Installation is done according to the documentation provided on this github. 
(sorry for the lengthy post)

ardera, 2023-02-14:
Hey,

if kmscube can output things, flutter-pi probably just does something wrong when speaking to the modesetting device. (Sometimes video drivers have very specific constraints that userspace applications like flutter-pi need to fulfill)

Can you do the following:
- Upload some details about your video driver using the instructions here: https://drmdb.emersion.fr/
  - (not technically required, just to keep the database up to date :) )
- Install `libdrm-tests` and run `modetest`, and send me the full output of that
- run:
  ```bash
  echo 0x1FF | sudo tee /sys/module/drm/parameters/debug # Enable verbose DRM logging
  sudo dmesg -C # Clear kernel logs
  dmesg -w >dmesg.log & # Continuously write DRM logs to a file, in the background
  flutter-pi ... # Run flutter-pi again, manually terminate it after a few errors have happened
  fg # Kill dmesg with Ctrl+C
  echo 0 | sudo tee /sys/module/drm/parameters/debug # Disable DRM logging
  ```
- and send me the `dmesg.log` file

Abnormalpingu, 2023-02-15:
Hi, 

yeah that's what i thought too, my skillset is somewhat limited with Linux so thank you so much for the quick and detailed reply ! 

I was able to dump the driver details which can be reached [here](https://drmdb.emersion.fr/snapshots/2649049f4709
) 
The modtest output looks like [this](https://pastebin.com/6bXFGdUx)
I was able to create the dmesg.log which is attached to this post. 
Let me know if you want to use something else for the file, it was the quickest thing I found.. :) 

-Dan
[dmesg.log](https://github.com/ardera/flutter-pi/files/10741430/dmesg.log)

ardera, 2023-02-15:
Okay, seems like your display controller doesn't support an ARGB as the pixel format for the output planes.

Should be easy to fix

Abnormalpingu, 2023-02-16:
Fantastic news, I was worried we had hit a hardware limitation, which wouldn't have suprised me due to the age of the hardware. 
Is there something specific we can do in our app to remedy this or is this a flutter-pi specific issue ?

ardera, 2023-02-17:
This is a flutter-pi specific issue, but if you want to try without a fix, you can try running with `--pixelformat XRGB8888` :)

Abnormalpingu, 2023-02-17:
Allright, Thank you! :) 
I will try this alternative method until a real fix is in place (take your time!) 

Not sure if you're aware but there seems to be a version issue between Dart and flutter-pi - just rebuilt our app and i get this; 
[ERROR:flutter/shell/common/shell.cc(89)] Dart Error: Can't load Kernel binary: Invalid SDK Hash
I was stupid and updated stuff on my build machine, but i reckon this will be an issue for any new users of flutter-pi. I might be wrong though.

ardera, 2023-02-17:
np, I already have it almost done

That issue is actually mentioned in the readme: https://github.com/ardera/flutter-pi#building-the-asset-bundle
TL;DR if you use the 3.7.3 engine binaries on the target device, you also need to use the 3.7.3 SDK on your development machine. If there's a slight mismatch it won't work. If you upgrade your SDK without also upgrading the engine binaries, that won't work.

Maybe I should clarify that in the readme, because a lot of people seem to run into that problem

Abnormalpingu, 2023-02-20:
Great, I've solved the issue, I just didn't read closely enough, thanks for the tip! 

I've rebuild the app and tested the --pixelformat RGB8888 option and I still get the same message on the older hardware, so guess I'll just wait until you're done.

ardera, 2023-02-27:
Can you try with the `feature/gstreamer-video-player` branch? (it should work without the `--pixelformat` option there)

Abnormalpingu, 2023-02-28:
Hello again, 

I've tried doing this without much success, seems I'm unable to build the gstreamer-video-player branch correctly. I'm pretty sure it's something I'm doing that's wrong though. 

I deleted the old flutter-pi git cloned folder, then did; 

git clone -b feature/gstreamer-video-player  https://github.com/ardera/flutter-pi.git
cd flutter-pi/ 
mkdir build
cd build
cmake .. 

and I get this [error](https://pastebin.com/i4tzNcq3)

[CMakeError.log](https://pastebin.com/nPVpdYwa)
[CMakeOutput.log](https://pastebin.com/61hBVEGR)

Cmake is installed; 
apt install cmake:

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
cmake is already the newest version (3.18.4-2+deb11u1).
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.

Abnormalpingu, 2023-02-28:
Hi, sorry for the spam :(

As suspected it was something wrong with cmake in my end, I was able to fix it by installing the 'build-essential' package.

However, we seem to have a different issue when building it - see [here](https://pastebin.com/MwGiZFiU)

ardera, 2023-02-28:
Are you sure you're building the latest sources? That error was fixed by a commit some time ago

The lines in the error message look different now: https://github.com/ardera/flutter-pi/blob/8eb4ccd37f04b6a6257a14b7457e464e0ca635ab/include/pixel_format.h#L36-L50

Maybe try deleting & cloning again

Abnormalpingu, 2023-02-28:
Well, that explains it, I did in fact remove the old sources, then I cloned; 
git clone -b feature/gstreamer-video-player https://github.com/ardera/flutter-pi.git
Took a look at the source files and they are indeed not updated. 

I looked around at the other branches and found a recently updated one; 
git clone -b feature/gstreamer-player-from-pipeline https://github.com/ardera/flutter-pi.git

It builds allthough I'm getting a lot of warnings, but it completed, was able to 'make install', and what do you know, tried starting our flutter app and it works! 

Amazing work, thank you so much for the help and patience!

ardera, 2023-02-28:
Nice! Glad it works now!

Can you send me the warnings you're getting?

Abnormalpingu, 2023-03-02:
Absolutely, 
[here ](https://pastebin.com/hGuZftWH)are the warnings :)

ardera, 2023-03-07:
Thanks, I'll look into it

Abnormalpingu, 2023-04-24:
Hello again, 

so it seems there are more hardware variations than we first expected out in the wild from our client, so we seem to have some more issues with rendering, I'm hoping you're able to help us out again.. :) Let me know if you want me to create a seperate post for this and I'll delete this post. 

I am able to get kmscube running as before, I assume this is a good sign, that the issue is related to flutter-pi and not the hardware itself. I'll post the information you requested last time here; 

Driver details on drmdb - [here](https://drmdb.emersion.fr/snapshots/a9c49f555c6d)
dmesg.log - [here](https://pastebin.com/KrpHgC4H)
modetest output - [here](https://pastebin.com/3ZvDyhS5)

The main differences now is that this is an AMD Sempron 145 cpu w/Radeon R600 onboard gpu. We're still running debian 11 etc. The celeron hw is still working perfectly after your earlier fix. 

I've installed the libdrm-amdgpu1 instead of libdrm-intel1; 

'apt install libgl1-mesa-dev libgles2-mesa-dev libegl1-mesa-dev libdrm-amdgpu1 libdrm-dev libgbm-dev fontconfig libsystemd-dev libinput-dev libudev-dev libxkbcommon-dev' (and of course all the gstreamer stuff). 
I am building the flutter-pi client from the gstereamer-player-from-pipeline branch

Radeon requires firmware to work in linux, so I've added the contrib and non-free repositories and installed; 
'apt install firmware-amd-graphics'

Should you need anything else please let me know ! 

-Dan

ardera, 2023-05-16:
Oof, legacy KMS. That's pretty old :smile: 

Can you try the `feature/compositor-ng` branch, configure with `cmake ... -DCMAKE_BUILD_TYPE=Debug -DDEBUG_DRM_PLANE_ALLOCATIONS=On ...` and send me the output of that?

That'd help me see what's going on. Thanks!

Abnormalpingu, 2023-05-22:
Yeah, this stuff is ancient, it will be replaced with RPI's as soon as they are back in stock, seems pretty much impossible to be able to get any these days, especially when we need about a thousand of them.. 

Anyway, sorry for the delay we've had some days off due to our consitution day and a some days off :) 

I did the following; 

git clone --single-branch --branch feature/compositor-ng https://github.com/ardera/flutter-pi
made a build folder, cd'ed into that and did; 
cmake .. -DCMAKE_BUILD_TYPE=Debug -DDEBUG_DRM_PLANE_ALLOCATIONS=On
Build is completed OK

output from running our app, also this is being ran as root, just to avoid any permission issues temporarily. 

flutter-pi client/ > tmp.txt

[locales] Warning: The system has no configured locale. The default "C" locale may or may not be supported by the app.
[keyboard] Could not load keyboard configuration from "/etc/default/keyboard". Default keyboard config will be used. load_file: No such file or directory
[flutter-pi] WARNING: display didn't provide valid physical dimensions. The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
[flutter-pi] Could not create GBM Surface. gbm_surface_create_with_modifiers. Will attempt with gbm_surface_create: Function not implemented
[modesetting] Could not find a suitable unused DRM plane for pushing the framebuffer.
[compositor] Couldn't add fb layer.
[compositor] Could not present backing store. rendertarget->present: Input/output error
[modesetting] Could not commit display update. drmModePageFlip: No such file or directory
[compositor] Could not present frame. kms_req_commit_blocking: No such file or directory
[modesetting] Could not find a suitable unused DRM plane for pushing the framebuffer.
[compositor] Couldn't add fb layer.
[compositor] Could not present backing store. rendertarget->present: Input/output error
[modesetting] Could not commit display update. drmModePageFlip: No such file or directory
[compositor] Could not present frame. kms_req_commit_blocking: No such file or directory
^C

Thanks again for your help and patience !

ardera, 2023-05-23:
Can you try cloning and reconfiguring again? The output looks a bit different than what expected. I removed the `[modesetting]`, `[compositor]` style logging in https://github.com/ardera/flutter-pi/commit/fd8ee917691143ea306122d04de135a4dad61193 :)

So it should print stuff like `compositor.c: ...`, `modesetting.c: ...` now. Also not sure debug mode is enabled correctly, it should output some debug information for locale, egl, opengl and display mode at the start in debug mode.

I'm also not sure if `>` will capture stderr too. Can you try `&>` to capture both stdout and stderr? Thanks! (Or did you redirect to `tmp.txt` but send me the console output (not the contents of `tmp.txt`) instead?)

Abnormalpingu, 2023-05-25:
Hello again ! 

Allright, I've recloned and rebuild flutter-pi as before, also changed to  &> file.txt
I also double-checked if it output all the console text, which it does :) 
I'm not really sure if debug is enabled correctly, as before I do the following as before in flutter-pi\build\ - 
cmake .. -DCMAKE_BUILD_TYPE=Debug -DDEBUG_DRM_PLANE_ALLOCATIONS=On

Here's a paste of the output, hopefully this helps ! 

[locales] Warning: The system has no configured locale. The default "C" locale may or may not be supported by the app.
[keyboard] Could not load keyboard configuration from "/etc/default/keyboard". Default keyboard config will be used. load_file: No such file or directory
[flutter-pi] WARNING: display didn't provide valid physical dimensions. The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
[flutter-pi] Could not create GBM Surface. gbm_surface_create_with_modifiers. Will attempt with gbm_surface_create: Function not implemented
==============Locale==============
Flutter locale:
  default: C
  locales: C
===================================
===================================
display mode:
  resolution: 1280 x 1024
  refresh rate: 60Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
===================================
EGL information:
  version: 1.5
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_device EGL_EXT_platform_wayland EGL_KHR_platform_wayland EGL_EXT_platform_x11 EGL_KHR_platform_x11 EGL_MESA_platform_gbm EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_create_context_robustness EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 3.0 Mesa 20.3.5"
  shading language version: "OpenGL ES GLSL ES 3.00"
  vendor: "X.Org"
  renderer: "AMD RS780 (DRM 2.50.0 / 5.10.0-20-amd64, LLVM 11.0.1)"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_filter_anisotropic GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_compression_rgtc GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_float_linear GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_draw_instanced GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_AMD_performance_monitor GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_NV_conditional_render GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_pack_reverse_row_order GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_robustness GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_robustness GL_KHR_texture_compression_astc_ldr GL_NV_pixel_buffer_object GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_copy_image GL_EXT_draw_buffers_indexed GL_EXT_draw_elements_base_vertex GL_EXT_polygon_offset_clamp GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_KHR_robust_buffer_access_behavior GL_OES_copy_image GL_OES_draw_buffers_indexed GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_EXT_blend_func_extended GL_EXT_float_blend GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_EXT_clip_cull_distance GL_EXT_disjoint_timer_query GL_EXT_texture_compression_s3tc_srgb GL_MESA_shader_integer_functions GL_EXT_clip_control GL_EXT_color_buffer_half_float GL_KHR_parallel_shader_compile GL_EXT_EGL_image_storage GL_EXT_texture_sRGB_R8 GL_MESA_framebuffer_flip_y GL_EXT_depth_clamp "
===================================
flutter: The Dart VM service is listening on http://127.0.0.1:46481/zAmW-d4hwRo=/
flutter: Main.dart file:///home/test/
flutter: GAMEMANAGER.INITIATE
flutter: ApplicationMenuViewModel._initiate
flutter: Trying to get audio implementation platform linux
[... cut ...]
[modesetting] Could not find a suitable unused DRM plane for pushing the framebuffer.
[compositor] Couldn't add fb layer.
[compositor] Could not present backing store. rendertarget->present: Input/output error
[modesetting] Could not commit display update. drmModePageFlip: No such file or directory
[compositor] Could not present frame. kms_req_commit_blocking: No such file or directory
flutter: INFO: 2023-05-25 13:10:51.951977: WebSocket connected to 'ws://192.168.100.10:8101/clienthub?id=kiG3qIwwJkJ0AlpCS0aing'.
flutter: Connected to server

Abnormalpingu, 2023-05-26:
Doesn't look like it, but I've been known to do.. things :D 

git log from what i tried yesterday - gitlog1.txt
just tried using git with '-b' and not '--single-branch --branch' options - gitlog2.txt

[gitlog1.txt](https://github.com/ardera/flutter-pi/files/11572730/gitlog1.txt)
[gitlog2.txt](https://github.com/ardera/flutter-pi/files/11572731/gitlog2.txt)

ardera, 2023-05-30:
maybe you're somehow not executing the flutter-pi you've built? 😅 because that's definitely the flutter-pi master branch output

Abnormalpingu, 2023-05-31:
You're absolutely correct, I'll stop git cloning other branches directly from the client, it obviously doesn't work for some reason. 
I'll just download a zip and transfer it manually to the client to make sure in the future, sorry for the confusion :) 

Got the expected output now;
[log.txt](https://github.com/ardera/flutter-pi/files/11610832/log.txt)

ardera, 2023-06-01:
Okay, apparently the `feature/compositor-ng` branch had a bug and it didn't even reach to point where to problem ocurred previously. I fixed it now, can you try again with the latest commit?

And also record the dmesg log with:
```bash
echo 0x1FF | sudo tee /sys/module/drm/parameters/debug # Enable verbose DRM logging
sudo dmesg -C # Clear kernel logs
dmesg -w >dmesg.log & # Continuously write DRM logs to a file, in the background
flutter-pi ... # Run flutter-pi again, manually terminate it after a few errors have happened
fg # Kill dmesg with Ctrl+C
echo 0 | sudo tee /sys/module/drm/parameters/debug # Disable DRM logging
```

(and still configure with `-DCMAKE_BUILD_TYPE=Debug -DDEBUG_DRM_PLANE_ALLOCATIONS=On` too)

Thanks!

Abnormalpingu, 2023-06-06:
Hello again, 

so I've been testing out the nvidia based client, that seems to have some issues as well (shocker :D) 
One odd thing is that when running kmscube, it works fine, but when ctrl-c'ing the screen remains black, this might of course be unrelated to what we're trying to do. Just a strange occurence. 
When starting the flutter-pi client the screen goes black, but it runs, so not sure what's going on here. I though it might be the resolution but the test monitor is the same and the resolution is still 1280x1024 when flutter-pi starts up. I tried starting it in 1024x768 and the same thing happened. 

Main hw differences; 
AMD Athlon(tm) II X2 250 CPU and GeForce 7025 / nForce 630a (rev a2)

Installed libdrm-nouveau2 which is for nvidia gpus (instead of libdrm-intel1 and libdrm-radeon1)
Using the compositor-ng branch (actually I've copied the actual build files from the amd client to make sure it's the same, then rebuilt on the nvidia client with the debug flags and so on)

I'm unable to post the drm-info to the db-site as the clients lives in a airgapped network, but i just piped the output to a file instead, hopefully this is ok. 
[drm_info.txt](https://github.com/ardera/flutter-pi/files/11663944/drm_info.txt)
[modetest.txt](https://github.com/ardera/flutter-pi/files/11663949/modetest.txt)
[dmesg.log](https://github.com/ardera/flutter-pi/files/11663951/dmesg.log)

This should be the last client we need, as these are the most numerous in the existing deployment, thanks again for your patience and help!

ardera, 2023-06-10:
Ah, I know what causes this. Flutter-pi tries to use ARGB but your GPU only supports XRGB for scanout.
Can you try if it works with latest `feature/compositor-ng`?

Abnormalpingu, 2023-06-12:
Hi, 

Interesting, I downloaded the latest compositor-ng (manually) and built it on the client (with the debug flags) and I could for a split second see graphics, then it goes back to black, system is not crashing or hanging, but needs to be rebooted to regain a video output. So it's absolutely a step in the right direction :) 
I also tried starting flutter-pi with the different XRGB-pixelformats manually without any difference (worth a shot) 

I've attached the dmesg logfile in case you'd need it

From the console I see some relevant text too; 
/home/test/flutter-pi/src/gl_renderer.c: Choosing EGL config with pixel format ARGB 8:8:8:8...
flutter: The Dart VM service is listening on http://127.0.0.1:38917/toBTcVMAOwE=/
flutter: Main.dart file:///home/bingo/
flutter: GAMEMANAGER.INITIATE
flutter: ApplicationMenuViewModel._initiate
flutter: Trying to get audio implementation platform linux
flutter: AudioImplementationLinuxLibao
ALSA lib pcm_hw.c:1829:(_snd_pcm_hw_open) Invalid value for card
couldn't open play str
[dmesg.log](https://github.com/ardera/flutter-pi/files/11720737/dmesg.log)
eam: No such file or directory
flutter: Automuting sound on startup to avoid mental issues
/home/test/flutter-pi/src/modesetting.c:   checking if plane with id 37 qualifies...
/home/test/flutter-pi/src/modesetting.c:     does qualify.
flutter: ConnectionState HubConnectionState.Connecting
../../third_party/skia/src/gpu/ganesh/SurfaceDrawContext.cpp:2110: fatal error: "assert(copy)"
Illegal instruction

ardera, 2023-06-12:
> ../../third_party/skia/src/gpu/ganesh/SurfaceDrawContext.cpp:2110: fatal error: "assert(copy)"
> Illegal instruction

That seems like a GL driver problem :/
Only way to debug this would be to build the flutter engine with debug symbols and step through the relevant skia code. I'm currently building the 3.10.4 engine with debug symbols here: https://github.com/ardera/flutter-ci/actions/runs/5242452070?pr=5

Abnormalpingu, 2023-06-14:
Gotcha, well, we've tried building the engine-binaries for x64 and we seem to have issues getting this to build correctly.. 
Would it be possible for you to build these binaries for x64 and provide them to us maybe? If you should need somewhere to store them we can provide somewhere to do that (or whatever works for you). 
Just for our temporary testing of course, we'd look further into building our own at a later point, just to save some time in our troubleshooting efforts. 

We also upgraded to flutter 3.10.4 and rebuilt our app. 
When we build the engine we only find one libflutter_engine.so file, I see in your builds you get them with .debug, .release and so forth. We can also find the flutter_embedder.h which we tried installing, as well as renaming the so-files. Not sure if when you build the engine the output is the same as the folder/file-structure of the git or you pick out the files manually and push them.

ardera, 2023-06-14:
Sure, here: https://1drv.ms/u/s!AsNHmDqMr-aVg65exuQp7lRxiSmsuA?e=Lsh22W

It's built with:
```
./src/flutter/tools/gn \
  --runtime-mode debug \
  --embedder-for-target \
  --disable-desktop-embeddings \
  --no-build-glfw-shell \
  --no-build-embedder-examples \
  --no-goma \
  --unoptimized --no-stripped
```

and some custom modifications so it's built with `-ggdb` and `-fdebug-default-version=4` (some gdbs have issues with DWARF v5) instead of `-g2`:
```diff
diff --git a/build/config/compiler/BUILD.gn b/build/config/compiler/BUILD.gn
index 4c9a1ee..b74ab87 100644
--- a/build/config/compiler/BUILD.gn
+++ b/build/config/compiler/BUILD.gn
@@ -951,7 +951,7 @@ config("symbols") {
     }
     ldflags = [ "/DEBUG" ]
   } else {
-    cflags = [ "-g2" ]
+    cflags = [ "-ggdb", "-fdebug-default-version=4" ]
   }
 }
```