ardera/flutter-pi #316: Mouse cursor isn't visible

zverbeta, 2023-02-16:
Hi. Thank you for your work!

Recently i used this library for my setup (rpi3 + blu-mouse + projector), but mouse cursor isn't visible. 

i think you should always display the mouse cursor, but display the cursor in the upper right corner (visual invisible), for example

https://github.com/ardera/flutter-pi/blob/ca62469570/src/user_input.c#L343

I'm not familiar with c code, so i can't propose pr

ardera, 2023-03-07:
Hey, can you try the `feature/gstreamer-player-from-pipeline` branch?

zverbeta, 2023-03-07:
I tried, in the current branch the cursor is displayed in the upper corner of the screen, but now it does not react to the movements of the mouse.

Last time I was able to click on buttons without a visible cursor ;)

ardera, 2023-03-07:
Ah, I have an idea what's wrong. I think the cursor is only shown at the new position when flutter draws a new frame.

zverbeta, 2023-03-07:
More interestingly, if you run an example from your video player library (https://pub.dev/packages/flutterpi_gstreamer_video_player), the cursor starts to move.

frederaik86, 2023-05-06:
Hi, this bug has been fixed or is there a workaround? 
I've had this issue this morning after update.

Thanks

ardera, 2023-08-22:
fixed by #323

closed 2023-08-22 by ardera