ardera/flutter-pi #317: Please add possibility to check engine version

eximius313, 2023-02-18:
Because every Flutter version must match exact version of the Engine, it is really hard to keep track which RPI has which version, if one has plenty of RPi operating.
Could you please add a command like `flutter-pi --engine-version` which displays the current engine version?

tobiasht, 2023-03-14:
That´s a great idea. I guess it could be as simple as looking up the hash code to get it´s corresponding version number. So when you get one of the following errors `Invalid kernel binary format version`, `Invalid SDK hash` or `Invalid engine hash` you would also get a message with the correct version.

ardera, 2023-03-24:
Showing the installed engine version is possible with some changes to the engine binaries installer, but showing the correct engine version for a flutter app is probably impossible.

But I think showing the installed engine version is still useful enough, so I'll implement that

eximius313, 2023-03-24:
`Showing the installed engine version is possible with some changes to the engine binaries installer, but showing the correct engine version for a flutter app is probably impossible.`

I think that the former should be enough.

`But I think showing the installed engine version is still useful enough, so I'll implement that`

That is awesome! Thank you

`Also, it's possible to bundle the flutter engine with an asset bundle, instead of installing it systemwide on a target machine. That'll probably result in less version mismatches. Currently, that's a bit harder to use than installing systemwide, but in the future, with some changes to the flutter SDK, that might be easier.`

How to do that?

By the way - today I was upgrading `flutter-pi` and I wanted to check the current version with `flutter-pi --version` and I've learnt that it is not possible either.

ardera, 2023-03-24:
> `Also, it's possible to bundle the flutter engine with an asset bundle, instead of installing it systemwide on a target machine. That'll probably result in less version mismatches. Currently, that's [...]

Just put the fitting `libflutter_engine.so` into the asset bundle directory (without `.debug` or `.release` suffix, just `libflutter_engine.so`) - flutter-pi will first attempt to load the engine from there and only fall back to the systemwide flutter engine if it doesn't a flutter engine there. (Not really documented, I know)

> By the way - today I was upgrading `flutter-pi` and I wanted to check the current version with `flutter-pi --version` and I've learnt that it is not possible either.

Right, yeah maybe some versioning mechanism there would be useful as well.