ardera/flutter-pi #319: Update ReadMe with how to build the app.so on Mac (release mode)

tobiasht, 2023-03-14:
Currently the instructions for how to build the app.so is only for Windows. Would be great if the Mac equivalent was present as well.

ardera, 2023-03-24:
I'm not sure that's possible. I haven't yet found a way to build the `gen_snapshot` for MacOS. The flutter engine doesn't support it. It seems like the Dart SDK doesn't either.

Other than that I don't have a MacOS machine to test the instructions on. Maybe there's some way to run linux binaries on MacOS, if yes I'd probably use that.

Though, if you have a mac and find a way to build the `app.so`, I'd be happy to include them in the readme!

toontoet, 2025-03-27:
```
flutterpi_tool build --release --arch=arm64
```
works on my recent Mac.