ardera/flutter-pi #321: Weird behaviour of `String.fromEnvironment` and `--dart-define` with `flutter-pi --release`

eximius313, 2023-03-24:
Having such application:
```dart
const constEnv = String.fromEnvironment('MY_ENV', defaultValue: 'FOO'); 

void main() async {
    final env = String.fromEnvironment('MY_ENV', defaultValue: 'BAR');
    print('TEST ENV:');
    print('constEnv: >$constEnv<');
    print('env: >$env<');
}
```

And building it with:
`flutter build bundle --dart-define=MY_ENV=BAZ`

When I run it with: `flutter-pi my_app`
Then I get:
```
flutter: TEST ENV:
flutter: constEnv: >BAZ<
flutter: env: >BAR<
```
It is nice, that `constEnv` get initialized correctly, but why the other one is not?

It is getting even worse with `flutter-pi --release my_app`
then, I get:
```
flutter: TEST ENV:
flutter: constEnv: >FOO<
flutter: env: >BAR<
```
so any value is initialized correctly

ardera, 2023-03-24:
`const` means the value is known at compile-time. For debug (JIT) mode, the compile-time is the time you run flutter-pi. For release and profile (AOT) mode, that's the time you run `gen_snapshot`. If you didn't specify `-DMY_ENV=BAZ` to `gen_snapshot`, the dart compiler will assume the default value for any `const` usages for `String.fromEnvironment('MY_ENV', ...)`.

The `final` usage probably doesn't work because it's non-const. The documentation says `String.fromEnvironment` is only guaranteed to work when used as `const`: https://api.flutter.dev/flutter/dart-core/String/String.fromEnvironment.html.
It says it might work, and I'm not sure of the exact reason it doesn't work here, but seems like that's within spec

eximius313, 2023-03-24:
> const means the value is known at compile-time. For debug (JIT) mode, the compile-time is the time you run flutter-pi. For release and profile (AOT) mode, that's the time you run gen_snapshot. If [...]

In my examples, no matter if I run it as `flutter-pi my_app` or `flutter-pi --release my_app`, I always build it like this:
```
FLUTTER=/c/User/flutter/bin/cache/

flutter build bundle --dart-define=My_ENV=BAZ

$FLUTTER/dart-sdk/bin/dart.exe \
 $FLUTTER/dart-sdk/bin/snapshots/frontend_server.dart.snapshot \
 --sdk-root $FLUTTER/artifacts/engine/common/flutter_patched_sdk_product \
 --target=flutter \
 --aot \
 --tfa \
 -Ddart.vm.product=true \
 --packages $packages \
 --output-dill build/kernel_snapshot.dill \
 --verbose \
 --depfile build/kernel_snapshot.d \
 package:my_app/main.dart
```

Does it mean that I provide MY_ENV to `gen_snapshot` or not? I'm not sure if I understand you well.

> The final usage probably doesn't work because it's non-const. The documentation says String.fromEnvironment is only guaranteed to work when used as const: https://api.flutter.dev/flutter/dart-core/String/String.fromEnvironment.html.
It says it might work, and I'm not sure of the exact reason it doesn't work here, but seems like that's within spec

ok, thanks

ardera, 2023-03-25:
> In my examples, no matter if I run it as `flutter-pi my_app` or `flutter-pi --release my_app`, I always build it like this:
> 
> ```
> FLUTTER=/c/User/flutter/bin/cache/
> 
> flutter build bundle --dart-define=MY_ENV=BAZ
> 
> $FLUTTER/dart-sdk/bin/dart.exe \
>  $FLUTTER/dart-sdk/bin/snapshots/frontend_server.dart.snapshot \
>  --sdk-root $FLUTTER/artifacts/engine/common/flutter_patched_sdk_product \
>  --target=flutter \
>  --aot \
>  --tfa \
>  -Ddart.vm.product=true \
>  --packages $packages \
>  --output-dill build/kernel_snapshot.dill \
>  --verbose \
>  --depfile build/kernel_snapshot.d \
>  package:my_app/main.dart
> 
> wsl ../engine-binaries/gen_snapshot_linux_x64_release \
>  --deterministic \
>  --snapshot_kind=app-aot-elf \
>  --elf=build/flutter_assets/app.so \
>  --strip \
>  build/kernel_snapshot.dill
> ```

I think the `--dart-define` option there is actually a no-op, just like the `--debug`, `--profile` and `--release` options. Or they aren't no-ops, they still do something, but they have no effect on the generated asset bundle.

> Does it mean that I provide MY_ENV to `gen_snapshot` or not? I'm not sure if I understand you well.

Sorry, no, not `gen_snapshot`, but to the frontend server. (The second command you listed in the code block) `-Ddart.vm.product=true` [is a dart define too](https://api.flutter.dev/flutter/foundation/kReleaseMode-constant.html). So the whole command could look like this:

```bash
$FLUTTER/dart-sdk/bin/dart.exe \
 $FLUTTER/dart-sdk/bin/snapshots/frontend_server.dart.snapshot \
 --sdk-root $FLUTTER/artifacts/engine/common/flutter_patched_sdk_product \
 --target=flutter \
 --aot \
 --tfa \
 -Ddart.vm.product=true \
 -DMY_ENV=baz \
 --packages $packages \
 --output-dill build/kernel_snapshot.dill \
 --verbose \
 --depfile build/kernel_snapshot.d \
 package:my_app/main.dart
```

eximius313, 2023-04-28:
@ardera I've done some tests, and come up with surprising results.

My build command is now:
```
flutter build bundle --dart-define=MY_ENV=MODE1 &&\            <-- one version

$FLUTTER/dart-sdk/bin/dart.exe \
 $FLUTTER/dart-sdk/bin/snapshots/frontend_server.dart.snapshot \
 --sdk-root $FLUTTER/artifacts/engine/common/flutter_patched_sdk_product \
 --target=flutter \
 --aot \
 --tfa \
 -Ddart.vm.product=true \
 -DMY_ENV=MODE2 \              <-- second version
 --packages $packages \
 --output-dill build/kernel_snapshot.dill \
 --verbose \
 --depfile build/kernel_snapshot.d \
 package:my_app/main.dart
```
and my main looks like this:
```dart
const _env = String.fromEnvironment('MY_ENV', defaultValue: 'FOO'); 

void main() async {
  WidgetsFlutterBinding.ensureInitialized();
  print('\n\nTESTING ENV: $_env');
}
```

Now - when I run the code with: `flutter-pi my_app`, I get `MODE1` so `--dart-define` is taken into account and `-DMY_ENV` is ignored (because I verified, that even when `--dart-define` is missing, then `-DMY_ENV` is still not taken into account), but when I run exactly the same code with `flutter-pi --release my_app`, it's exactly the opposite - I get `MODE2`, so `-DMY_ENV` is taken into account and `--dart-define` is completly ignored in `--release`

Is it desired behaviour?

ardera, 2023-05-13:
AFAICT, that is desired behaviour.
flutter-pi has absolutely no influence on that anyway, so even if it wasn't desired behaviour, I wouldn't be able to change it. That's all handled by the dart vm/sdk internally.

ardera, 2023-05-16:
Closing for now. If there's any questions remaining, feel free to reopen or create a new issue :)

closed 2023-05-16 by ardera