ardera/flutter-pi #322: Get sound decibel of microphone

BigGitWorld, 2023-03-27:
Hello.
How can I measure the decibel level of a microphone connected to a Raspberry Pi?

BigGitWorld, 2023-07-05:
No help?

ardera, 2023-07-21:
That's definitely something you'll have to write a plugin for, unfortunately it's not possible in pure dart (except if you want a hacky solution)

For the implementation, I'd recommend you use gstreamer and the `level` plugin to detect the noise level.
- examples on how to use gstreamer is already in [`src/plugins/gstreamer_video_player/player.c`](https://github.com/ardera/flutter-pi/blob/feature/compositor-ng/src/plugins/gstreamer_video_player/player.c)
- the `level` plugin also has some examples: https://gstreamer.freedesktop.org/documentation/level/index.html?gi-language=c#level
- some flutter-pi plugin examples are here: https://github.com/ardera/flutter-pi/blob/feature/camera-plugin/src/plugins/camera/plugin.c

As a proof of concept, this here will log the noise level of some auto-detected microphone to console (look at the `message: RMS ...dB` logs):
```bash
GST_DEBUG=level:7 gst-launch-1.0 autoaudiosrc ! audioconvert ! level post-messages=true ! fakesink sync=true
```

If you want a really _really_ hacky solution you can just run that with normal dart `Process.start` and use regexes to extract the noise level from the process output.