ardera/flutter-pi #324: How to make it run even if no screen is connected

moham96, 2023-04-16:
If the screen is not powered on then flutter-pi fails with this:

```
[flutter-pi] Device "/dev/dri/card0" doesn't have a display connected. Skipping.
[flutter-pi] Could not find a connected connector!
```
In my case I can't guarantee that the screen will be on by the time `flutter-pi` is ran so how can I make flutter-pi run headless if no screen is connected/powered on and then maybe adjust after a screen is on?

Previously my app was written in `pyqt` and the pi was using the old framebuffer driver where I had to force a specific HDMI mode so that it works if the screen is connected or not, now I have rewritten my app in flutter and I'm stuck

moham96, 2023-04-19:
Using fkms driver and setting `hdmi_force_hotplug=1` seems to work and the app is able to start at boot even when no screen is connected.and apparently adjust the resolution automatically when a screen is later connected.
But does this have any drawback?
Is the fkms driver well supported by flutter-pi?
Is it possible to achieve the same results with `kms` driver?

ardera, 2023-04-20:
> But does this have any drawback?

The fkms driver showed some flickering in the past (even though vsync is enabled), at least on DSI. Not sure that's still the case

> Is the fkms driver well supported by flutter-pi?

Yep, same as the KMS one

> Is it possible to achieve the same results with kms driver?

Don't think so I'm afraid, flutter itself doesn't support display hotplugging or multi-display, so any solution would require flutter-pi to fake a display itself. That's quite hacky and also requires some fundamental modesetting changes but the benefit is pretty small. I'll rather wait till flutter has true multi-display & display hotplugging support (that's WIP at the moment, but you can already see some of the work in flutter stable) and then implement it using that

FlorianAlt, 2023-06-09:
I had the same issue and was able to get this running by setting this options:

```
hdmi_force_hotplug=1

hdmi_group=1
hdmi_mode=76
hdmi_force_mode=1

dtoverlay=vc4-fkms-v3d
```

moham96, 2025-05-04:
@ardera is there an update to this issue?
Did flutter/flutter-pi gain the ability to hot plugging/Multi display?

ardera, 2025-05-05:
@moham96 Multi-display no, but to run without a display you can use `--dummy-display`. To customize the simulated display size you can use `--dummy-display-size=width,height`

ardera, 2025-05-05:
@moham96 That would require multi-display / hotplugging functionality (as you correctly noted); what's your use-case? Would it also be possible to start the flutter-pi app on display connect?

moham96, 2025-05-05:
> [@moham96](https://github.com/moham96) That would require multi-display / hotplugging functionality (as you correctly noted); what's your use-case? Would it also be possible to start the flutter-pi app on display connect?

I am using flutter-pi to run a turnstile entry based on RFID, the device is connected to a big screen that shows various information, sometimes the device is turned on but the display is turned off, this prevents the app from starting and no entry is allowed, i would like to keep partial functionality of the app (verifying rfid and opening the gate) while the display is turned off and when the display is turned on it should gain the full functionality of entry and displaying information information

moham96, 2025-05-07:
@ardera can you introduce a workaround like this:
Introduce a  flag named "--conditional-dummy-display" or make the "--dummy-display" flag take a value l(e.g "conditional" or  "always") to define which mode of operation to use.
If the mode is "conditional" then flutter-pi checks if the device has a display connected when starting up, if so it will ignore 'dummy-display" flags and just start on that display, but it find no display it will create a dummy display using the "--dummy-display-size" flag.

This can probably be implemented using a warpper script around flutter-pi that checks `/sys/class/drm` path for displays and conditonaly modify flutter-pi's flags, but it would be nice to have it built into flutter-pi untill we have true hot plugging support