ardera/flutter-pi #325: Engine binaries not found on the Pi

Binibeno, 2023-04-22:
I am facing an issue with flutter-pi on a fresh OS install on a Raspberry Pi 4 Model B Rev 1.4. I have installed Flutter 3.7.11 (stable) and the matching arm engine binaries. I followed the steps in the README (no errors in the process), however when I try to run flutter-pi, I receive the following error: `"[flutter-pi] Error: Could not load flutter engine from any location. Make sure you have installed the engine binaries."`

I've checked the install script of the `flutter-engine-binaries-for-arm` repo, and it does succeed, the files get copied into /usr/lib and /usr/include.

I have checked all the necessary commands and done a fresh installation, but the problem persists. The `uname -m` command returns aarch64 (32-bit OS).

I would appreciate any help to get flutter-pi working correctly. Let me know if you need any additional information.

ardera, 2023-04-23:
Can you run with `LD_DEBUG=libs flutter-pi ...` and send me the output?

> I have checked all the necessary commands and reinstalled the os, but the problem persists. The uname -m command returns aarch64 (32-bit OS).

If you're running a 32-bit OS, `uname -m` should print `armv7l`, that's what it prints for me

Binibeno, 2023-04-23:
Thanks for the help!

[output.txt](https://github.com/ardera/flutter-pi/files/11303437/output.txt)
 
I've attached the output. 

Weird,  uname -m prints out "aarch64" for me, however I installed rapsberry pi os 32bit (picture attached from pi imager). 

[image]

Should I try to overwrite the install script in [flutter-engine-binaries-for-arm](https://github.com/ardera/flutter-engine-binaries-for-arm) to install binaries from the 'arm' directory as opposed to the 'arm64' directory?

ardera, 2023-04-23:
that's strange, maybe they changed something

> Should I try to overwrite the install script in [flutter-engine-binaries-for-arm](https://github.com/ardera/flutter-engine-binaries-for-arm) to install binaries from the 'arm' directory as opposed to the 'arm64' directory?

yep you can try that. (or just manually copy the arm32 binaries into `/usr/lib`)

You can also try configuring & building with `-DCMAKE_BUILD_TYPE=Debug`, flutter-pi will print more info on engine load failure in that case

Binibeno, 2023-04-23:
Good news! I've installed the binaries from the 'arm' folder as opposed the 'arm64', and it works without the problem. 

Thanks for the help!

closed 2023-04-23 by Binibeno