ardera/flutter-pi #327: symbol lookup error: flutter-pi: undefined symbol: eglDestroyImage

felix2049, 2023-05-07:
I got an error message when I was running on [RK3588](https://www.rock-chips.com/a/en/index.html). How to solve it? 

flutter-pi /home/myhome/Documents/flutter_gallery .
flutter-pi: symbol lookup error: flutter-pi: undefined symbol: eglDestroyImage

ardera, 2023-05-16:
Can you try the `feature/compositor-ng` branch?

felix2049, 2023-05-18:
> 

Many thanks!
It Works!  !

Firstly I try compile with "gstreamer video player" related packages, failed with compile error. 
Then use apt remove them all, compile successed. 
Does it will be fixed and merged to master soon?

ardera, 2023-06-10:
Not sure how soon, but i will be merged to master yes.

What's the compile error when compiling with the gstreamer video player packages?

closed 2023-07-01 by github-actions[bot]