ardera/flutter-pi #333: wrong touch position when using orientation argument

AndroidDesigner, 2023-05-28:
Hello
I want to show my app as landscape, so I used ``sudo flutter-pi --release -o landscape_left /path-to-app`` to run my app.
But I don't know why touch positions are wrong in the landscape mode!
for example, suppose I have a button in bottom-left of the screen. So if I click on it, nothing happens while if I click on bottom-right side of the screen, the button's event triggers.
what is the reason?

closed 2023-06-01 by AndroidDesigner

ardera, 2023-06-01:
Nice that you got it working!

flutter-pi uses `libinput`, but other than that it's correct :) I also have [flutter_libinput_calibrator](https://github.com/ardera/flutter_libinput_calibrator) for this, but specifying an exact rotation matrix like you did is better

AndroidDesigner, 2023-06-02:
The [flutter_libinput_calibrator](https://github.com/ardera/flutter_libinput_calibrator) is so useful too.
Thank you.