ardera/flutter-pi #335: Matching SDK/Engine

zmeggyesi, 2023-06-02:
When trying to run a test application, I got the error message `Dart Error: Can't load Kernel binary: Invalid SDK hash.` - the problem is fairly clear.

What is less clear is the _correct_ SDK hash/engine version. I'm okay with hacking about in the SDK to set it to the correct version, but because I can't run `flutter --version` on the Raspi - since there's no _actual_ Flutter installation - I don't know what to set it to.

Would it be possible to have the embedder print the version and has of the engine/SDK it's using to make matching an easier task?

P.S.: the `#custom-embedders` channel seems to be gone from the Flutter Discord...

eximius313, 2023-09-23:
seems similar to: https://github.com/ardera/flutter-pi/issues/317

ardera, 2023-09-24:
@eximius313 True, thanks.

Closing this for now. See #317 instead

closed 2023-09-24 by ardera