ardera/flutter-pi #338: [fs layout] App bundle directory "(null)" does not exist.

amirazizi4861, 2023-06-12:
Hi there,
I've already build for release but when i want to run flutter-pi the following message pop up:
`[fs layout] App bundle directory "(null)" does not exist.
`

amirazizi4861, 2023-06-12:
I've already tried to reinstall flutter-pi using the latest version hoping it would solve the issue but it still happen. Please help.

ardera, 2023-06-12:
Can you paste the commandline you're using to start flutter-pi? I.e. `flutter-pi ... ~/my-app`

amirazizi4861, 2023-06-12:
`sudo flutter-pi -o portrait_down --release  /home/pi/frontend_amir`
Hi there, this is the command that i used.
I've managed to resolve the issue by upgrading both engine-binaries to the latest one.
Thank you.

ardera, 2023-06-12:
Glad you could resolve it, but tbh it seems unlikely this is related to the engine binaries. I'm not even sure how the `NULL` value made it into the app bundle directory variable so it's printing out `(null)` there.

> my app seems to crash when running with the latest engine which it didnt happen when i was using 3.7.11 version.

You mean a different crash than the `App bundle directory "(null)" does not exist` crash, right? Can you please paste the complete commandline output for the new crash?

closed 2023-07-03 by github-actions[bot]