ardera/flutter-pi #343: Flutter engine binaries for engine 987b621eac4e1cbe12894d35127e3e80869591f2 are not available.

Teddywonseokyoo, 2023-07-04:
hello.

I am getting the following error while building the example. 

flutter pub global run flutterpi_tool build --arch=arm64 --cpu=pi4 --release
Building package executable... (1.2s)
Built flutterpi_tool:flutterpi_tool.
Flutter engine binaries for engine 987b621eac4e1cbe12894d35127e3e80869591f2 are not available.

The flutter version is

Flutter 3.12.0-13.0.pre.40 • channel master • https://github.com/flutter/flutter.git
Framework • revision ddb16b7d92 (13 hours ago) • 2023-07-03 14:27:33 -0400
Engine • revision 987b621eac
Tools • Dart 3.1.0 (build 3.1.0-262.0.dev) • DevTools 2.25.0

thank you.

ardera, 2023-07-04:
Hey,
for flutterpi_tool only stable channel (so currently 3.10.5) is supported.

The flutterpi_tool downloads the binaries from: https://github.com/ardera/flutter-ci/releases
and I only build them for stable there.

Teddywonseokyoo, 2023-07-04:
Thanks for the quick reply.

closed 2023-07-04 by Teddywonseokyoo