ardera/flutter-pi #344: Check audio plugin against audioplayers v4

DoumanAsh, 2023-07-05:
This is sort of remainder to do.
There are a lot of changes in audioplayers, including FFI part.

I will need to get myself a raspberry pi again and then I'll start checking it

DoumanAsh, 2023-07-06:
audioplayers 4.0.0 is not compatible with plugin:

```
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: MissingPluginException(No implementation found for method create on channel xyz.luan/audioplayers)
#0      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:308)
<asynchronous suspension>
#1      AudioPlayer._create (package:audioplayers/src/audioplayer.dart:148)
<asynchronous suspension>
```

There is also incompatibilities with older version:

```
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(illegalargument, Expected `arg['is_local']` to be a bool., null, null)
#0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310)
<asynchronous suspension>
#2      AudioPlayer.play (package:audioplayers/src/audioplayer.dart:131)
<asynchronous suspension>
#3      TicketsProvider.callTicket (package:panel_fe/providers/tickets_provider.dart:85)
<asynchronous suspension>
```

bojidartonchev, 2023-10-17:
Hello @DoumanAsh,

Do you have info what is the latest supported version atm?

DoumanAsh, 2023-10-17:
@bojidartonchev my branch `audioplayers_v4` should work with any `4.*.*` but I never had time to work on testing it after great refactoring in flutter-pi
Branch is rebased, but not well tested yet

DoumanAsh, 2023-10-18:
@bojidartonchev Got it, probably some issue after rebase, although it is weird that resume is not called at all
Sorry I never had time to finalize it after rebasing, I will find some time this weekend to flash fresh OS image on my raspberry pi and test it again

Btw did it have sound when you played via `gst-launch-1.0`?
I only used audio jack and never bothered to configure audio devices properly, not sure what is your setup

bojidartonchev, 2023-10-19:
@DoumanAsh Yes, I have sound with `gst-launch-1.0`.

DoumanAsh, 2023-10-22:
@bojidartonchev I think dudes responsible for `audioplayers` changed something again
I just tested and it fails with timeout error (as if no response is received) even though response is sent back
I will try to figure out what changed and maybe just re-write code to align with latest version

bojidartonchev, 2023-10-23:
@DoumanAsh That would be great, thanks.

I have tried with latest version and the issue is the same as with `4.x.x`.

DoumanAsh, 2023-10-23:
@bojidartonchev Latest version is aligned with `5.*` but because I tested it against that version you cannot expect it to work for now.
I'm pretty sure API wise I covered all changes with my last[commit](https://github.com/ardera/flutter-pi/pull/345/commits/58dfcaf210a9fb046b15f0cddf73dbb5c285b02c)

Unfortunately it doesnt work which sucks and I suspect there is something wrong with how I use flutter channels when replying back...
Not clear what yet, I will try to figure it out

bojidartonchev, 2023-10-23:
@DoumanAsh Found the issue.

In `audio_player_on_media_state_change` there is a check by the source object by name

`if (streq(GST_OBJECT_NAME(src), "playbin")) {`

but the playbin source's name is actually `playbin0`. So we can either set a name of the playbin element and check by name, or we can check this way:

`if (src == GST_OBJECT(self->playbin)) {`

DoumanAsh, 2023-10-23:
Ok this is interesting, because I'm pretty sure it worked like that in past.
I will change it as per your recommendation then and will test it later after work.

If you have time feel free to test me and let me know if there are any other issues.

I will need to run under ASAN and fix all possible issues in flutter-pi code. I found one minor problem outside of plugin

bojidartonchev, 2023-10-24:
[This](https://github.com/DoumanAsh/flutter-pi/commit/d4c44ce5bdf4ca73a39752e761d4fdcb84782383#diff-da34e6bd9767081da8520c288e2f7bea4f717b12dfaa7c88df511716c102727dL76-R101:~:text=self%2D%3Eplaybin%20%3D,playbin%22%2C%20NULL)%3B) change is the reason.

DoumanAsh, 2023-10-24:
oooh I see
that's unexpected, I probably didn't follow though refactoring for new version.
Good catch, thank you

DoumanAsh, 2023-10-28:
@bojidartonchev In case you're working on more extensive audio usage, please try my branch in its current state to see if there is any problems with it.
I've been doing some simple testing my side, but it would be good to have someone else to cross check it just in case

closed 2024-01-15 by ardera