ardera/flutter-pi #348: Allow Non-interactive

maxime-guerin-biprep, 2023-08-01:
Dart is not compatible with ARMv7 since version 2 (i think) but this repo allow us to use dart with a more recent flutter version.

I would like to know if its possible to build an flutter app so that there no UI to allow dart program to run in ARMv7 computer, in my case a steamlink.

I am currently trying to compile flutter-pi for the steamlink

ardera, 2023-08-01:
Dart is in fact compatible with armv7[^1], but it's not with armv6, maybe you meant that?

What are you trying to do? Do you need any flutter specific features or would pure dart (no flutter) work for you too?
I think a non-interactive mode is a good idea for flutter-pi and I have that planned. It's also useful for headless testing. But if you're on armv7 and don't need anything specific to flutter, you can just use vanilla dart.

[^1]: see linux releases here: https://dart.dev/get-dart/archive#stable-channel

maxime-guerin-biprep, 2023-08-01:
Yes maybe, i firstly has this probleme with the raspberry zero 1.
In the end i changed my dart program so it can work with the version 2.0.0-dev.50.0, because i found it precompiled.
For that case a pure dart version would work, it is more of a dart version problem, because i firstly did it with the version 3++.
In the end i made it work but maybe a headless flutter version could have make it easier ^^