ardera/flutter-pi #349: Can't build for arm64 on raspberry pi 4b

Rob-bb, 2023-08-05:
Hi, just trying flutter-pi out, I am building on the raspberry pi 4b running raspberry pi os bullseye.

` $ flutter --version
Flutter 3.10.6 • channel stable • https://github.com/flutter/flutter.git
Framework • revision f468f3366c (3 weeks ago) • 2023-07-12 15:19:05 -0700
Engine • revision cdbeda788a
Tools • Dart 3.0.6 • DevTools 2.23.1
`

The project is the generic one from "flutter create projectname" 

This seems weird that it can't build for arm64.  What am I doing wrong?

` $ flutterpi_tool build --release --arch=arm64
Unhandled exception:
Unsupported operation: Unsupported host platform: HostPlatform.linux_arm64
#0      FlutterpiArtifactPaths.getHostDirName (package:flutterpi_tool/src/build_bundle.dart:171:12)
#1      FlutterpiArtifactPaths.getEngineSource (package:flutterpi_tool/src/build_bundle.dart:227:25)
#2      CopyFlutterpiEngine.inputs (package:flutterpi_tool/src/build_bundle.dart:575:24)
#3      Target.resolveInputs (package:flutter_tools/src/build_system/build_system.dart:195:34)
#4      Target._toNode (package:flutter_tools/src/build_system/build_system.dart:148:39)
#5      Target._toNode (package:flutter_tools/src/build_system/build_system.dart:155:58)
#6      FlutterBuildSystem.build (package:flutter_tools/src/build_system/build_system.dart:603:30)
#7      buildFlutterpiBundle (package:flutterpi_tool/src/build_bundle.dart:892:36)
#8      BuildCommand.run.<anonymous closure> (package:flutterpi_tool/src/build_bundle.dart:1179:17)
<asynchronous suspension>
#9      AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
<asynchronous suspension>
#10     BuildCommand.run (package:flutterpi_tool/src/build_bundle.dart:1166:5)
<asynchronous suspension>
#11     CommandRunner.runCommand (package:args/command_runner.dart:212:13)
<asynchronous suspension>
#12     main (package:flutterpi_tool/src/build_bundle.dart:1269:16)
<asynchronous suspension>
#13     main (file:///home/rob/.pub-cache/hosted/pub.dev/flutterpi_tool-0.1.2/bin/flutterpi_tool.dart:4:3)
<asynchronous suspension>
`

ardera, 2023-08-05:
Hey, currently flutterpi_tool is not supported on arm64. Only linux x64 is supported atm. (Note that's the host platform, i.e. the one you're running tje tool on, not the target platform)

Rob-bb, 2023-08-06:
Thanks for the info.  Not really a bug or issue, but how do I exit a flutter app that is running on the pi?

closed 2023-08-18 by ardera