ardera/flutter-pi #353: Error MissingPlatformDirectoryException when running flutter gallery

yaliv, 2023-08-26:
I got this error when running flutter gallery with flutter-pi:
```dart
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: MissingPlatformDirectoryException(Unable to get application documents directory)
#0      GetStorage._init (package:get_storage/src/storage_impl.dart:49)
<asynchronous suspension>
#1      new GetStorage._internal.<anonymous closure> (package:get_storage/src/storage_impl.dart:30)
<asynchronous suspension>
#2      main (package:gallery/main.dart:29)
<asynchronous suspension>
```
just stuck on that output.

---

### Specifications

#### Development machine:
- **Flutter SDK version:** 3.10.6
- **Build method:** flutterpi_tool

#### Raspberry Pi:
- **Hardware:** Pi 3 Model B v1.2
- **OS:** 2023-05-03-raspios-bullseye-arm64-lite

### Executed Commands

#### Development machine:
```sh
flutter pub get
flutter run        # the gallery app runs fine
flutterpi_tool build --arch=arm64 --cpu=pi3 --release
rsync -a --info=progress2 ./build/flutter_assets/ pi@192.168.1.70:/home/pi/flutter_gallery
```

#### Raspberry Pi:
```sh
flutter-pi --release ~/flutter_gallery
# or
flutter-pi --release /home/pi/flutter_gallery
```

I always got the same error, both in debug mode and release mode.

chochstedler, 2023-10-01:
Seeing the same issue

pieter-scholtz, 2023-10-02:
https://github.com/flutter/gallery/issues/979

ardera, 2023-10-12:
You need to install xdg-user-dirs: `sudo apt-get install xdg-user-dirs`

It's mentioned in the readme now with #352 .

closed 2023-10-12 by ardera