ardera/flutter-pi #355: Run flutter on x64 linux

sparrowu93, 2023-09-02:
Hey, thank you for the great works.
I have some issues when trying to run flutter code on a x64 debian 12 server with a monitor connected. 
After clone flutter-pi repo and install it and build with flutterpi_tool , then I sync it to the server 
When I run command 
```
flutter-pi --release my_apps_flutter_assets/
```
the out put is
```
flutter: (elided 3 frames from class _Timer and dart:async-patch)
modesetting.c: Couldn't add buffer as DRM fb. drmModeAddFB2WithModifiers: Invalid argument
egl_gbm_render_surface.c: Couldn't add GBM buffer as DRM framebuffer.
window.c: Couldn't present flutter layer on screen. surface_present_kms: Input/output error
```
I searched them and did not got some informations, and hope if any adverse I could get from here would be really appreciated, thanks again !

ardera, 2023-09-02:
This sounds like some driver incompatibility.

What hardware (specifically GPU) are you using? Some Nvidia model?

Does `kmscube` (install with `sudo apt install kmscube`) work?

Can you follow https://github.com/ardera/flutter-pi/wiki/debugging-output-issues and attach the `dmesg.log` file here?

sparrowu93, 2023-09-03:
It's j4125 intel cpu and GPU is build-in.
```
sudo lshw -C display
  *-display                 
       description: VGA compatible controller
       product: GeminiLake [UHD Graphics 600]
       vendor: Intel Corporation
       physical id: 2
       bus info: pci@0000:00:02.0

       logical name: /dev/fb0
       version: 06
       width: 64 bits
       clock: 33MHz
       capabilities: pciexpress msi pm vga_controller bus_master cap_list rom fb
       configuration: depth=32 driver=i915 latency=0 resolution=1366,768
       resources: irq:132 memory:a0000000-a0ffffff memory:90000000-9fffffff ioport:f000(size=64) memory:c0000-dffff
```
kmscube works fine.
[video card info](https://drmdb.emersion.fr/snapshots/7a6f1b8ccd0f)

[dmesg.log](https://github.com/ardera/flutter-pi/files/12505060/dmesg.log)

c68r69s, 2023-09-04:
I just had seemingly the same issue on a Zotac mini pc running debian linux. But here it does work when specifying the pixelformat as RGB656 via:
`flutter-pi --pixelformat RGB565 --release flutter_assets `

sparrowu93, 2023-09-04:
> I just had seemingly the same issue on a Zotac mini pc running debian linux. But here it does work when specifying the pixelformat as RGB656 via: `flutter-pi --pixelformat RGB565 --release flutter_assets `

Oh, thanks for tip, I just tried the arguments and seems those errors are gone, and I will check the monitor later.
Anyway, it was so helpless I felt when I encounter this error, and I am so execitd found the way to walk through.
I will update the result after testing. 
Good day, folks!

ardera, 2023-09-04:
`--pixelformat RGB565` probably works, but that doesn't really fix the bug, only makes it not ocurr.

Issue is that flutter-pi lets the the graphics driver choose the framebuffer format & layout, but that can (for Intel) result in a framebuffer format/layout that uses fb compression and has more than one plane.
And then in some other place in flutter-pi, it's quietly assumed there's only one plane at most, and this assumption is causing this issue here.

bojidartonchev, 2023-09-07:
Hello guys,

I am having the same issue and indeed `--pixelformat RGB565` did the trick.
However, gstreamer after that only shows green screen. @ardera do you think this is connected?

Best regards!

ardera, 2023-09-07:
Just merged a fix. Tested on my own Intel notebook to confirm.

@bojidartonchev Interesting, that shouldn't be happening. Seems like a mesa bug. I'll try reproducing on my own intel hw

closed 2023-09-07 by ardera

bojidartonchev, 2023-09-07:
@ardera Thank you, please let me know if you need any info regarding the video I am playing.

ispo82, 2025-02-09:
@ardera Hi, we got a similar error:
"plugins/gstreamer_video_player/frame.c: Couldn't determine video frame plane sizes. Without plane sizes, only planar framebuffer formats are supported, but the supplied format 
was not planar."

gstreamer version is 1.16
System is Ubuntu Focal X64 on Intel Architecture

  *-display
       description: VGA compatible controller
       product: HD Graphics 500
       vendor: Intel Corporation
       physical id: 2
       bus info: pci@0000:00:02.0
       version: 0d
       width: 64 bits
       clock: 33MHz
       capabilities: pciexpress msi pm vga_controller bus_master cap_list rom
       configuration: driver=i915 latency=0
       resources: irq:139 memory:90000000-90ffffff memory:80000000-8fffffff ioport:f000(size=64) memory:c0000-dffff

Any suggestions for us? Thank you in advance.

PR #357 (2023-09-07): fix multiplanar framebuffers
Implement uploading multi-planar framebuffers. Mutli-planar framebuffers can arise from normal RGB pixel formats when using modifiers (on Intel, the second plane is used as a Color-Control-Surface for framebuffer compression for example)

- fixes #355